Koenkk/zigbee2mqtt #6601: TS0601 motorized roller blinds position not updating (started with 1.18 release)

ASNNetworks, 2021-03-06:
## What happened
I use this product: https://zigbee.blakadder.com/Moes_AM43-0_45_40-ES-EB.html. And with the latest Z2M update, the position does not update anymore. The service calls do still work fine, through Z2M GUI and also HA. I can open, close and set to a desired position. However, the current position is always stuck at 1 within Z2M. I also removed and paired the device again. This worked fine using 1.17 (both stable and dev branch). Using the latest 1.18 release (hassio addon) it now has this bug.

## What did you expect to happen
The position needs to reflacte the actual position, like before. 

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: Conbee II
Adapter firmware version:  latest

playersct, 2021-03-10:
Yes, i have this same problem on CC2652p

Koenkk, 2021-03-10:
Can you provide the debug log of this when changing the position?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

ASNNetworks, 2021-03-11:
Closing:
```
Debug Received MQTT message on 'zigbee2mqtt/Kleedkamer Gordijnen/set' with data 'CLOSE'
Debug Received Zigbee message from 'Kleedkamer Gordijnen', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,255],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'Kleedkamer Gordijnen', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":7,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'Kleedkamer Gordijnen', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID null
```

Opening:
```
Debug Received MQTT message on 'zigbee2mqtt/Kleedkamer Gordijnen/set' with data 'OPEN'
Debug Received Zigbee message from 'Kleedkamer Gordijnen', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'Kleedkamer Gordijnen', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,255],"type":"Buffer"},"datatype":2,"dp":105,"fn":0,"status":0,"transid":0}' from endpoint 1 with groupID null
```

Koenkk, 2021-03-11:
@DomiStyle do you also have this issue with your curtain?

ASNNetworks, 2021-03-11:
Just to make sure it's clear, this is about the roller blind, not the curtains. Since they both have the TS0601 name, I figured to re-iterate. I saw you mention DomiStyle, who also chimed in on a thread about the TS0601 curtains before when that had a bug. The curtains are fine, it's the roller blinds with the issue now. Perhaps DomiStyle has the roller blinds as well and not only the curtains ofcourse. But I just wanted to make sure :)

DomiStyle, 2021-03-11:
@Koenkk Nope, works fine here with 1.18.1.

But I see the problem, DP 1 is treated as position here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/91b494cf2a3f3cfd5aa210c0c112c5c89ade03a8/converters/fromZigbee.js#L1545

However, there is no position in DP 1, it always returns 1.

It works for me because DP 1 is reported before DP 3 (the actual position) but in the log above it gets returned last, setting the position to 1.

I'm not sure why or when this was changed but I can make a pull request for it.

ASNNetworks, 2021-03-11:
This makes perfect sense. Because I can see the correct position flash very briefly before it changes to 1. So that explains it.

closed 2021-03-12 by Koenkk

blowbuh, 2021-03-16:
Hi, my setup ist the following: Debian + Home Assistant (supervised) incl. AddOns (Z2MQTT + MQTT broker). 

I've used the auto-discovery function and was able tu successfully implement this device. However, I cannot see any position data at all, I am not able to adjust this value. As soon as I try to move the position slider in my Z2MQTT add-on, nothing happens.
 
At least I can close and fully open it when I press the up and down arrows.
 
Is it this based on the bug described here or do I have to configure it manually to be be able to see and set position data?

DomiStyle, 2021-03-16:
@blowbuh It's possible. Can you try the latest dev version and see if it still happens? Also, what's your exact model?

blowbuh, 2021-03-16:
@DomiStyle thank you for your quick response. Was not even aware of the fact that there is a dev version for Home Assistant. However I've installed this version and deleted/reconfigured my device. 

Its recognised as TS0601_cover which is correct I guess (some call it AM-43!?). However, I cannot change the position. Neither In Z2M nor in Home Assistant. I can move the slider and see the value changing accordingly in the state section, e.g.:

```
{
    "linkquality": 44,
    "motor_speed": 41,
    "position": 55,
    "running": true,
    "state": "OPEN"
}
```

but nothing happens (device does not react). Other functions are working as before (up, down, stop)

Log:
`Info MQTT publish: topic 'zigbee2mqtt/0x5c0272fffeca9dcc', payload '{"linkquality":44,"motor_speed":41,"position":51,"running":true,"state":"OPEN"}'`

ASNNetworks, 2021-03-17:
@DomiStyle The dev branch has fixed the issue for me (topic starter). The positions are reflected correctly now after a command. Also open/close/set position continues to work fine. Should I leave this ticket open for other users, or should other users create a new separate ticket since my mentioned issue is resolved?

santirguez, 2021-07-23:
Hello,

I am using the dev branch, but still have the same issue. Position is not updating and changing the position manually is not making the motor move.
I tried to remove and re-pair it again but still the same issue. Is thare any specific configuration that should be done?

Thank you!