Koenkk/zigbee2mqtt #6608: SOMGOMS BHT-6000-GCLZ

irzayev, 2021-03-06:
Hi all,
I'm newbie, could you help me to add SOMGOMS BHT-6000-GCLZ device? it is not supported by default:

Support status Unsupported
IEEE address 0x5c0272fffeecccc2
Network address 0x75CC

I'm using zigbee 2538 usb stick

Thank you!

sjorge, 2021-03-06:
Do you have a link to the product page and the database.db entry for the device?

irzayev, 2021-03-06:
> Do you have a link to the product page and the database.db entry for the device?

https://www.aliexpress.com/item/1005001473536551.html

{"id":3,"type":"Unknown","ieeeAddr":"0x5c0272fffeecccc2","nwkAddr":36894,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1615057363699}

sjorge, 2021-03-06:
That's not a whole lot to go on unfortunately, also looks like it's TuYa.
Hopefully someone else more experience with those can help.

insipiens, 2021-03-06:
It looks like a white label of the existing Moes "Tuya" Thermostat BHT-6000 that is already set up, so might just need a changed fingerprint.

closed 2021-04-13 by github-actions[bot]