Koenkk/zigbee2mqtt #6609: No Switch TS0043 Benexmart

nonome1, 2021-03-06:
I have a TS0043 3 button switch which pairs well but I don't have the switches. I tested my deconz key on the deconz program and I have no problem.

```0x60a423fffeb7d131
Device type
EndDevice
Zigbee model
TS0043
Description
Wireless switch with 3 buttons
Support status
Supported

IEEE address
0x60a423fffeb7d131
Network address
0xA81F
Vendor
TuYa
Model
TS0043
Power source
Battery 
Interview completed
Yes

battery
Remaining battery in %
N/A%
action
Triggered action (e.g. a button click)
N/A
linkquality
Link quality (signal strength)
N/Alqi
```

## System Health

version | core-2021.3.2
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 4.19.0-13-amd64
arch | x86_64
timezone | Europe/Paris

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4380
Installed Version | 1.11.3
Stage | running
Available Repositories | 752
Installed Repositories | 53

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Debian GNU/Linux 10 (buster)
-- | --
update_channel | beta
supervisor_version | supervisor-2021.03.4
docker_version | 20.10.2
disk_total | 113.6 GB
disk_used | 13.1 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.12.5), Samba share (9.3.1), MQTT Server & Web client (1.2.0), File editor (5.2.0), Terminal & SSH (9.0.2), ESPHome (1.16.2), Spotify Connect (0.8.2), InfluxDB (4.0.3), Grafana (6.1.3), Portainer (1.4.0), Node-RED (8.1.2), Tydom2MQTT - Beta (release_0.99), Zigbee2mqtt Edge (edge), Zigbee2mqtt (1.18.1-1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 33
views | 13
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

nonome1, 2021-03-06:
I have all which works but I did not have the battery reassembled it is identical to the TS0042 and TS0041

closed 2021-04-13 by github-actions[bot]

CrYoGeNiC6977, 2022-05-10:
> I have all which works but I did not have the battery reassembled it is identical to the TS0042 and TS0041

Can you share your updated blueprint please?