Koenkk/zigbee2mqtt #6615: Aldi C422AC14D41H140.0W supported and not supported

KarlKartoffel, 2021-03-07:
Hi there, 
I'm trying to connect a Aldi C422AC14D41H140.0W to Zigbee2mqtt witch is listed as supported.
By running the paring the interview failed a view times but after some repeating the interview succeeded.
The meta data contains "supported":false.

Client mosqsub|14666-raspberry received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/log', ... (115 bytes))
{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x842e14fffe0e0b0e","supported":false}}

There is no status message or anything comming up by turning the light off/on, but it is listed in the configuration.yaml.

Im not 100% sure if its a bug or if I forgot any thing. I tried to update the light throu the smart phone-app but the app is acutally shit. 

Would be awsome if someone could help me cause my girlfriend is allready annoyed.

thx in advance Karl

Koenkk, 2021-03-07:
Are you on zigbee2mqtt 1.18.1?

KarlKartoffel, 2021-03-07:
> 
> 
> Are you on zigbee2mqtt 1.18.1?

I just pulled an update with the comands listed here:
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html 

by calling --help the version is 1.5.7

so maybe my problem is the updating it self.

KarlKartoffel, 2021-03-08:
I had to install dockers to get over the update issue. but now everything seems tro work. 

Thx again for the fast reply and this beutifull Project.

Grilfriend is happy again and so am I.
have agreat week.

closed 2021-03-08 by KarlKartoffel