Koenkk/zigbee2mqtt #6617: EPERM: operation not permitted, rename database.db.tmp

malosaa, 2021-03-07:
Getting this error once a day, causes to stop and i need to manual restart .

what can be the problem ? 

```
internal/fs/utils.js:220`
    throw err;
    ^

Error: EPERM: operation not permitted, rename 'C:\zigbee2MQTT\zigbee2mqtt\data\database.db.tmp' -> 'C:\zigbee2MQTT\zigbee2mqtt\data\da    at Object.renameSync (fs.js:643:3)
    at Database.write (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\database.js:67:22)
    at Device.save (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\device.js:215:35)
    at Timeout._onTimeout (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\controller.js:152:61)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7) {
  errno: -4048,
  syscall: 'rename',
  code: 'EPERM',
  path: 'C:\\zigbee2MQTT\\zigbee2mqtt\\data\\database.db.tmp',
```

nurikk, 2021-03-07:
C'mon, it'w windows. It requires reboot even if you insert a cd disk to the cd drive 🤣

malosaa, 2021-03-07:
lol

i don't have to restart windows at all.. just re-run npm start.. but its kinda crap this happens every 24h

Koenkk, 2021-03-07:
`operation not permitted` -> permission issue, run Zigbee2MQTT as admin or fix permission on the data directory.

malosaa, 2021-03-11:
i did run as administrator like before and aswell the directory has all admin permissions, it works now for 3 days, but when i edit the configuration file it starts all over again. 

don;t know why

malosaa, 2021-03-16:
@Koenkk  i did run it for a week getting this message again... 
what can cause this problem? the only thing i did was update 1 week ago... 

also i did try to re-install it all but no luck

Koenkk, 2021-03-16:
@malosaa did you restart your computer in the mean time? I have no clue what causes the change of rights.

malosaa, 2021-03-17:
yes i did that, well i did try again. i will report back within a week see if its fixed..

malosaa, 2021-03-26:
Hi @Koenkk 

well i report back, no issue till now, so i waited 9 days. all stable with no errors. i think we can close it now

closed 2021-03-26 by Koenkk

malosaa, 2021-04-14:
@Koenkk 

today i got this issue again, but i could not start the script after the event.. 

```
Error: EPERM: operation not permitted, rename 'C:\zigbee2MQTT\zigbee2mqtt\data\database.db.tmp' -> 'C:\zigbee2MQTT\zigbee2mqtt\data\database.db'
    at Object.renameSync (fs.js:643:3)
    at Database.write (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\database.js:67:22)
    at Database.update (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\database.js:40:14)
    at Device.save (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\device.js:215:35)
    at Controller.databaseSave (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\controller.js:238:20)
    at Timeout._onTimeout (C:\zigbee2MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\controller.js:152:61)
    at listOnTimeout (internal/timers.js:531:17)
    at processTimers (internal/timers.js:475:7) {
  errno: -4048,
  syscall: 'rename',
  code: 'EPERM',
  path: 'C:\\zigbee2MQTT\\zigbee2mqtt\\data\\database.db.tmp',
  dest: 'C:\\zigbee2MQTT\\zigbee2mqtt\\data\\database.db'
}

C:\zigbee2MQTT\zigbee2mqtt>npm start

> zigbee2mqtt@1.18.1 start
> node index.js

Zigbee2MQTT:info  2021-04-14 09:00:19: Logging to console and directory: 'C:\zigbee2MQTT\zigbee2mqtt\data\log\2021-04-14.09-00-10' filename: log.txt
Zigbee2MQTT:info  2021-04-14 09:00:20: Starting Zigbee2MQTT version 1.18.1 (commit #a43d1f3)
Zigbee2MQTT:info  2021-04-14 09:00:20: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:error 2021-04-14 09:00:24: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-04-14 09:00:24: Failed to start zigbee
Zigbee2MQTT:error 2021-04-14 09:00:24: Exiting...
Zigbee2MQTT:error 2021-04-14 09:00:24: Error: Error while opening serialport 'Error: Open (SetCommState): The parameter is incorrect'
```