Koenkk/zigbee2mqtt #6621: CC2351 works on Windows, but not on Linux

dvandonkelaar, 2021-03-07:
## What happened
After flashing my freshly bought CC2531 with the zStack12 [CC2531_DEFAULT_20201127](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20201127.zip)-firmware, I wanted to pair my first zigbee devices.

The host that was running zigbee2mqtt is an Intel NUC NUC8i5BEK with XCP-ng installed on it. The VM that runs zigbee2mqtt is a Debian 10 VM with zigbee2mqtt in a homeassistant-container (with the [Home Assistant Add-on](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt)). The initial connection from the host to the CC2531 was with USB-passthrough. As advised, the CC2531 is connected through an USB extension cable.

Zigbee2mqtt is starting as expected. The device firmware is detected and no errors are logged.
The devices I tried to pair are 2 [ECO-DIM.07 LED dimmers](https://www.zigbee2mqtt.io/devices/Eco-Dim.07.html), which are supported zo no problem (I thought).
Zigbee2mqtt was set to `permit_join: true` and the dimmer was set in pairing mode.
After a while nothing showed up, so I suspected a problem with the USB-passthrough in XCP-ng.
Running `test -w [PORT] && echo success || echo failure` returned success, so I do not have any write problems.

The connection from passthrough was changed to ser2net. In XCP-ng ser2net was configured and within zigbee2mqtt socat whas configured.
Zigbee2mqtt was started and the socat connection was working. XCP-ng netstat showed an established connection to the VM and zigbee2mqtt returned the correct device firmware.
Zigbee2mqtt was set to `permit_join: true` and the dimmer was set in pairing mode.
After a while nothing showed up, so I suspected a problem with interference.
Running `test -w [PORT] && echo success || echo failure` returned success, so I do not have any write problems.

To rule out interference, a raspberry pi (Model B Rev 2) was setup to check if it works on other places in my house.
Zigbee2mqtt was installed and running as a service. The CC2531 was recognized and logging returned the correct firmware version.
Zigbee2mqtt was set to `permit_join: true` and the dimmer was set in pairing mode.
After a while nothing showed up, so I suspected a problem with the firmware.
Running `test -w [PORT] && echo success || echo failure` returned success, so I do not have any write problems.

Using socat to connect to my XCP-ng host also had the same results.

Thereafter, the above steps are reproduced with the following firmwares:
 - Z-Stack 3.0.x: [CC2531_20190425](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.0.x/bin/CC2531_20190425.zip)
 - Z-Stack Home 1.2: [CC2531_SOURCE_ROUTING_20201128](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20201128.zip)

All firmware versions have the same result, absolutely no logging.

To rule out a bad CC2531, I installed zigbee2mqtt on my Windows 10 laptop.
The CC2531 was recognized and logging returned the correct firmware version.
Zigbee2mqtt was set to `permit_join: true` and the dimmer was set in pairing mode.
After a while, both dimmers showed up in zigbee2mqtt and could be controlled through homeassistant. This is on all three CC2531 firmwares.

Using [Serial over Ethernet software](https://www.serial-over-ethernet.com/) to connect to my XCP-ng host also resulted in no logging.

There is probably something simple which is not set up, but after 3-4 days of testing, changing and checking I don't have any idea what could be wrong.

## What did you expect to happen
After setting `permit_join: true` in the zigbee2mqtt config and setting the dimmer in pairing mode, I expected it to (at least) show up in the logging.

## Debug Info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2531
Adapter firmware version: zStack12 (revision [20201128](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20201128.zip))

chris-chensy, 2021-03-08:
I am having the same issue using VM hosted by Mac. The zigbee2mqtt started successfully. But there is no logs or mqtt messages from my network since I updated to HA 2021.03.x. It worked pretty well over the last year until I recently updated the home assistant version (from 2021.02.x). Can I ask which home assistant and supervisor versions you are testing? I haven't tried the windows option yet.

dvandonkelaar, 2021-03-08:
Hi,

The test is with the following versions:
 - host: docker 20.10.5
 - core: core-2021.3.2
 - supervisor: supervisor-2021.03.4

kinappy, 2021-03-08:
I just re-flashed a spare CC2531 using CC2531_DEFAULT_20201127.
Raspberry pi 3b running Raspbian 32bit lite.
Zigbee2mqtt running in docker koenkk/zigbee2mqtt:1.18.1

Zigbee2MQTT:info  2021-03-08 17:52:46: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"a43d1f3","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.18.1"}'

I'm able to pair a device.

Zigbee2MQTT:info  2021-03-08 17:55:34: Successfully interviewed '0x00158d00045fbce8', device has successfully been paired
Zigbee2MQTT:info  2021-03-08 17:55:34: Device '0x00158d00045fbce8' is supported, identified as: Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)

chris-chensy, 2021-03-11:
> Just tested with MAc OS on version 1.18.1. The CC2531 is correctly recognized, but no logs when trying to pair my dimmers (placing the macbook nest to the dimmer).
> 
> After a few days of researching, I still cannot find out what the problem could be.

do you have any devices connected at the moment? do they post any message? My issue is that my existing network is not responding plus no pairing messages.

dvandonkelaar, 2021-03-11:
These are the first devices I try to pair to my setup, so no existing connected devices.

dvandonkelaar, 2021-04-11:
After buying the CC2652RB from slaesh, I got my setup working. The attached USB stick was passed throught to the VM and all was correctly detected.
Why my setup isn't working with the CC2531 is still the question, but it works now.
Technically, the issue is not resolved. But it is fiexed for now.

closed 2021-04-11 by dvandonkelaar