Koenkk/zigbee2mqtt #6623: Philips Hue Liane (4090230P9) not supported

XaviosTheBoss, 2021-03-08:
## What happened
I reseted my bulb (4090230P9) via Bluetooth. My device has successfully been paired but is NOT supported.

info  2021-03-08 01:05:35: Successfully interviewed '0x00178801085807ce', device has successfully been paired
warn  2021-03-08 01:05:35: Device '0x00178801085807ce' with Zigbee model '4090230P9' and manufacturer name 'Philips' is NOT supported.

## What did you expect to happen
My paired device should be supported.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: CC2531
Adapter firmware version: /

database.db:
{"id":10,"type":"Router","ieeeAddr":"0x00178801085807ce","nwkAddr":27726,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"4090230P9","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"4090230P9","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20201012","swBuildId":"1.76.10"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20201012","swBuildId":"1.76.10","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1615161935662}

Thank you in advance

XaviosTheBoss, 2021-03-08:
info  2021-03-08 13:13:16: Starting Zigbee2MQTT version 1.18.1 (commit #a43d1f3)
info  2021-03-08 13:13:16: Starting zigbee-herdsman (0.13.71)

Koenkk, 2021-03-08:
Added, after upgrading to latest dev branch can you share your data/database.db entry of this device?
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

XaviosTheBoss, 2021-03-08:
With dev branch, it works like a charm :) Thank you a lot !

closed 2021-03-09 by Koenkk