Koenkk/zigbee2mqtt #6631: EZSP no longer works after restarting Docker container.

danmed, 2021-03-08:
## What happened
I've got zigbee2mqtt (dev-latest) running in Docker connected to a Sonoff Zigbee Bridge flashed with Tasmota (9.3.1)
I have a Xiaomi Temperature sensor and a Sonoff PLug connected. All works fine.. 
I altered the configuration.yaml as i had set permit_join to true and the button in the GUI wasn't working.. So i altered it to false and then i restarted the docker container.
Once it came back up i can no longer control the plug and the sensor is not updating. Any attempt to change the state of the plug results in an error

https://paste.ubuntu.com/p/mBJQYmVKn6/

## What did you expect to happen
I expected to be able to restart the docker container and still be able to use my devices.

## How to reproduce it (minimal and precise)
Add some devices, and then restart the docker container.

## Debug info
Zigbee2MQTT version: Dev-latest
Adapter hardware: EZSP
Adapter firmware version: Tasmota 9.3.1

danmed, 2021-03-08:
Additionally, though the logs indicate it connected to the coordinator, i cannot actually do anything with it including allow permit.

kirovilya, 2021-03-09:
At the moment, support for the EZSP is at an early stage, so there may be various errors. More detailed logging is needed for detection - https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

https://github.com/Koenkk/zigbee-herdsman/issues/319

danmed, 2021-03-09:
> At the moment, support for the EZSP is at an early stage, so there may be various errors. More detailed logging is needed for detection - https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging
> 
> [Koenkk/zigbee-herdsman#319](https://github.com/Koenkk/zigbee-herdsman/issues/319)

Thanks, i've turned this on and started the container again and magically it's working.. so i'll experiment and get back to you.

danmed, 2021-03-09:
So i've re-paired a device and just lef tit.. it was working fine.. but now it's broken (and i didn't restart the container this time)..

How do i provide the debug logs for you guys?

danilov7624, 2021-03-14:
I am trying to set up Sonoff Zigbee Bridge, but I still can't. Could you please reset the Zigbee2MQTT addon

Hedda, 2021-03-29:
> I am trying to set up Sonoff Zigbee Bridge, but I still can't.

Of-topic but FYI; a tip is to first follow these instructions on how to flash Zigbee chip with ncp-uart-sw_6.7.8_115200.ota firmware:

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

Only after once managed to successfully flashed the ncp-uart-sw_6.7.8_115200.ota to the Zigbee chip should you attemt to follow:

https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

Seams that many miss that they also need to flash the Zigbee chip or read other docs that refer to older or withdrawn versions:

https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

PS: Currently it is only 6.7.8 that is the recommended version however 6.7.9 version should also work (but not any older or newer).

Hedda, 2021-03-29:
By the way, note that Zigbee to WiFi bridges for EZSP is not recommended for production and it does not do well with packed loss:

https://github.com/zigpy/bellows#warning-about-zigbee-to-wifi-bridges

https://www.home-assistant.io/integrations/zha/#warning-about-wi-fi-based-zigbee-to-serial-bridgesgateways

closed 2021-05-07 by github-actions[bot]