Koenkk/zigbee2mqtt #6636: Incorrect voltage value for Mi/Aqara smart home cube

deosrc, 2021-03-09:
## What happened

The state for the cube includes a "voltage" value of almost 3000V:

```json
{
    "angle": 93.87,
    "battery": 86,
    "linkquality": 71,
    "side": 0,
    "voltage": 2975
}
```

I don't actually use the voltage value and suspect most don't so this is probably a low priority to fix. I suspect it is just an issue with decimal positioning.

## What did you expect to happen

The state to have a more realistic voltage value.

## How to reproduce it (minimal and precise)

Not sure as it's just there in the state. Happy to perform steps to provide additional debug information if someone can guide me. 

## Debug info

Zigbee2MQTT version: 1.18.1 commit: a43d1f3
Adapter hardware: ConBee2/RaspBee2
Adapter firmware version: 0x26660700

I'm running in a container on a Raspberry Pi.

Koenkk, 2021-03-09:
It's indeed a known issue, the unit is millivolt. Won't be changed for now since its a breaking change (e.g. people could have setup automations to notify battery low when voltage <= 2500)

deosrc, 2021-03-10:
Thanks for the quick response!

Sorry I realised I didn't actually check what the units are and just assumed it was meant to be volts. Looking at the documentation, it says that the units are volts. Perhaps the documentation should just be updated for now? Maybe make it stand out a bit as it's not using the SI unit.

Koenkk, 2021-03-10:
Good idea, unit will now be marked as millivolts. Assuming this can be closed now (docs will be updated automatically with new release).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-10 by Koenkk