Koenkk/zigbee2mqtt #6638: Philips Hue Edison and Globe re-pair after ~ 15min

rperier, 2021-03-09:
## What happened

Hi,
Recently I have bought one "Hue white Filament bulb G93 E27 bluetooth" (model 8718699688882) and two "Hue white filament Edison E27 LED" (model 929002241201). Everything is detected fine by zigbee2mqtt, from the features point of view everything work. After 15-20 min of uses all my lamps switch to the highest brigthness, suddently. This occurs with my 3 lamps connected or either with only one lamp (with one lamp, other lamps are electrically off).

After investigation, everything time when this problem occurs, I have the following events in zigbee2mqtt:

`
Zigbee2MQTT:debug 2021-03-09 19:26:06: Saving state to file /share/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2021-03-09 19:28:06: Device 'Hue_white_edison_02' announced itself
Zigbee2MQTT:info  2021-03-09 19:28:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Hue_white_edison_02","ieee_address":"0x0017880108ef3e92"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-03-09 19:28:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Hue_white_edison_02"},"type":"device_announced"}'
`

That's the same kind of message when a new device pairs with the network, it seems that my lamps re-pair after some minutes of use.

I have also a lot of periodic:

`
Zigbee2MQTT:debug 2021-03-09 19:35:13: Received Zigbee message from 'Hue_white_edison_02', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16784128,"imageType":274,"manufacturerCode":4107}' from endpoint 11 with groupID 0
`

But it seems to be unrelated to my issue.

I can provide additionnal infos , test stuffs or enable stuffs if necessary

## What did you expect to happen

My lamps work without periodic re-pair

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.1-1  (via Home Assistant 2021.3.x stable channel)
Adapter hardware: zig-a-zig-ah (CC2652R stick)
Adapter firmware version: zStack3x0 revision 20201026

rperier, 2021-03-10:
So I have tested this evening and unfortunately I have exactly the same issues with the new firmware. I have also noticed that the LQI is not very high, around 99 most of the time , the coordinator being at ~3m of the lamp. Other sensors, like a xiaomi aqara temperature at 3m, or even ~6m have a better LQI (and I have a wall beetween the coordinator and my xiaomi sensors, no wall at all between my coordinator and my lamps).

Koenkk, 2021-03-10:
Do you have a CC2531 to sniff the traffic with this when this happens? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

rperier, 2021-03-11:
Unfortunately not (yet), I can try to find one if it is required.

rperier, 2021-03-19:
Another fact, when I replug my edison (one lamp) and move it near to my coordinator , let's say... less than 1 meter, the problem does not happen, no re-pairing.

closed 2021-04-26 by github-actions[bot]