Koenkk/zigbee2mqtt #6645: log_directory is ignored.

freekode, 2021-03-10:
## What happened

I changed value for `log_directory` to:
```
log_directory: /app/data/log/zzz
```
But logs stopped writing at all, in console I have:
```
Zigbee2MQTT:info  2021-03-10 14:36:16: MQTT publish: topic 'zigbee2mqtt/bridge/response/restart', payload '{"data":{},"status":"ok","transaction":"dw7pu-2"}'
Zigbee2MQTT:info  2021-03-10 14:36:18: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
Zigbee2MQTT:info  2021-03-10 14:36:18: Disconnecting from MQTT server
Zigbee2MQTT:info  2021-03-10 14:36:19: zigbee-herdsman stopped
Zigbee2MQTT:info  2021-03-10 14:36:19: Stopped Zigbee2MQTT
Zigbee2MQTT:info  2021-03-10 14:36:19: Logging to console and directory: '/app/data/log/2021-03-10.14-30-31' filename: log.txt
Zigbee2MQTT:info  2021-03-10 14:36:19: Starting Zigbee2MQTT version 1.18.1 (commit #a43d1f3)
Zigbee2MQTT:info  2021-03-10 14:36:19: Starting zigbee-herdsman (0.13.71)
```

## What did you expect to happen
Logs should be saved in directory `/app/data/log/zzz`

## How to reproduce it (minimal and precise)
Change `log_directory` value

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: `revision":20190608`

fkh-bims, 2021-03-10:
advanced:
   log_directory: /var/log
   log_file: zigbee2mqtt.log

Works here

freekode, 2021-03-11:
hm...I'm using official docker image, does it matter?

freekode, 2021-03-17:
Yeah, working, dunno why it doesn't work before.

closed 2021-03-17 by freekode