Koenkk/zigbee2mqtt #6652: Timeout when moving to usb3

Toukite, 2021-03-10:
Hello,

I am running HassOS on a rpi 4b and ssd.
Up until now, I had a sabrent case which was only working with ssd plugged on usb2 port (not booting on usb3, known HA issue).

Everything was running fine for months (and still is, since I rollbacked to this config)
However, I got another case which is compatible with usb3. When I plug my ssd on it and start HA on usb3, everything is working fine (and much more fluid), except zigbee2mqtt ...
I am able to retrieve some values like batteries or temperatures, but I cannot switch the lights on and off anymore (with sonoff basic or xiaomi ssm-u02): I am getting a timeout error message and nothing happens, and I also see the error "failed to execute lqi for ..."

If I shutdown and replug ssd with old casing and on usb2, everything works fine again

Does anyone have an idea about this ? I would much more prefer to be able to migrate to usb3 and this is my only issue !
Thanks a lot !

m0wlheld, 2021-03-11:
Do not operate a Zigbee coordinator device close to an USB3 device. There is a known interference between USB3 and 2.4Ghz networks like Zigbee. 

Use an extension cable, as recommended in the FAQ.

Toukite, 2021-03-11:
Ok ... Thanks for the reply but that's not it.
My rpi and ssd are inside an IT metal box and I already have an extension cable going outside the box to 1 meter of distance where the conbee is plugged. There is nothing near the conbee...
Any other ideas ?

m0wlheld, 2021-03-11:
Did the device change when you re-assembled the case and re-plugged the Conbee? /dev/ttyAMA0 etc.?

Toukite, 2021-03-11:
I use the device path as recommended (with format /dev/serial/by-id/...) so no, it is unchanged :(

closed 2021-04-19 by github-actions[bot]

alexryd, 2021-05-12:
@Toukite Did you manage to solve this? I think I have the same issue with my rpi 4b. I'm migrating from an sd card to an ssd (usb3) and I'm getting the same behaviour (timeouts, "failed to execute lqi for...", "no network route") even though nothing else has changed. My zzh! dongle is also on an extension cable.

Toukite, 2021-05-13:
Nope ... I tried multiple times and each time the same behavior: everything works fine except for zigbee.

So sadly no, I choose zigbee over speed and thus I'm still on usb2 because of this :(