Koenkk/zigbee2mqtt #6657: Sonoff SNZB-02 not reporting battery voltage

avvalex, 2021-03-10:
In many sensors i've found more reliable the voltage than the battery %
Is it possible to expose it, since it's reported in state?

{
    "battery": 100,
    "humidity": 40.9,
    "linkquality": 54,
    "temperature": 25.65,
    "voltage": 3200    <-----------
}

Thanks

sjorge, 2021-03-10:
@Koenkk can it be that battery is sometimes based of batteryPercentRemainint and sometimes calculate from the voltage? Some of my SONOFF device seems to jump around a bit based on what the last attreport was?

Koenkk, 2021-03-10:
@sjorge yes this is indeed the case. If the device also reports the battery percentage remaining by itself `battery: {voltageToPercentage: '3V_2500'}` has to be removed from: https://github.com/Koenkk/zigbee-herdsman-converters/blob/f2e05c44bb94aa2993a584219401da009f2d4b48/devices.js#L14281

sjorge, 2021-03-10:
I’m fairly confident I saw report in my logs. Let me dig into that next time I am at the computer.

sjorge, 2021-03-10:
Actually grabbing the database entry is easy
```
"id":68,"type":"EndDevice","ieeeAddr":"0x00124b0022417e62","nwkAddr":33805,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2316}},"msRelativeHumidity":{"attributes":{"measuredValue":5775}},"genPowerCfg":{"attributes":{"batteryVoltage":29,"batteryPercentageRemaining":164}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":5,"maxRepIntval":1800,"repChange":50},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20200718","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1615412997197}
```

the device is reporting it on its own, there is no reporting entry for it.

but it’s possible to configure reporting via the frontend.

I think we should remove the voltageToPercenrage and add a second reporting line to the configure function.

sjorge, 2021-03-10:
Same for the -03 contact sensor! I do not own the -01 motion sensor to check.

blakadder, 2021-05-15:
I can confirm that percentage remaining and battery_low reports are meaningless.

I have all 3 sonoff sensors in testing for 8 moths now: 
- Their batteries die while the last report in Z2M shows 52% to 56% capacity (varies in that range). 
- Battery low warning never triggered once for any sensor
- Voltage shows a number around 2800 as the last reported one