Koenkk/zigbee2mqtt #6661: Ghost device joining? 0x0000000000000000

jd1900, 2021-03-11:
I'm seeing something like this:
[image]

From time to time, despite having joining disabled.

I force remove it, and disable joining again, but after a few hours or days a new ghost device appears again.

## Debug info
Zigbee2MQTT version: 1.18.1-1 
Adapter hardware: zzh
Adapter firmware version: CC2652R_coordinator_20210120

Koenkk, 2021-03-11:
Can you provide the herdsman debug log of when  this device pairs?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

jd1900, 2021-03-12:
I've added it.
But I'll be quite difficult to find it, as it is not anything that appears all the time.
And if the log disappears...

jd1900, 2021-04-12:
I haven't been able to collect the logs before the dissapear. 

For instance right now I have this:
[image]

closed 2021-04-20 by github-actions[bot]