Koenkk/zigbee2mqtt #6671: Can't bind IKEA E1743 to group

freekode, 2021-03-11:
## What happened
I successfully unbound device from default_bind_group. Then I created a new group, where I added IKEA bulb, then tried to add remote control E1743 to the group. It's failing.
In logs I have
```
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x5c0272fffeacfb9b/1","group":"corridor_light"},"error":"Failed to add from group (Command 0x5c0272fffeacfb9b/1 genGroups.add({\"groupid\":1,\"groupname\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 37989 - 1 - 34 - 4 - 0 after 10000ms))","status":"error"}'
Error Failed to add from group (Command 0x5c0272fffeacfb9b/1 genGroups.add({"groupid":1,"groupname":""}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37989 - 1 - 34 - 4 - 0 after 10000ms))
```

## What did you expect to happen
Remote control should be added to a group

## How to reproduce it (minimal and precise)
Just try to add E1743 to any group

## Debug info
Zigbee2MQTT version: 1.18.1 on docker
Adapter hardware: CC2531
Adapter firmware version: 20190608

MattWestb, 2021-03-12:
You cant adding one IKEA remote to one group ten  its one controller and not one light.
You need doing one new binding of the remote to the group (after unbinding it that you have done from the coordinator).

freekode, 2021-03-12:
Ok, thanks, I fixed it. I did all via web UI.
I removed previously created group.
* Create a new group
* Go to remote control page, open Bind tab
* Select destination: your group, clusters: OnOff. Click bind
* Binding was successful
* Then to the group I added a bulb and all is working. On\Off and brightness too.

closed 2021-03-12 by freekode

MattWestb, 2021-03-12:
If adding light level 2 you can using long press for dimming the lights ;-))