Koenkk/zigbee2mqtt #6673: Problem after update to 1.18.1-1: Error: Command 0x842e14fffea04b1f/1 genOnOff.off({}

r77s, 2021-03-12:
## What happened
I use a CC2531 about a few month and updated now to the latest version 1.18.1-1. Since 2 days I get a error when I try to change for example the lampe state.

`Zigbee2MQTT:info  2021-03-12 07:55:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'LAMPE_E14_1' failed: 'Error: Command 0x842e14fffea04b1f/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"LAMPE_E14_1"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2021-03-12 07:55:51: Publish 'set' 'state' to 'LAMPE_E14_1' failed: 'Error: Command 0x842e14fffea04b1f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))'`

## How to reproduce it (minimal and precise)
Every time I try to change the lamp state

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: I don't know. I bought this already flashed

arripeo, 2021-03-13:
same for me after updating a lot of lag, and error on device under genoff cluster, I have 4 mini sonoff switch rele (router) and 4 lonshono 2 gang rele switch (router)

closed 2021-04-13 by r77s

Faustino-ai, 2021-05-20:
Hello did you solve this issue?