Koenkk/zigbee2mqtt #6675: Xiaomi QBKG03LM hold action

bublikOff, 2021-03-12:
Can not catch hold action on Xiaomi QBKG03LM. 
Instead of it it sends unknown_left.
The same situation with double_left

Koenkk, 2021-03-12:
Can you provide the debug log when this is send?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

bublikOff, 2021-03-12:
Was sligtly wrong about HOLD ... it does sends it as single action instead of hold

**Here is hold log**
`Debug Received Zigbee message from 'hall_light', type 'attributeReport', cluster 'genBasic', data '{"65520":{"data":[170,16,5,65,135,1,1,16,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'hall_light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 5 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/hall_light', payload '{"action":"","linkquality":81,"state_left":"OFF","state_right":"ON","temperature":30}'
Info MQTT publish: topic 'zigbee2mqtt/hall_light/action', payload 'single_right'`

**Here is double click**
`Debug Received Zigbee message from 'hall_light', type 'attributeReport', cluster 'genOnOff', data '{"61440":52742464,"onOff":0}' from endpoint 3 with groupID 0
Debug Received Zigbee message from 'hall_light', type 'attributeReport', cluster 'genOnOff', data '{"61440":203737408,"onOff":0}' from endpoint 3 with groupID 0
Debug Received Zigbee message from 'hall_light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":2}' from endpoint 5 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/hall_light', payload '{"action":"","linkquality":84,"state_left":"ON","state_right":"OFF","temperature":30}'
Info MQTT publish: topic 'zigbee2mqtt/hall_light/action', payload 'undefined_right'`

It seams that the problem is located between HA and z2m as I can cleary see that z2m HOLD 
handles it
[https://i.ibb.co/dBWKQv7/viber-image-2021-03-13-00-02-24.jpg](url)

Koenkk, 2021-03-12:
Can you try disabling the legacy behaviour? (https://www.zigbee2mqtt.io/devices/QBKG03LM.html#device-type-specific-configuration)

Change your config like:

```yaml
devices:
  '0x12345678':
    friendly_name: 'hall_light'
    legacy: false
```

closed 2021-04-12 by Koenkk