Koenkk/zigbee2mqtt #6683: Error: AREQ - ZDO - activeEpRsp after 10000ms - Is something wrong with my adapter?

b0wtie, 2021-03-13:
## What happened
- From one minute to the other, my perfectly working zigbee2mqtt setup stopped working and would not start again.
- Debug log points to a problem with zigbee-herdsman which indicates a problem with the adapter communication.
- Here is the logfile: https://pastebin.com/zAENsNGB
## What did you expect to happen
- I expected it to start normally.
## How to reproduce it (minimal and precise)
### Simply try and start zigbee2mqtt on my Pi3B with my CC2531 adapter.
### I've tried the following things:
- Try different USB Ports / USB cables and such
- Start zigbee2mqtt (via docker-compose) with a fresh configuration
- Reinstall Pi-OS completely and run a fresh docker-image without any configurational changes
- Try a baremetal installation
- Try the dev-branch
#### However:
- None of these things changed the error I am receiving. That is why I guess it has to do with my adapter.
## Debug info
Zigbee2MQTT version: 1.18.1

Adapter hardware: CC2531
Adapter firmware version: 20190608

Please be aware that I am unable to upgrade my adapter. I would have to buy the programming kit - which I have not done yet. If you think the adapter is broken I'd rather spend money on a new one.

Koenkk, 2021-03-13:
Looks like your CC2531 crashes, I suggest using the following firmware instead (switching does not require repairing of devices): https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

b0wtie, 2021-03-17:
Hi @Koenkk,

thank you very much. Today the Debugger and Downloader-Cable arrived - I flashed the source_routing firmware and z2mqtt started and worked like a charm again.

Thanks for the great project!

closed 2021-03-17 by b0wtie

marcelbrueckner, 2021-09-06:
I'm facing the same issue with [cod.m's ZigBee CC2538 Raspberry Pi module](https://github.com/codm/cc2538-raspberry-pi-module).

As OP described, this issue occured out of nowhere. From one moment to the other, everything stopped working. I didn't change anything in my rather small (~20 devices) setup before (that is, no new devices have been added, no update of zigbee2mqtt has been made, no "intense" usage of my devices, ...)

What are my options? I'm using a firmware with revision `20201010` which is afaik the latest version for my module.

```
> zigbee2mqtt@1.18.3 start /app
> node index.js
[34mZigbee2MQTT:debug[39m 2021-09-06 01:00:27: Loaded state from file /share/zigbee2mqtt/state.json
[32mZigbee2MQTT:info [39m 2021-09-06 01:00:27: Logging to console and directory: '/share/zigbee2mqtt/log' filename: zigbee2mqtt_2021-09-06.01-00-25.log
[32mZigbee2MQTT:info [39m 2021-09-06 01:00:27: Starting Zigbee2MQTT version 1.18.3 (commit #1.18.3)
[32mZigbee2MQTT:info [39m 2021-09-06 01:00:27: Starting zigbee-herdsman (0.13.92)
[34mZigbee2MQTT:debug[39m 2021-09-06 01:00:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[25],"extendedPanID":[HIDDEN],"networkKey":"HIDDEN","panID":HIDDEN},"serialPort":{"baudRate":115200,"path":"tcp://192.168.178.45:20108","rtscts":false}}'
2021-09-05T23:00:27.727Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2021-09-05T23:00:27.732Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[HIDDEN],"panID":HIDDEN,"extendedPanID":[HIDDEN],"channelList":[25]},"serialPort":{"baudRate":115200,"rtscts":false,"path":"tcp://192.168.178.45:20108"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}'
2021-09-05T23:00:27.738Z zigbee-herdsman:adapter:zStack:znp:log Opening TCP socket with 192.168.178.45:20108
2021-09-05T23:00:28.285Z zigbee-herdsman:adapter:zStack:znp:log Socket connected
2021-09-05T23:00:28.288Z zigbee-herdsman:adapter:zStack:znp:log Socket ready
2021-09-05T23:00:28.290Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
# ...
2021-09-05T23:00:29.355Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - version - {"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20201010}
2021-09-05T23:00:29.355Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-05T23:00:29.356Z zigbee-herdsman:adapter:zStack:adapter Adapter concurrent: 2
2021-09-05T23:00:29.356Z zigbee-herdsman:adapter:zStack:adapter Detected znp version 'zStack30x' ({"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20201010})
# ...
2021-09-05T23:00:29.521Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- UTIL - getDeviceInfo - {"status":0,"ieeeaddr":"HIDDEN","shortaddr":0,"devicetype":7,"devicestate":9,"numassocdevices":0,"assocdeviceslist":[]}
2021-09-05T23:00:29.522Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-05T23:00:29.522Z zigbee-herdsman:adapter:zStack:startZnp ZNP is already started as coordinator
2021-09-05T23:00:29.524Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":0,"nwkaddrofinterest":0}
2021-09-05T23:00:29.525Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,0,0,0,0,36]
2021-09-05T23:00:29.537Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
2021-09-05T23:00:29.537Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2021-09-05T23:00:29.538Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2021-09-05T23:00:29.538Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2021-09-05T23:00:29.539Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[31mZigbee2MQTT:error[39m 2021-09-06 01:00:39: Error while starting zigbee-herdsman
[31mZigbee2MQTT:error[39m 2021-09-06 01:00:39: Failed to start zigbee
[31mZigbee2MQTT:error[39m 2021-09-06 01:00:39: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
[31mZigbee2MQTT:error[39m 2021-09-06 01:00:39: Exiting...
[31mZigbee2MQTT:error[39m 2021-09-06 01:00:39: Error: AREQ - ZDO - activeEpRsp after 10000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-09-05T23_00_39_604Z-debug.log
```

Koenkk, 2021-09-06:
@marcelbrueckner did you already try reflashing your stick?

marcelbrueckner, 2021-09-06:
I didn't re-flash my module as I have to press some buttons on it and my Raspberry Pi is hid in a somewhat difficult to access area of my house. I hoped to get around it :D

However, I took your question as occasion to do so. I've flashed the [JetHome firmware](https://github.com/jethome-ru/zigbee-firmware/tree/master/ti/coordinator/cc2538_cc2592) *JH_2538_2592_ZNP_UART_20201010* as suggested in [this discussion](https://github.com/Koenkk/zigbee2mqtt/discussions/1568#discussioncomment-342868) (which should be the same my module was shipped) and rebooted my Raspberry Pi my module is attached to (it's forwarded via ser2net to zigbee2mqtt).

zigbee2mqtt started and I was able to access the GUI, but none of my Zigbee devices is pingable, ex.

```
Zigbee2MQTT:debug 2021-09-07 00:40:05: Failed to ping 'bm_of_floor_lamp'
2021-09-06T22:40:05.836Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,182,189]
2021-09-06T22:40:05.836Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,182,189]
2021-09-06T22:40:05.837Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,182] - 189
2021-09-06T22:40:05.837Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":182}
2021-09-06T22:40:05.838Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-06T22:40:05.839Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x0017880109828600:27159,205,0)
2021-09-06T22:40:05.839Z zigbee-herdsman:controller:endpoint Read 0x0017880109828600/11 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
```

I'm afraid I will have to re-pair all devices?

Any idea what causes the module to stop service from one moment to the other? I would really like to avoid this situation in the future because re-flashing and re-pairing all devices isn't a feasible permanent solution.

Koenkk, 2021-09-07:
After reflashing the network route are losts, they should be rebuild automatically, but powering on/off some routers may help to speed this up.

Note sure what the reason is of the `AREQ - ZDO - activeEpRsp after 10000ms`, maybe the firmware crashes but I suggest creating an issue on the firmware repo.

marcelbrueckner, 2021-09-07:
I've power cycled the nearest routers but the network doesn't seem to recover itself. Some routers are seen by zigbee2mqtt, but then they go offline just seconds after it.

The reason I stayed with this CC2538 module until now is that I would have to re-pair everything when changing the coordinator. As I now have to do this anyway, I will switch it out with cod.m's CC2652P which I have lying around here. It's listed as a [supported adapter](https://www.zigbee2mqtt.io/information/supported_adapters.html) and it's using [your Z-Stack firmware](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin), so it might work out well :)

However, some logs for completeness:

<details><summary>Logs from the z2m GUI</summary>

```
debug Device 'bm_of_main_front' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"bm_of_main_front","ieee_address":"0x001788010453adcc"},"type":"device_announce"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"bm_of_main_front"},"type":"device_announced"}'
info MQTT publish: topic 'zigbee2mqtt/bm_of_main_front/availability', payload 'online'
debug Device 'bm_of_main_back' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"bm_of_main_back","ieee_address":"0x0017880104d81898"},"type":"device_announce"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"bm_of_main_back"},"type":"device_announced"}'
info MQTT publish: topic 'zigbee2mqtt/bm_of_main_back/availability', payload 'online'
error Failed to read state of 'bm_of_main_front' after reconnect
error Failed to read state of 'bm_of_main_back' after reconnect
error Failed to read state of 'bm_of_main_back' after reconnect
error Failed to read state of 'bm_of_main_front' after reconnect

... omitted some "failed to ping" messages for other devices ...

info MQTT publish: topic 'zigbee2mqtt/bm_of_main_back/availability', payload 'offline'
error Failed to ping 'bm_of_main_back'
info MQTT publish: topic 'zigbee2mqtt/bm_of_main_front/availability', payload 'offline'
error Failed to ping 'bm_of_main_front'
```
</details>

<details><summary>Debug logs</summary>

```log
Zigbee2MQTT:debug 2021-09-07 22:10:56: Device 'bm_of_main_front' announced itself
Zigbee2MQTT:info  2021-09-07 22:10:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"bm_of_main_front","ieee_address":"0x001788010453adcc"},"type":"device_announce"}'
2021-09-07T20:10:56.245Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2021-09-07 22:10:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"bm_of_main_front"},"type":"device_announced"}'
2021-09-07T20:10:56.249Z zigbee-herdsman:controller:endpoint Read 0x001788010453adcc/11 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-09-07T20:10:56.250Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001788010453adcc:25305/11 (0,0,1)
2021-09-07T20:10:56.251Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":25305,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":230,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,150,0,0,0]}}
2021-09-07T20:10:56.251Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,217,98,11,1,6,0,230,0,30,5,16,150,0,0,0,230]
Zigbee2MQTT:info  2021-09-07 22:10:56: MQTT publish: topic 'zigbee2mqtt/bm_of_main_front/availability', payload 'online'
2021-09-07T20:10:56.255Z zigbee-herdsman:controller:endpoint Read 0x001788010453adcc/11 genOnOff(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-09-07T20:10:56.272Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-09-07T20:10:56.273Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-09-07T20:10:56.273Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-09-07T20:10:56.273Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-09-07T20:10:56.274Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-07T20:10:56.828Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,230,237]
2021-09-07T20:10:56.829Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,230,237]
2021-09-07T20:10:56.829Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,230] - 237
2021-09-07T20:10:56.830Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":230}
2021-09-07T20:10:56.831Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-09-07T20:10:56.832Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x001788010453adcc:25305,205,0)
2021-09-07T20:10:56.832Z zigbee-herdsman:adapter:zStack:adapter Wait 2000ms
[... cut ...]
2021-09-07T20:12:37.063Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-09-07T20:12:37.064Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-09-07T20:12:37.064Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-09-07T20:12:37.064Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-09-07T20:12:37.064Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```
</details>

Thank you @Koenkk for your quick reply and your help on this matter.

Koenkk, 2021-09-08:
> The reason I stayed with this CC2538 module until now is that I would have to re-pair everything when changing the coordinator. As I now have to do this anyway,

This is not true, see https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices

marcelbrueckner, 2021-09-08:
Good to know, thanks. No idea why I thought I'd have to re-pair all devices.

In the meantime I've updated my Home Assistant OS (from 5.13 to 6.3) and after a reboot zigbee2mqtt (which is running as an add-on) is suddenly able to talk to my ZigBee devices. (This doesn't imply power recycling the CC2538 ZigBee module as it is attached to another Pi). I don't know why it suddenly works, but I won't complain :D