Koenkk/zigbee2mqtt #6684: z2m has been working very strangely and delayed for about a week.

salopette, 2021-03-13:
Hello, for about a week I have been observing the z2m working very strangely and with a delay.
My Hue motion detectors react very badly, sometimes not at all.
Has anything been changed in the stability?

Zigbee2MQTT version
1.18.1-dev commit: 07c02f9
Coordinator type
zStack3x0
Coordinator revision
20210120
Frontend version
0.3.81
TI Board -

salopette, 2021-03-13:
`Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"WC-TRV"},"type":"device_announced"}'
Error Publish 'set' 'occupancy_timeout' to 'BWM-Flur' failed: 'Error: Write 0x0017880104b4de0f/2 msOccupancySensing({"pirOToUDelay":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 57669 - 2 - 243 - 1030 - 4 after 10000ms)'
Debug Error: Write 0x0017880104b4de0f/2 msOccupancySensing({"pirOToUDelay":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 57669 - 2 - 243 - 1030 - 4 after 10000ms) at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35) at listOnTimeout (internal/timers.js:554:17) at processTimers (internal/timers.js:497:7)
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupancy_timeout' to 'BWM-Flur' failed: 'Error: Write 0x0017880104b4de0f/2 msOccupancySensing({\"pirOToUDelay\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 57669 - 2 - 243 - 1030 - 4 after 10000ms)'","meta":{"friendly_name":"BWM-Flur"},"type":"zigbee_publish_error"}'`

Koenkk, 2021-03-13:
Can you provide your networkmap?

salopette, 2021-03-13:
[image]
[image]

salopette, 2021-03-13:
I have just reflashed the Ti board again, but it doesn’t bring any improvement.
I haven't changed anything in my setup.

Koenkk, 2021-03-14:
I see the BWM-Flur has no parent, can you try repairing it?

salopette, 2021-03-14:
I already re-paired it this morning!
But it happens again and again that it freezes and no longer reacts to movement.
I can change values such as TimeOut, sensitivity in the front end, this is also practiced, but he and other Hue alarms react very strangely to movement.

Koenkk, 2021-03-14:
After repairing, what is the parent of the problematic hue motion sensor? It looks like there is some incompatibility with its parent.

salopette, 2021-03-14:
How can I see / find out which parent the Hue is looking for?

salopette, 2021-03-15:
I have set log to debug, it behaves very strangely, heaps of messages come in every second, maybe that's why it makes everything slow?
[Neues Textdokument.txt](https://github.com/Koenkk/zigbee2mqtt/files/6138741/Neues.Textdokument.txt)
[Neues Textdokument (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/6138742/Neues.Textdokument.2.txt)

arripeo, 2021-03-15:
I got same delay on my xiaomi wall button WXKG02LM_rev2, and lag on my relay SONOF ZBMINI but also on Lonsonho QS-Zigbee-D02-TRIAC-2C-LN. I got same error on log, so it's seems a diffused problem. More a want to Underline that my CC2530 take 11 end device directly connected, meanwhile by the documentation it's seem that CC2530 can only support 10 end devices.

arripeo, 2021-03-15:
[image]
This is my CC2530 with 11 end devices, instead of 10 devices, I will investigate if each time one of the 11 will be disconnected

Koenkk, 2021-03-16:
@salopette it looks like your TRVS are very spammy indeed. Can you try turning them off completely?

arripeo, 2021-03-16:
@salopette  which is your adapter? cc2530 or cc2531?

salopette, 2021-03-16:
TI LAUNCHXL-CC26X2R1

arripeo, 2021-03-16:
@salopette  it's seem you don't have signal power, what power supply have you for Raspi3? Pay attention that if power supply is not enough it's possible that as RPI work correctly on usb the power out could be cut off.

salopette, 2021-03-17:
I have now completely removed z2m and reinstalled it, but restored the configuration and database without re-pairing the devices. The errors are gone, and everything reacts normally again.

arripeo, 2021-03-18:
My network crashed, my cc2530 + cc2592 stopped to paring, I did the reinstall but I did a downgrade to cc2531 the lag is reduced but it's clear that as the number of devices is not limited (using more routers) when the network grows the coordinator should have more capabilities in terms of memory and baud rate for communication with the Mqtt broker. This is what I've understood anyway could be interesting having some parameters to change polling and cache (if any) in order to tuning the Mqtt data exchange

ASNNetworks, 2021-03-31:
Could be related to my problem: https://github.com/Koenkk/zigbee2mqtt/issues/6304#issuecomment-811472747

hirsts, 2021-04-02:
The symptoms I am seeing are very similar to those mentioned in this thread. I have around 10 Hue Motion sensors which all worked fine using ZHA and a Tasmota flashed Sonoff bridge. They seemed to work initially but are now basically unusable.

I moved to Z2M as ZHA didn't support the Zemismart blinds and the Z2M UI has some great features, much more complete than ZHA. Looks like I might need to run two seperate Zigbee networks. 

I'll try and do some more diagnosis this weekend update this thread.

choeflake, 2021-04-03:
Same here, I had issues several months ago. See #5110. As adviced, I installed the 20201113 firmware (from the dev branch which is not available any more). From that moment everything runs very smooth, better than ever before. And I was very happy!

Now with 1.18.1 and 20210120, the problems are back. I'm not sure if it is exactly the same. Bottom line the overal experience is worse than ever!

McGiverGim, 2021-04-04:
Another one... since yesterday my network is not working anymore... I tried repairing but it seems I can't join the network again.

choeflake, 2021-04-21:
Is there any progress on this issue? 
@Koenkk Will it work when I set back the firmware of issue #5110 (end of november 2020) in combination with Zigbee2mqtt 1.18.2? 
I have the following (old development) binaries: 20201026, 20201113

Koenkk, 2021-04-21:
@choeflake yes this will work.

closed 2021-05-29 by github-actions[bot]

salopette, 2021-06-17:
Hello, my entire network has been very slow for a few days.
And mistakes happen very often.

Error Publish 'set' 'state' to 'Keller-3d-Licht' failed: 'Error: Command 0x01124b001c2e7150/11 genOnOff.off({}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 39627 - 11 - 236 - 6 - 11 after 10000ms)'

salopette, 2021-06-18:
Error Publish 'set' 'led_indication' to 'BWM Bad' failed: 'Error: Write 0x0017880106f48c37/2 genBasic({"51":{"value":false,"type":16}}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4107,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":49127,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'