Koenkk/zigbee2mqtt #6697: Expose switch as light

biau, 2021-03-13:
## Is your feature request related to a problem? Please describe
I'm using z2m with hass and installing zigbee switch micro modules behind my house wall switch.
If te module is a dimmer then z2m expose the endpoints as lights.
If the module is a simple switch, entity is a switch and is not displayed as a light in hass.

## Describe the solution you'd like
Having the opportunity to tell z2m to expose the switch endpoint as light

## Describe alternatives you've considered

## Additional context
Actually I'm using the light switch platform in hass but choosing the type in hass like we can do it in tasmota for example would be great!

m0wlheld, 2021-03-13:
See https://www.zigbee2mqtt.io/integration/home_assistant.html#exposing-switch-as-a-light and close this issue, please.

biau, 2021-03-14:
Really sorry that I missed that.
Thanks for the link

closed 2021-03-14 by biau