Koenkk/zigbee2mqtt #6698: SONOFF SNZB-02 Pairing success but no data

robin-s-007, 2021-03-13:
# Bug Report
SONOFF SNZB-02 is pairing successfully but no data is received.
I have paired several times with 2 devices, and both fail in the same manner.
When looking into this issue, I can across bug "Failure to configure Sonoff SNZB-02 #5562", and used this to try and further diagnose.
I am using MQTT explorer to catch any published data, which remains absent what every I try.

I can not remove the units from the Node-red GUI, and have tried the force_remove method.
And followed the outlined procedure:
"force remove the device
enable pairing
remove/wait 3 sec/insert the batter
It should, hopefully rejoin and pair successfully this time."

I get the following results:
Note two entries below are from a second force_remove as I missed the log output:
Mar 13 20:29:45 zigbee2mqttpi npm[1163]: Zigbee2MQTT:error 2021-03-13 20:29:45: Cannot force remove, device '-m '0x00124b0022692ecc'' does not exist
Mar 13 20:29:45 zigbee2mqttpi npm[1163]: Zigbee2MQTT:info  2021-03-13 20:29:45: MQTT publish: topic '/huis/zigbee2mqtt/bridge/log', payload '{"message":"-m '0x00124b0022692ecc'","type":"device_force_removed_failed"}'

{"id":33,"type":"EndDevice","ieeeAddr":"0x00124b0022692ecc","nwkAddr":44056,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20200718","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1615666167349}

>>enable pairing + remove bat for 30 sec
Mar 13 20:32:49 zigbee2mqttpi npm[1163]: Zigbee2MQTT:info  2021-03-13 20:32:49: MQTT publish: topic '/huis/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"zolderkamer-3d-printer-TH-spool2","ieee_address":"0x00124b0022692ecc"},"type":"device_announce"}'
Mar 13 20:32:49 zigbee2mqttpi npm[1163]: Zigbee2MQTT:info  2021-03-13 20:32:49: MQTT publish: topic '/huis/zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"zolderkamer-3d-printer-TH-spool2"},"type":"device_announced"}'

>>After pairing:
Mar 13 20:34:31 zigbee2mqttpi npm[1163]: Zigbee2MQTT:info  2021-03-13 20:34:31: MQTT publish: topic '/huis/zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"zolderkamer-3d-printer-TH-spool2","ieee_address":"0x00124b0022692ecc"},"type":"device_announce"}'
Mar 13 20:34:31 zigbee2mqttpi npm[1163]: Zigbee2MQTT:info  2021-03-13 20:34:31: MQTT publish: topic '/huis/zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"zolderkamer-3d-printer-TH-spool2"},"type":"device_announced"}'
{"id":33,"type":"EndDevice","ieeeAddr":"0x00124b0022692ecc","nwkAddr":36026,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"TH01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"TH01","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20200718"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20200718","zclVersion":1,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1615667569511}

## What happened
Device paired, but no data is coming in, not immediately, not after several hours, not after breathing in sensors. 

## What did you expect to happen
Device would provide data (and that this is published on MQTT), on timely basis and/or on changing environment.
Both should happen according to z2m device description page 

## How to reproduce it (minimal and precise)
Pair device, blow in to sensor, and wait -> no data.

## Debug Info
Zigbee2MQTT version: 1.17.1
Adapter hardware: CC2531
Adapter firmware version: 20190619
Nr of devices: 33

sjorge, 2021-03-14:
Does it show anything under the reporting tab in the frontend?
If it keeps failing to setup reporting, after the change from the issue you found, the error will be eaten so you won't see it.

robin-s-007, 2021-03-14:
Could you tell me were I can find this reporting tab ? 
As in my Node-red page, there is no reporting tab, just the device list, device info, rename and group button and the network maps and version info. 
On my Pi I have the logs, but they contain huge amounts of recurring info that I have no idea what to look for.
Thanks,
Robin

sjorge, 2021-03-14:
It should be in the frontend that  comes with z2m under de device page. So probably not in the node-red one.

robin-s-007, 2021-03-14:
Until your remark, I was unaware of this feature, and it was not configured (not even in the file). 
So I enabled it, and it is a really nice feature, Kudos for the developement. 

I have rejoined the device, and map screenshots of all the screens that might be interesting. 
See attached images. 
Please let me know if your missing something.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sjorge, 2021-03-14:
OK, no wonder your not getting data.
You need to have 4 entries on the reporting tab. (On phone so can’t screenshot)

@koenkk I think due to the try catch you added in the earlier linked issue, the device silently fails to configure and won’t retry unless unless manually requested. Does that make sense?

sjorge, 2021-03-14:
[image]

should look like this

robin-s-007, 2021-03-14:
Ok, thanks for the insight.
I guess that this is something that z2m needs to setup automatically ?

sjorge, 2021-03-14:
Yes, when pairing the device. But I think it’s failing but eating the error so not retrying

robin-s-007, 2021-03-27:
Thanks for the insights.
I was wondering, if this issue has now been 'automatically' been registered on the bug list ?  Or do I need to do take additional action ?
For now I will put the SONOFF SNZB-02 sensors aside.

sjorge, 2021-03-27:
@Koenkk ping

Koenkk, 2021-03-27:
@sjorge it's indeed eating the error, error will be logged now in the latest dev

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

NijsKei, 2021-03-30:
sorry dat ik in het nederlands reageer. Mijn engels is niet helemaal top.
Ik heb al dagen dat mijn CC2531 uitvallen. Ik heb er 6 en als ik ze opnieuw flash dan werken ze weer.
Jammer genoeg maar voor 1 dag of korter. Nou had ik net weer een uitval en toen heb ik en client Sonoff ZBR03 van de stroom gehaald en weer op de stroom en alles werkte weer. Kan het zijn dat er apparaten zijn die storen op de CC2531?

Koenkk, 2021-03-30:
@NijsKei waarschijnlijk is je adapter niet sterk genoeg, ik beveel een upgrade naar de ZZH of Slaesh aan: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

NijsKei, 2021-03-30:
@KoenKK Bedankt voor je snelle reactie. Ik heb een ZZH in bestelling maar duurt nog een week. Ik vind het wel vreemd dat ik met de 2531 al 2 jaar geen problemen heb. Ik had er toen alleen temperatuur sensoren aanhangen. Het probleem is ontstaan toen ik de sonoff ZBR03 aan het netwerk hing. Daar heb ik er 6 van en nu 4 in gebruik. Ik hoop dat maar 1 van de 4 het probleem geeft dan vervang ik die. De zbr03's refreshen wel erg vaak misschien dat dat het probleem is. Nu nog uitzoeken hoe ik dat kan verminderen. Het is een hele studie op zich maar doe het met veel plezier.

robin-s-007, 2021-04-06:
@Koenkk and @sjorge 
I just installed the latest zigbee2mqtt version 1.18.2, and tested the SNZB-02 pairing, and am very happy to report that the issue is fixed.
The SNZB-02s are now finishing their interrogation and are completing the pairing process successfully.
Many thanks for your support and fixing the issue.

Best Regards,

Robin

closed 2021-04-06 by Koenkk

crooy, 2021-09-25:
I am on the latest dev branch using docker, and i'm still getting this error.
```
Configure failed: Error: Bind 0x00124b0022692499/1 msTemperatureMeasurement from '0x00124b002390d9de/1' failed (AREQ - ZDO - bindRsp after 10000ms)
info Successfully configured '0x00124b0022692499'
```

syfq91, 2021-09-30:
I’m facing the same issues as well for my 2 Sonoff sensors. 
I'm using the CC2652P usb stick from ebony, Zigbee2mqtt using docker-compose.

bvbunnik, 2021-10-01:
I can report the same issue as well.
zzh-stick, SNZB-02 sensor, reporting to pair successfully but will not report any data.
Same error as crooy two comments above.
N.B. I have two other SNZB-2 sensors that are functioning perfectly fine...

sjorge, 2021-10-01:
They are extreme picky when pairing, when replacing a broken one with a fresh one it took 5 tries before it was OK.
Make sure to force remove the device between tries otherwise z2m won't poke the device in a way where it stays awake.

nickmcgill, 2021-10-08:
I have just pulled the latest-dev docker image, and the configuration fails.

Koenkk, 2021-10-09:
Locking this in favour of https://github.com/Koenkk/zigbee2mqtt/issues/8971 

TLDR; I've ordered this device to further debug it.