Koenkk/zigbee2mqtt #6704: Request for new device - Aurora Aone Smart Double Socket UK

Napandee, 2021-03-14:
Tried following "How to support new devices" but running into issues under step 3 "Adding converter(s) for your device"
Debug is enabled and the only action is to power each socket on and off but cannot see any clear indication on said action in debug log or in "fromzigbee.js"

Please note, new to Zigbee, this is the first device I tried to add and running this in docker.

There is an entry in the - https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js
This is similar to the socket I've been trying to add - The information below
 {
        zigbeeModel: ['SingleSocket50AU'],
        model: 'AU-A1ZBPIAB',
        vendor: 'Aurora Lighting',
        description: 'Power plug Zigbee EU',
        fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement],
        exposes: [e.switch(), e.power(), e.voltage(), e.current()],
        toZigbee: [tz.on_off],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
            await reporting.onOff(endpoint);
        },

Tried modifying this as well but couldn't get that to work either

## Information about the device + link
This is the entry I was trying to add/generate
{
    zigbeeModel: ['DoubleSocket50AU'],
    model: 'AU-A1ZBDSS',
    vendor: 'Aurora Lighting',
    description: 'Aurora Double Smart Socket UK',
    supports: 'On/Off, Dimmable Power buttons, Power consumption',
    fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.power(), e.voltage(), e.current()],
},

Link to the product - https://auroralighting.com/gb/ProductDetail/AU-A1ZBDSS

## data/database.db entry of the device
Can't find any instructions on how to show database entry information - Please instruct if needed.
Only entry Debug wise received from the socket is the following,

Received Zigbee message from 'Cabin Power Socket', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":20,"imageType":5129,"manufacturerCode":4636}' from endpoint 1 with groupID 0

If more information is needed/missing let me know and I'll gather/add more information.

Koenkk, 2021-03-14:
Can you try with:

```js
{
    zigbeeModel: ['DoubleSocket50AU'],
    model: 'AU-A1ZBDSS',
    vendor: 'Aurora Lighting',
    description: 'Double smart socket UK',
    fromZigbee: [fz.identify, fz.on_off, fz.electrical_measurement],
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'),
        e.power().withEndpoint('left'), e.voltage().withEndpoint('left'), e.current().withEndpoint('left'),
        e.power().withEndpoint('right'), e.voltage().withEndpoint('right'), e.current().withEndpoint('right')],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1, multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genIdentify', 'genOnOff', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint2);
    },
},
```

You should be able to control it via the frontend by clicking on the device -> exposes tab.

Napandee, 2021-03-15:
Seem to be working, I now have the following under the exposes tab,

[image]
[image]

On/off switches are working 

And devices and entities are being displayed in Home Assistant as expected.
Huge thanks

Koenkk, 2021-03-16:
The voltage and current show as N/A, looks like the device doesn't support it? If yes I will remove this part and integrate this.

Napandee, 2021-03-16:
Yeah, it looks like it only supports current power consumption only, so
feel free to remove it from the device.

Thanks for the help.

Koenkk, 2021-03-16:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-16 by Koenkk