Koenkk/zigbee2mqtt #6707: Awox / EGLO id:33944 LED E27 RGB support needed

willyfog78, 2021-03-14:
Hello

This is my first post so be gentle pls. 
CASE: AwoX E27 Tunable White RGB bulb under brand EGLO is not allowing for tempcontrol of LEDs.
Model number on the box ID 33944.
First you need to connect it to phone via bluetooth. At that point colortemp is working. 
Then you need to update software via bluetooth. Only then you can pair it with zigbee.
Unfortunatelly it pairs as AwoX | 33951/33948 but there is no colortemp control.

Since I am running home assistant with zigbee2mqtt add-on I have tried to modify devices.js by:
- adding colortemp feature in AwoX | 33951/33948 but this did not worked.
- create new device AwoX 33944 with the following setup:
{
    zigbeeModel: ['TLSR82xx'],
    model: '33944',
    vendor: 'AwoX',
    description: 'LED E27 RGB',
    extend: preset.light_onoff_brightness_colortemp(),
},
which was also unsucessfull.

Since this product attracted a lot of attention in Poland it would be nice if we could figure it out.
-- | --

## Logs when pairing
Zigbee2MQTT:debug 2021-03-14 12:32:03: Received Zigbee message from '0xa4c1389fb31d02fc', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"0122052017"}' from endpoint 1 with groupID 0
2021-03-14T11:32:03.849Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-14T11:32:03.850Z zigbee-herdsman:controller:device:log Interview - got 'softwareBuildID' for device '0xa4c1389fb31d02fc'
2021-03-14T11:32:03.851Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - simpleDescReq - {"dstaddr":47028,"nwkaddrofinterest":47028,"endpoint":3}
2021-03-14T11:32:03.852Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,4,180,183,180,183,3,39]
2021-03-14T11:32:03.861Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96]
2021-03-14T11:32:03.861Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,4,0,96]
2021-03-14T11:32:03.862Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 4 - [0] - 96
2021-03-14T11:32:03.863Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - simpleDescReq - {"status":0}
2021-03-14T11:32:03.863Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-14T11:32:03.891Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,22,69,132,180,183,0,180,183,16,3,143,18,13,1,1,2,80,255,81,255,2,80,255,81,255,84]
2021-03-14T11:32:03.892Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,22,69,132,180,183,0,180,183,16,3,143,18,13,1,1,2,80,255,81,255,2,80,255,81,255,84]
2021-03-14T11:32:03.892Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 22 - 2 - 5 - 132 - [180,183,0,180,183,16,3,143,18,13,1,1,2,80,255,81,255,2,80,255,81,255] - 84
2021-03-14T11:32:03.894Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":47028,"status":0,"nwkaddr":47028,"len":16,"endpoint":3,"profileid":4751,"deviceid":269,"deviceversion":1,"numinclusters":2,"inclusterlist":[65360,65361],"numoutclusters":2,"outclusterlist":[65360,65361]}
2021-03-14T11:32:03.895Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-03-14T11:32:03.896Z zigbee-herdsman:controller:device:log Interview - got simple descriptor for endpoint '3' device '0xa4c1389fb31d02fc'
2021-03-14T11:32:03.900Z zigbee-herdsman:controller:device:log Interview - completed for device '0xa4c1389fb31d02fc'
2021-03-14T11:32:03.903Z zigbee-herdsman:controller:log Succesfully interviewed '0xa4c1389fb31d02fc'
Zigbee2MQTT:info  2021-03-14 12:32:03: Successfully interviewed '0xa4c1389fb31d02fc', device has successfully been paired
Zigbee2MQTT:info  2021-03-14 12:32:03: Device '0xa4c1389fb31d02fc' is supported, identified as: AwoX LED white (33951/33948)
Zigbee2MQTT:info  2021-03-14 12:32:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"LED white","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"33951/33948","supports_ota":false,"vendor":"AwoX"},"friendly_name":"0xa4c1389fb31d02fc","ieee_address":"0xa4c1389fb31d02fc","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-03-14 12:32:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"LED white","friendly_name":"0xa4c1389fb31d02fc","model":"33951/33948","supported":true,"vendor":"AwoX"},"type":"pairing"}'
Zigbee2MQTT:info  2021-03-14 12:32:03: MQTT publish: topic 'homeassistant/light/0xa4c1389fb31d02fc/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_temp":false,"command_topic":"zigbee2mqtt/0xa4c1389fb31d02fc/set","device":{"identifiers":["zigbee2mqtt_0xa4c1389fb31d02fc"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c1389fb31d02fc","sw_version":"Zigbee2MQTT 1.18.1"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"hs":false,"json_attributes_topic":"zigbee2mqtt/0xa4c1389fb31d02fc","name":"0xa4c1389fb31d02fc","schema":"json","state_topic":"zigbee2mqtt/0xa4c1389fb31d02fc","unique_id":"0xa4c1389fb31d02fc_light_zigbee2mqtt","xy":false}'
Zigbee2MQTT:info  2021-03-14 12:32:03: MQTT publish: topic 'homeassistant/sensor/0xa4c1389fb31d02fc/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c1389fb31d02fc"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c1389fb31d02fc","sw_version":"Zigbee2MQTT 1.18.1"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c1389fb31d02fc","name":"0xa4c1389fb31d02fc linkquality","state_topic":"zigbee2mqtt/0xa4c1389fb31d02fc","unique_id":"0xa4c1389fb31d02fc_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

## Information about the device + link
This was purchased in Castorama in Poland: https://www.castorama.pl/zarowka-led-awox-g95-e27-1300-lm-2700-6500-k-id-1134885.html
On an official Eglo page there is no such product with 1300 lumen and E27 RGB - ID: 33944. There is something similar but smaller: https://datasheet.eglo.com/tds/pol-PL/11586

## data/database.db entry of the device
I don't know how to do that.
[image]
[image]

Koenkk, 2021-03-14:
Can you provide the data/database.db entry of this device?

willyfog78, 2021-03-14:
Here you go, found it

{"id":11,"type":"Router","ieeeAddr":"0xa4c1389fb31d02fc","nwkAddr":47028,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1615721523845}

willyfog78, 2021-03-14:
in the mean time we have temporary solution -  one of the contributors from facebook suggested external converter such as avox.js:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const libExposes = require('zigbee-herdsman-converters/lib/exposes');
const e = libExposes.presets;
const exposes = [e.light_brightness_colortemp_colorxy(true), ...([e.effect()])];
const localFromZigbee = [fz.color, fz.on_off, fz.brightness, fz.level_config,fz.ignore_basic_report];
const localToZigbee = [
    tz.light_onoff_brightness, tz.light_color_colortemp, tz.ignore_transition, tz.ignore_rate, tz.light_brightness_move,
    tz.light_colortemp_move, tz.light_brightness_step, tz.light_colortemp_step, tz.light_hue_saturation_move,
    tz.light_hue_saturation_step, tz.light_colortemp_startup, tz.level_config, tz.power_on_behavior, tz.light_color_options,
    ...([tz.effect])];
const device = {
    zigbeeModel: ['TLSR82xx'],
    model: '33944',
    vendor: 'AwoX',
    supports: '',
    description: 'LED RGB & Brightness',
    fromZigbee: localFromZigbee,
    toZigbee: localToZigbee,
    exposes: exposes,
};

module.exports = device;

and it works after copying to location from configuration file: data_path
then I have enabled awox.js in the zigbee2mqtt settings/external converters

Koenkk, 2021-03-14:
Added it, can you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

willyfog78, 2021-03-14:
Thanx will check and revert back with outcome

Pobierz aplikację Outlook dla systemu Android<https://aka.ms/AAb9ysg>

Koenkk, 2021-03-16:
Thanks, if it works please also provide the data/database.db entry after that

willyfog78, 2021-03-19:
Hi , you are The Man

 

Thanks for adding it works, however color temp at startup is not working. 

Please find attached database.db entry for this one.

 

Best regards

Piotr

Koenkk, 2021-03-19:
- I can't find the attachment.

> Thanks for adding it works, however color temp at startup is not working.

Then this device doesn't support it (not uncommon)

Remigiusz2020, 2021-04-10:
> Czy możesz podać wpis data / database.db tego urządzenia?

Hi, I just want to know if you add this update to newest version z2m? I got 2 bulbs rgb  from awox and it dosnt work propely. HA is discover them as normal bulbs. Please let me know. 
thanks for help

Remigiusz2020, 2021-04-22:
> Dodałem, czy możesz sprawdzić, czy działa?
> 
> Zmiany będą dostępne w gałęzi deweloperów za kilka godzin. ( https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html )

Hej, czy będzie to dostępne również dla zwykłych urzytkowników - nie developer branch?

closed 2021-05-30 by github-actions[bot]