Koenkk/zigbee2mqtt #6712: Heiman Smart Temperature & Humidity Sensor

jeremybreeschoten, 2021-03-14:
## Information about the device + link
Heiman Smart Temperature & Humidity Sensor / https://nl.aliexpress.com/item/32908832168.html?spm=a2g0s.9042311.0.0.6b4a4c4d6nkhmp

## data/database.db entry of the device
I found this all in the database.db entry

`{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014d95e2c","nwkAddr":0,"manufId":0,"epList":[242,47,13,12,110,11,10,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1024,"inClusterList":[25],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":2080,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"EndDevice","ieeeAddr":"0x000d6f0016195351","nwkAddr":43873,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"COSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"COSensor-EM","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.9","swBuildId":"1.1.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014d95e2c","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d95e2c","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.3.9","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1615744441112}
{"id":3,"type":"EndDevice","ieeeAddr":"0x000d6f0014f6ab8e","nwkAddr":41471,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"SmokeSensor-EM","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,1282],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SmokeSensor-EM","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":1,"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.4.20","swBuildId":"1.1.1"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014d95e2c","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d95e2c","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":18,"stackVersion":2,"hwVersion":16,"dateCode":"2018.4.20","swBuildId":"1.1.1","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1615744871252}
{"id":4,"type":"EndDevice","ieeeAddr":"0xbc33acfffe058c7b","nwkAddr":38533,"manufId":4619,"manufName":"HEIMAN","powerSource":"Battery","modelId":"HT-EF-3.0","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,1026,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"HT-EF-3.0","manufacturerName":"HEIMAN","powerSource":3,"zclVersion":2,"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2019.9.6"}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":770,"inClusterList":[0,3,1029],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":2,"hwVersion":16,"dateCode":"2019.9.6","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1615745037783}`

jeremybreeschoten, 2021-03-14:
they look the same as: HS1HT

sjorge, 2021-03-14:
Can you try by updating

https://github.com/Koenkk/zigbee-herdsman-converters/blob/f8c9564b23dbcebd2ed02f2fb02ffefacf0cdb83/devices.js#L9787

```
    {
        zigbeeModel: ['HT-EM', 'TH-EM', 'TH-T_V14'],
        model: 'HS1HT',
        vendor: 'HEIMAN',
        description: 'Smart temperature & humidity Sensor',
        exposes: [e.battery(), e.temperature(), e.humidity()],
        fromZigbee: [fz.temperature, fz.humidity, fz.battery],
        toZigbee: [],
        meta: {configureKey: 2, battery: {voltageToPercentage: '3V_2500'}},
        whiteLabel: [{vendor: 'Ferguson', model: 'TH-T_V14'}],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['msTemperatureMeasurement']);
            const endpoint2 = device.getEndpoint(2);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['msRelativeHumidity', 'genPowerCfg']);
            await reporting.temperature(endpoint1);
            await reporting.humidity(endpoint2);
            await reporting.batteryVoltage(endpoint2);
            await reporting.batteryPercentageRemaining(endpoint2);
        },
    },
```

to 

```
    {
        zigbeeModel: ['HT-EM', 'TH-EM', 'TH-T_V14', 'HT-EF-3.0'],
        model: 'HS1HT',
        vendor: 'HEIMAN',
        description: 'Smart temperature & humidity Sensor',
        exposes: [e.battery(), e.temperature(), e.humidity()],
        fromZigbee: [fz.temperature, fz.humidity, fz.battery],
        toZigbee: [],
        meta: {configureKey: 2, battery: {voltageToPercentage: '3V_2500'}},
        whiteLabel: [{vendor: 'Ferguson', model: 'TH-T_V14'}],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['msTemperatureMeasurement']);
            const endpoint2 = device.getEndpoint(2);
            await reporting.bind(endpoint2, coordinatorEndpoint, ['msRelativeHumidity', 'genPowerCfg']);
            await reporting.temperature(endpoint1);
            await reporting.humidity(endpoint2);
            await reporting.batteryVoltage(endpoint2);
            await reporting.batteryPercentageRemaining(endpoint2);
        },
    },
```

jeremybreeschoten, 2021-03-14:
@sjorge I dont know exactly what to do? I am new to this. I have the addon in my Home Assistant installation. I cant find that devices.js file.

sjorge, 2021-03-15:
There are some instructions here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#2-adding-your-device

jeremybreeschoten, 2021-03-15:
> There are some instructions here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#2-adding-your-device

I have the addon in Home Assistant. that sends me to: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#home-assistant-add-on-hassio-zigbee2mqtt

Multipage, 2021-03-15:
@jeremybreeschoten : it's a pain in the *ss to do it within home assistant and the info you'll find is outdated.

I have 2  CC2531' s for this. 1 is a sniffer, the other a coordinator. So what i did was:
On a local machine ( Ubuntu )  i installed an MQTT server and installed and configured Zigbee2MQTT.
Connected a CC2531 coordinator  edited the devices.js there.
Restart the zigbee2mqtt. 
Paired the device. Check if everything works and then created pull request here.
So no HA installation . Just those 2 programs.

You can connect to the Zigbee2MQTT  webUI through http://127.0.0.1:8099 ( well , in my case it is configured at that port) 
This way is SO much easier then fiddling around in your HA and some files you CAN'T overrule ( e.g. fromZigee and toZigbee ) so you still won't be able to do all the testing if you need to change those files. I found that problem when i was helping out to get the xfinity keypad working on ZIgbee2MQTT.

So unless you want frustration and headaches i advice you to do the same as i did. 
DON'T USE YOUR 'LIVE' ZIGBEE COORDINATOR . Always use a spare dedicated coordinator for this kind of work.

jeremybreeschoten, 2021-03-15:
> @jeremybreeschoten : it's a pain in the *ss to do it within home assistant and the info you'll find is outdated.
>
> I have 2 CC2531' s for this. 1 is a sniffer, the other a coordinator. So what i did [...]

Thank you for the information, So i am little forced to buy me a second one for this. Or to just leave the device. It is a pity that for Noobs with the zigbee2mqtt addin it is difficult to do this.
Cause i think that the update of sjorge would be the fix :)

Multipage, 2021-03-15:
@jeremybreeschoten If i were you i would buy a second CC2531. They are cheap and  in the future if you have an unsupported device you are all set to build in support very fast.  It's worth the investment.

I ran into a doorsensor that wasn't supported the other day. Fired up everything and added support. It's already merged and will be available in the next version. It' s either that ...or hoping ...and waiting for someone else to get the same device and build support for it. I personally wouldn't wait for someone else.

jeremybreeschoten, 2021-03-15:
> @jeremybreeschoten If i were you i would buy a second CC2531. They are cheap and in the future if you have an unsupported device you are all set to build in support very fast. It's worth the [...]

Multipage, thank you for the feedback. Is there a clear manual to do this at home? TO do those actions then?

Multipage, 2021-03-15:
For MQTT installation i would use ( don't look at version number Ubuntu, it doesn't matter):
https://www.digitalocean.com/community/tutorials/how-to-install-and-secure-the-mosquitto-mqtt-messaging-broker-on-ubuntu-18-04

For Zigbee2MQTT use:
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
BUT!:
Replace in step 2, this line:
sudo chown -R pi:pi /opt/zigbee2mqtt
into 
sudo chown -R $USER:$USER /opt/zigbee2mqtt
Also after installation make sure you have rights to open the port by :
chown $USER:USER /dev/ttyACM0 ( or whatever th port the CC2531 is on) 

When done (including starting it as a service)switch to dev branch:
https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html
I do this different because i'm not running it as a service so the switch to dev fails. I'm cloning the full dev git repo but then i have to edit the configuration.yaml again.

BTW: you have 99,9999% change your sensor will work with the solution from sjorge. Heiman has different versions of the same hardware and they all work with the same code. 
I posted something simulair for their doorsensors so hopefully the devices.js will be cleaned from this.
Do you know how to make a PR ? Otherwise i will do it for you. 
Since you are living in Limburg (as far as i can see) and i'm near Rotterdam it's a bit to far to come over here to test it :) It would be 3+ hour drive for 10 seconds of work. A bit to much, right?
I'm convinced it will work!

jeremybreeschoten, 2021-03-24:
Dear team,

i managed to adjust the devices.js file and paired it in Ubuntu with the zigbee2mqtt and the Mosquitto Broker

THis is debug result 
`Zigbee2MQTT:debug 2021-03-24 20:39:14: Received MQTT message on 'zigbee2mqtt/bridge/extensions' with data '[]'
Zigbee2MQTT:info  2021-03-24 20:39:15: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"b6085fc","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.18.1"}'
Zigbee2MQTT:info  2021-03-24 20:40:04: Device '0xbc33acfffe058c7b' joined
Zigbee2MQTT:info  2021-03-24 20:40:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-03-24 20:40:04: Starting interview of '0xbc33acfffe058c7b'
Zigbee2MQTT:info  2021-03-24 20:40:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-03-24 20:40:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe058c7b"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-03-24 20:40:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe058c7b"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-03-24 20:40:04: Device '0xbc33acfffe058c7b' announced itself
Zigbee2MQTT:info  2021-03-24 20:40:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-03-24 20:40:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe058c7b"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-03-24 20:40:17: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"modelId":"HT-EF-3.0"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:20: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"HEIMAN"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:23: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:26: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:29: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"appVersion":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:32: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:35: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"hwVersion":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:38: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2019.9.6"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:41: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-24 20:40:45: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2021-03-24 20:40:45: Successfully interviewed '0xbc33acfffe058c7b', device has successfully been paired
Zigbee2MQTT:info  2021-03-24 20:40:45: Device '0xbc33acfffe058c7b' is supported, identified as: HEIMAN Smart temperature & humidity Sensor (HS1HT)
Zigbee2MQTT:info  2021-03-24 20:40:45: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart temperature & humidity Sensor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HS1HT","supports_ota":false,"vendor":"HEIMAN"},"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-03-24 20:40:45: Configuring '0xbc33acfffe058c7b'
Zigbee2MQTT:info  2021-03-24 20:40:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart temperature & humidity Sensor","friendly_name":"0xbc33acfffe058c7b","model":"HS1HT","supported":true,"vendor":"HEIMAN"},"t [... cut ...] ceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:419:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
`

please tell me what to do? @Multipage @Koenkk

Multipage, 2021-03-26:
@jeremybreeschoten It seems the config isn't right for this one.
You could try to edit this one :
fingerprint: [{modelID: 'HT-N', manufacturerName: 'HEIMAN'}],

Change to :
zigbeeModel: ['HT-N', 'HT-EF-3.0'],

And see if that's working better

Remove the 'HT-EF-3.0' from the other HS1HT

jeremybreeschoten, 2021-03-26:
@Multipage 
Here is the output now.

`Zigbee2MQTT:info  2021-03-26 14:19:56: Device '0xbc33acfffe058c7b' joined
Zigbee2MQTT:info  2021-03-26 14:19:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-03-26 14:19:56: Starting interview of '0xbc33acfffe058c7b'
Zigbee2MQTT:info  2021-03-26 14:19:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-03-26 14:19:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xbc33acfffe058c7b"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-03-26 14:19:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xbc33acfffe058c7b"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-03-26 14:20:10: Device '0xbc33acfffe058c7b' announced itself
Zigbee2MQTT:info  2021-03-26 14:20:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-03-26 14:20:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xbc33acfffe058c7b"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-03-26 14:20:23: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"modelId":"HT-EF-3.0"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:26: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"HEIMAN"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:29: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:32: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:35: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"appVersion":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:38: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:41: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"hwVersion":16}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:44: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2019.9.6"}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2021-03-26 14:20:44: Failed to interview '0xbc33acfffe058c7b', device has not successfully been paired
Zigbee2MQTT:info  2021-03-26 14:20:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-03-26 14:20:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xbc33acfffe058c7b"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-03-26 14:20:47: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-03-26 14:20:51: Received Zigbee message from '0xbc33acfffe058c7b', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2021-03-26 14:20:51: Successfully interviewed '0xbc33acfffe058c7b', device has successfully been paired
Zigbee2MQTT:info  2021-03-26 14:20:51: Device '0xbc33acfffe058c7b' is supported, identified as: HEIMAN Smart temperature & humidity Sensor (HS1HT-N)
Zigbee2MQTT:info  2021-03-26 14:20:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart temperature & humidity Sensor","exposes":[{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HS1HT-N","supports_ota":false,"vendor":"HEIMAN"},"friendly_name":"0xbc33acfffe058c7b","ieee_address":"0xbc33acfffe058c7b","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-03-26 14:20:51: Configuring '0xbc33acfffe058c7b'
Zigbee2MQTT:info  2021-03-26 14:20:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart temperature & humidity Sensor","friendly_name":"0xbc33acfffe058c7b","model":"HS1HT-N","supported":true,"vendor":"HEIMAN"},"type":"pairing"}'
[... cut ...]
Zigbee2MQTT:info  2021-03-26 14:21:24: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe058c7b', payload '{"battery":92.5,"linkquality":131,"temperature":22.82}'
Zigbee2MQTT:debug 2021-03-26 14:21:24: Received Zigbee message from '0xbc33acfffe058c7b', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":4703}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2021-03-26 14:21:24: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe058c7b', payload '{"battery":92.5,"humidity":47.03,"linkquality":131,"temperature":22.82}'
`

Multipage, 2021-03-26:
'Successfully interviewed '0xbc33acfffe058c7b'

And no errors. So..the question is ...is it working :) It seems so
'{"battery":92.5,"humidity":47.03,"linkquality":131,"temperature":22.82}'

jeremybreeschoten, 2021-03-26:
it looks like yes. Indeed. The temperature can be correct and the battery also. The linkquality also. Cause it is 1 mt away from the zigbee stick.

How to proceed? Can you make that further happen? Or what to do. How can i contact you outside of this?

Multipage, 2021-03-26:
I made the changes in the file and created a PR. If all goes OK it will make it to the final that will be released in april.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2417

So from the next release you will be able to use it in your 'production' enviroment.

Multipage, 2021-03-28:
It's merged in the dev branch now so it definitely will be available for you in the next release.

jeremybreeschoten, 2021-03-28:
Ahh cool. But is the dev branche not reachable by Zigbee2mqtt Edge?

Multipage, 2021-03-29:
Sure you can but why would you do that? It's only a couple of days away from being released.

jeremybreeschoten, 2021-03-29:
Okay. But with that release. Do I need to reflash my usb device? Or is this
something for the addin for HA

Multipage, 2021-03-29:
No, if it's released you just update your Zigbee2MQTT to 1.19.1( i presume)  and pair your device. 

That's all.

Multipage, 2021-03-29:
I'm sorry ..missed the part of HA add-in . You'll have to wait for the addin to get updatet.

closed 2021-05-07 by github-actions[bot]