Koenkk/zigbee2mqtt #6713: BlitzWolf (Tuya  TS0121_plug) sockets problem

opirob64, 2021-03-14:
with the update to the latest version, my BlitzWolf sockets  (identify Tuya 
TS0121_plug) no longer detect electricity consumption. Any suggestions?

Multipage, 2021-03-15:
What do you mean with 'latest version'? Dev or master?
With version 1.18.1-1 ( master) i have no problem with exactly the same Blitzwolf sockets

ASNNetworks, 2021-03-15:
Running 1.18.1-2 and no issues here with the same socket. I have 4 of these (EU model), all working without issues. 

Also had no issues on 1.18.1-1.

m0wlheld, 2021-03-16:
Also, no issues using Blitzwolf SHP-13, Z2M 1.18.1 and Home Assistant 2021.3.4.

Mr-Groch, 2021-03-17:
I'm also have problems since update to 1.18.1 (from 1.17-dev). I'm seeing that `power `and `current `are always 0. `voltage `and `energy `are working as before.

m0wlheld, 2021-03-17:
[image]

Zigbee2MQTT 1.18.1, HA 2021.3.4, Blitzwolf SHP-13 via CC2652RB coordinator device.

ASNNetworks, 2021-03-17:
If you're having issues, you should post all information of the specific device. It's possible there are different model numbers and manufacturers that Blitzwolf uses (happens alot with Tuya stuff). Otherwise it will be impossible to solve this, and chances are it will break the devices for people that currently have no issues (like other users en myself).

Mr-Groch, 2021-03-17:
I have 4 of them, and all have the same problem - only `energy` is updating. That's why I haven't noticed it before (HA utility_meters were working). Just when I saw this issue, I've checked it...

I'm sure that about week ago it was working (I have been installing one of them in backyard and checked then all measurements.

What changed in my configuration?

- I've updated addon to 1.18.1-2
- I've added to my zigbee network 2 IKEA Tradfri remotes (and noticed that new group then was created - `default_bind_group (#901)`

Maybe that Ikea remotes are responsible for that... I saw here similar issue...

I've tried to remove one of Blitzwolf SHP-13 from my newtork, then turn it off from power, allow join, and repaired it - doesn;t help... Still no `power` and `current`... `voltage` also disapeared after repairing... Only `energy` working... Lucky - that is what I need most :)

This is my env:

Zigbee2MQTT version: 1.18.1 commit: 1.18.1
Coordinator type: zStack3x0 (zzh!)
Coordinator revision: 20210120

All Blitzwolf SHP-13 are the same - labeled Blitzwolf SHP-13, and Z2M detecting it as:
```
Device type
Router
Zigbee model
TS0121
Description
10A UK or 16A EU smart plug
Support status
Supported

IEEE address
0x588e81fffeff5d84
Network address
0xEC1D
Vendor
TuYa
Model
TS0121_plug
```

This is device record from database.db:

`{"id":33,"type":"Router","ieeeAddr":"0x588e81fffeff5d84","nwkAddr":60445,"manufId":4098,"manufName":"_TZ3000_g5xawfcq","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TZ3000_g5xawfcq","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00215fff6e","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00215fff6e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1615988372679}`

Mr-Groch, 2021-03-17:
I've just restarted zigbee2mqtt addon (without changing anything!) and electrical measurements are back!

[image]

My last restart was when updating addon to 1.18.1-2 few days ago...

luismalddonado, 2021-03-17:
I have the same problem:
MQTT publish: topic 'zigbee2mqtt/Lavadora', payload '{"**current":0**,"linkquality":55,"**power":0,**"state":"ON","voltage":237.54}'
Info MQTT publish: topic 'zigbee2mqtt/Secadora', payload '{"**current":0**,"linkquality":31,"**power":0**,"state":"ON","voltage":236}'

Voltage is fine, but power and current disappeared.  

My device:
https://www.zigbee2mqtt.io/devices/HS2SK.html#heiman-hs2sk

I am in the master branch:
pi@raspberrypi:/opt/zigbee2mqtt $ git branch
* master
pi@raspberrypi:/opt/zigbee2mqtt $ git pull
Already up to date.

ASNNetworks, 2021-03-18:
> I have the same problem:
> MQTT publish: topic 'zigbee2mqtt/Lavadora', payload '{"**current":0**,"linkquality":55,"**power":0,**"state":"ON","voltage":237.54}'
> Info MQTT publish: topic [...]

This is a completely different device. You should create a new ticket for that one.

GSzabados, 2021-03-19:
@m0wlheld, @ASNNetworks  could you please try to disable the LED on the adapter from the Frontend. (Settings/Serial/Check box checked at the bottom/Submit) and then restart Z2M from the Frontend. (Or enable the Last seen option from the Frontend and restart Z2M from the Frontend.)

That's the two things what I have done and those values stayed on the last reading, which was 0 and hasn't updated since.

But when I restart Z2M from Supervisor, as the Frontend Settings are not persistent, the LED comes back on and the Last seen option is not checked anymore.

My guess, the restart from the Frontend causes the values not to update.

Just give it a try to see can you reproduce the issue.

(I am using a CC2531 adapter.)

Mr-Groch, 2021-03-19:
> @m0wlheld, @ASNNetworks could you please try to disable the LED on the adapter from the Frontend. (Settings/Serial/Check box checked at the bottom/Submit) and then restart Z2M from the Frontend. (Or enable the Last seen option from the Frontend and restart Z2M from the Frontend.)
> 
> That's the two things what I have done and those values stayed on the last reading, which was 0 and hasn't updated since.
> 
> But when I restart Z2M from Supervisor, as the Frontend Settings are not persistent, the LED comes back on and the Last seen option is not checked anymore.
> 
> My guess, the restart from the Frontend causes the values not to update.
> 
> Just give it a try to see can you reproduce the issue.
> 
> (I am using a CC2531 adapter and having the same issue.)

Yes, I can confirm that electrical measurements arre gone after restarting Z2M from frontend, and the're back after restarting whole adddon.

GSzabados, 2021-03-19:
Then can someone advise how this issue can be tracked down further? What difference does the restart from Frontend makes? And how to pinpoint it to this issue...

luismalddonado, 2021-03-20:
> > I have the same problem:
> > MQTT publish: topic 'zigbee2mqtt/Lavadora', payload '{"**current":0**,"linkquality":55,"**power":0,**"state":"ON","voltage":237.54}'
> > Info MQTT publish: topic 'zigbee2mqtt/Secadora', payload '{"**current":0**,"linkquality":31,"**power":0**,"state":"ON","voltage":236}'
> > Voltage is fine, but power and current disappeared.
> > My device:
> > https://www.zigbee2mqtt.io/devices/HS2SK.html#heiman-hs2sk
> > I am in the master branch:
> > pi@raspberrypi:/opt/zigbee2mqtt $ git branch
> > 
> > * master
> >   pi@raspberrypi:/opt/zigbee2mqtt $ git pull
> >   Already up to date.
> 
> This is a completely different device. You should create a new ticket for that one. Chances are your problem will otherwise be burried and not solved soon.

Thanks. I fixed the problem rejoining the device.

GSzabados, 2021-04-08:
I've just updated to 1.18.2 and if I restart Z2M from the frontend the plug stops reporting voltage, current and power, so the issue still exists. If the whole addon is restarted from the Supervisor menu, it starts to work again.

@Koenkk, can you give some hint how to track down this issue?

It feels like that it starts to use some different Herdman or the binding is broken. But I am just guessing. 

@m0wlheld, @ASNNetworks have you tried to reproduce the issue on your setup?

Koenkk, 2021-04-08:
@GSzabados can you check if this is fixed in the latest dev branch?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

GSzabados, 2021-04-08:
> @GSzabados can you check if this is fixed in the latest dev branch?
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Yes, I will do, but probably only tomorrow morning, tonight I am a bit busy (cooking and helping my wife to put the children to the bed), and the plug is next to the bedroom with a fan heater connected to it, so it is a bit noisy.

How many devices were affected? I've just started to look at the herdsman's details in the device.js. My first thought was that it for some reason not handling the onEvent correctly.

GSzabados, 2021-04-09:
> @GSzabados can you check if this is fixed in the latest dev branch?
> 
@Koenkk

I've just updated to the edge version, and have confirmed that it was using the latest commit. 

With the change what you have implemented after restarting Zigbee2MQTT from the frontend the device reports correctly voltage, current and power as expected. 

And I was happy to see as well that the typo in Settings(specific), "Get possible device specific praemeters" has been fixed as well to "parameters".

I think this issue can be closed then.

closed 2021-05-17 by github-actions[bot]

faeton, 2021-12-09:
Doesn't seem to be fixed ~ same topic developing here https://github.com/Koenkk/zigbee2mqtt/issues/7272