Koenkk/zigbee2mqtt #6714: Lidl smart plug HG06337 not detected

marcgarciamarti, 2021-03-14:
## What happened
I have a very stable network holding some 30 devices by xiaomi, osram, ikea. I've purchased 3 lidl silvercrest HG06337 wall plug but they won't be recognized/noticed.

## What did you expect to happen
I would have expected to be able to pair and use this model HG06337

## How to reproduce it (minimal and precise)
Attempt to pair and use HG06337. Zigbee2mqtt won't see or detect any of these new devices. Not a single message is logged when initiating pairing on the device.

## Debug info
Zigbee2MQTT version: 1.18.1 commit: 1.18.1
Adapter hardware: CC2531
Adapter firmware version: 20190619

Gis70, 2021-04-02:
Hello, there are maybe only compatible with zigbee 3.0 and cc2531 is 1.2.

I just paired one on a cc2562 with zstack3.0, i had to try several time to have a good pairing.

marcgarciamarti, 2021-04-02:
Wow! I think you've hit the nail right on its head, to my profound despair... and I've just browsed quickly the internet, and it looks like I would need to repair all my devices if I want to migrate my setup to zigbee 3.0. This hurts.

@Koenkk do you see any room for eventually managing to transparently migrate 1.2 setups to 3.0? I could wait even months if needed because some of my current devices sit very far apart from my coordinator and repairing them feels daunting...

Gis70, 2021-04-02:
It's not possible. It's a hardware "cause". CC2531 is said has not powerfull enough to support zstack3.0 as coordinator if I'm not wrong.

marcgarciamarti, 2021-04-03:
@Gis70 sure, I don't mind purchasing a new dongle. What I do mind about it repairing all my devices because some of them are temperature sensors for example, and they are stuck to the walls; bringing them close to the coordinator is not easy at all. 

My main question above was whether Koenkk sees some possibility, even in the long term, to facilitate a migration path to 3.0 without the need of repairing all the existing devices. I don't mind waiting months if that is needed.

Gis70, 2021-04-03:
You don't need to put them close to the coordinator. For example mine is on
a server in my garage, and 2 floors above i was able to pair windows
sensor, temperature sensor...

I suggest you to buy a new more powerful dongle compatible z3.0 and try
with just a few device to not break all you network and configuration.

marcgarciamarti, 2021-04-03:
Ok. I will give this a thought, although I would appreciate @Koenkk 's view on whether there is room for hope here in the long term, prior to jumping the gun.

@Gis70 what dongle do you use? Any of the recommended ones on this list? https://www.zigbee2mqtt.io/information/supported_adapters.html 

If so, where did you get yours?

Thanks

Gis70, 2021-04-03:
I have a slaesh dongle. CC2652RB. Could takes time to buy one and receive it. So be patient. But you can find similar chipset on aliexpress maybe.

Koenkk, 2021-04-04:
Migrating from 1.2 to 3.0 (including a different dongle) MIGHT work after https://github.com/Koenkk/zigbee-herdsman/pull/303 (no ETA)

castorw, 2021-04-04:
> Migrating from 1.2 to 3.0 (including a different dongle) MIGHT work after [Koenkk/zigbee-herdsman#303](https://github.com/Koenkk/zigbee-herdsman/pull/303) (no ETA)

Yes, migration from 1.2 to 3.0 should work fine after the PR, which is now ready for review.

marcgarciamarti, 2021-04-04:
@Koenkk @castorw thank you so much for stepping by and for those good news. I'm sooo happy to see I may be able to avoid repairing all my devices maybe... I will keep my fingers crossed and hope that you guys eventually make it happen!!!

marcgarciamarti, 2021-04-11:
@castorw @Koenkk I'm not knowledgeable at all with GitHub. What indicator should I track if I want to know when the above work has been incorporated in the productive assets? As a user of z2m I see new updates to the add-on but this is part of the device firmware and not the add-on if my understanding is right. 
I see new updates are being posted in the PR and I assume those have to do with the testing and validation process, but I'm really not sure why to track. 
Additionally, I would also like to know what you guys suggest we test this new versions. I have a backup dongle ready... just trying to be ready when the time comes to test out this!! I can't wait!!
Thanks a million to the both of you!

castorw, 2021-04-11:
Me an @Koenkk are currently in the process of final testing, debugging and polishing. To my knowledge - as Koen told me - the soonest this may get in stable release is beginning of May. If you wish to test this yourself you could pull https://github.com/castorw/zigbee-herdsman/tree/znp-provisioning-refactor and use `npm link` to link it to your Z2M instance (you can google procedures to do that). However I would not encourage you to do so, since it's not ready yet. I would wait at least until the PR is merged in `zigbee-herdsman` master.

Koenkk, 2021-05-13:
https://github.com/Koenkk/zigbee-herdsman/pull/303 has been merged now, but restoring a 1.2 -> 3.0 network is not possible yet.

marcgarciamarti, 2021-05-16:
> FYI; https://github.com/Koenkk/zigbee-herdsman/pull/303 has been merged now, but restoring a 1.2 -> 3.0 network is not possible yet.

Thanks for keeping this thread up to date @Koenkk !! I'm keeping my fingers crossed that may work at some point!

pawol, 2021-05-19:
I confir that HG06337 works perfectly on z2m 1.18.3 and the latest zstack 12 (202011)

marcgarciamarti, 2021-06-20:
> > FYI; https://github.com/Koenkk/zigbee-herdsman/pull/303 has been merged now, but restoring a 1.2 -> 3.0 network is not possible yet.
> 
> Thanks for keeping this thread up to date @Koenkk !! I'm keeping my fingers crossed that may work at some point!

Hi @Koenkk ! In your opinion, do you see the 1.2 to 3.0 networker migration happening anytime soon?

Thanks!!

Koenkk, 2021-06-20:
@marcgarciamarti , @castorw do you still pan to work on this?

closed 2021-06-28 by github-actions[bot]

marcgarciamarti, 2021-07-18:
@castorw I would love to read your plans going forward. Do you plan to do this?

castorw, 2021-07-22:
@marcgarciamarti I will be working on this. Unfortunately I am a bit short on ￼time. But I think I may get to this pretty soon. Wanna finish up what I have started.

marcgarciamarti, 2021-07-22:
> @marcgarciamarti I will be working on this. Unfortunately I am a bit short on ￼time. But I think I may get to this pretty soon. Wanna finish up what I have started.

@castorw I'm so much looking forward to this... I'm pretty useless when it comes to coding, but please if there is anything I can help with, drop me a line! Thank you so much!!

castorw, 2021-07-28:
@marcgarciamarti The PR is in (https://github.com/Koenkk/zigbee-herdsman/pull/395).

marcgarciamarti, 2021-07-29:
> @marcgarciamarti The PR is in (https://github.com/Koenkk/zigbee-herdsman/pull/395).

@castorw that was quick! Thanks you so much!
If it is already the time and you or somebody else could kindly point me to any guidance or step-by-step, I'm willing test this on my own environment. My only constraint is that I will be on holidays starting on Saturday, and testing this when you are not at home can be very tricky...