Koenkk/zigbee2mqtt #6718: Groups handling error

jd1900, 2021-03-15:
## What happened
I've tried to change just the color_temp of a lights' group. However, all of the other ligths became "ON" on the eyes of Z2M and HA, but that switch on is not real. So they appear to be "ON" while being "OFF".
It is fine to reamain "OFF", as I didn't change the "ON/OFF" status, but the state should remains as it was. I only want to change it colorTemp to be ready for the time I turn it ON.
It happen as well with "optimistic" equalt to false. But I think it sould never happen regardless of the optimistic value.

## What did you expect to happen
Only change color_temp and not the state.

## How to reproduce it (minimal and precise)
Create a group with one or more devices.
Keep at least one off, or all of them off.
Send a message to:
with this payload or similar { "color_temp": 455 }
See all the lights becoming alledgelly "on" in the state but not on reality.

## Debug info
Zigbee2MQTT version: 1.18.1-2
Adapter hardware: zzh

Koenkk, 2021-03-16:
I couldn't manage to reproduce this. Could you provide the debug logs? 
See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging

Mine:

```
# Turned the bulb OFF
Zigbee2MQTT:debug 2021-03-16 20:10:10: Received MQTT message on 'zigbee2mqtt/my_bulb/set' with data '{"state": "OFF"}'
Zigbee2MQTT:debug 2021-03-16 20:10:10: Publishing 'set' 'state' to 'my_bulb'
Zigbee2MQTT:info  2021-03-16 20:10:10: MQTT publish: topic 'zigbee2mqtt/my_bulb', payload '{"color":{"x":0.415211980162654,"y":0.395434886759171},"color_temp":300,"linkquality":126,"state":"OFF"}'

# Set color temp for bulb, bulb stays off
Zigbee2MQTT:debug 2021-03-16 20:10:21: Received MQTT message on 'zigbee2mqtt/my_bulb/set' with data '{"color_temp": 300}'
Zigbee2MQTT:debug 2021-03-16 20:10:21: Publishing 'set' 'color_temp' to 'my_bulb'

# Set color temp for group, bulb stays off, also STATE stays OFF
Zigbee2MQTT:debug 2021-03-16 20:10:31: Received MQTT message on 'zigbee2mqtt/my_group/set' with data '{"color_temp": 300}'
Zigbee2MQTT:debug 2021-03-16 20:10:31: Publishing 'set' 'color_temp' to 'my_group'
Zigbee2MQTT:info  2021-03-16 20:10:32: MQTT publish: topic 'zigbee2mqtt/my_bulb', payload '{"color":{"x":0.415211980162654,"y":0.395434886759171},"color_temp":300,"linkquality":126,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 20:10:32: MQTT publish: topic 'zigbee2mqtt/my_group', payload '{"color":{"x":0.415211980162654,"y":0.395434886759171},"color_temp":300,"state":"OFF"}'
```

jd1900, 2021-03-16:
The behaviour similar to the described here: https://www.zigbee2mqtt.io/information/groups.html#state-changes

I've created a new group, with 2 lights:

- Group name: **z2mtestgroup**
- Light 1: LED Driver RGB+CCT. Named: **LED_01_test**
- Light 2: LED Driver RGB+CCT. Named: **LED_02_test**

```
# Turned the LED_01_test OFF
Zigbee2MQTT:debug 2021-03-16 22:19:36: Received MQTT message on 'zigbee2mqtt/LED_01_test/set' with data '{"state":"OFF"}'
Zigbee2MQTT:debug 2021-03-16 22:19:36: Publishing 'set' 'state' to 'LED_01_test'
(more here)
```

```
# Set color temp for LED to 333, LED stays off
Zigbee2MQTT:debug 2021-03-16 22:25:24: Received MQTT message on 'zigbee2mqtt/LED_01_test/set' with data '{"color_temp":333}'
Zigbee2MQTT:debug 2021-03-16 22:25:24: Publishing 'set' 'color_temp' to 'LED_01_test'
(more here)
```

```
# Set color temp (444) for group, it stays off as well when both LEDs are off. **THIS IS FINE**
Zigbee2MQTT:debug 2021-03-16 22:52:54: Received MQTT message on 'zigbee2mqtt/z2mtestgroup/set/color_temp' with data '444'
Zigbee2MQTT:debug 2021-03-16 22:52:54: Publishing 'set' 'color_temp' to 'z2mtestgroup'
Zigbee2MQTT:info  2021-03-16 22:52:54: MQTT publish: topic 'zigbee2mqtt/LED_01_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"color_temp_startup":65535,"last_seen":"2021-03-16T21:52:48.471Z","linkquality":144,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 22:52:54: MQTT publish: topic 'zigbee2mqtt/z2mtestgroup', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 22:52:54: MQTT publish: topic 'zigbee2mqtt/LED_02_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"last_seen":"2021-03-16T21:52:48.490Z","linkquality":51,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 22:52:54: MQTT publish: topic 'zigbee2mqtt/LED_01_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 22:52:54: MQTT publish: topic 'zigbee2mqtt/LED_02_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"state":"OFF"}'
Zigbee2MQTT:debug 2021-03-16 22:52:55: Received Zigbee message from 'LED_01_test', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":444}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-03-16 22:52:55: MQTT publish: topic 'zigbee2mqtt/LED_01_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 22:52:55: MQTT publish: topic 'zigbee2mqtt/LED_01_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"color_temp_startup":65535,"last_seen":"2021-03-16T21:52:55.531Z","linkquality":141,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 22:52:55: MQTT publish: topic 'zigbee2mqtt/z2mtestgroup', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"state":"OFF"}'
Zigbee2MQTT:debug 2021-03-16 22:52:55: Received Zigbee message from 'LED_02_test', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":444}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-03-16 22:52:55: MQTT publish: topic 'zigbee2mqtt/LED_02_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"state":"OFF"}'
Zigbee2MQTT:info  2021-03-16 22:52:55: MQTT publish: topic 'zigbee2mqtt/LED_02_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.500605373718514,"y":0.415275958087495},"color_temp":444,"last_seen":"2021-03-16T21:52:55.554Z","linkquality":51,"state":"OFF"}'
```

However, if I previously turn on the other LED (LED_02_test), things change:
```
# Set color temp (222) for group, LED_01_test now turns on. **THIS IS NOT FINE**
Zigbee2MQTT:debug 2021-03-16 22:57:16: Received MQTT message on 'zigbee2mqtt/z2mtestgroup/set/color_temp' with data '222'
Zigbee2MQTT:debug 2021-03-16 22:57:16: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2021-03-16 22:57:16: Publishing 'set' 'color_temp' to 'z2mtestgroup'
Zigbee2MQTT:info  2021-03-16 22:57:16: MQTT publish: topic 'zigbee2mqtt/LED_01_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.360786471097048,"y":0.363543544699483},"color_temp":222,"color_temp_startup":65535,"last_seen":"2021-03-16T21:54:22.490Z","linkquality":144,"state":"ON"}'
Zigbee2MQTT:info  2021-03-16 22:57:16: MQTT publish: topic 'zigbee2mqtt/z2mtestgroup', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.360786471097048,"y":0.363543544699483},"color_temp":222,"state":"ON"}'
Zigbee2MQTT:info  2021-03-16 22:57:16: MQTT publish: topic 'zigbee2mqtt/LED_02_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.360786471097048,"y":0.363543544699483},"color_temp":222,"last_seen":"2021-03-16T21:56:09.828Z","linkquality":99,"state":"ON"}'
[... cut ...]
Zigbee2MQTT:debug 2021-03-16 22:57:17: Received Zigbee message from 'LED_02_test', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":222}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-03-16 22:57:17: MQTT publish: topic 'zigbee2mqtt/LED_02_test', payload '{"brightness":1,"color":{"hue":0,"saturation":100,"x":0.360786471097048,"y":0.363543544699483},"color_temp":222,"last_seen":"2021-03-16T21:57:17.100Z","linkquality":96,"state":"ON"}'

```

I then changed to `optimistic: false` to change the behaviour.
However, then, even if both bulbs are turned off, they all turn on with every color_temp change.
I can provice logs as well if needed for this.

Koenkk, 2021-03-17:
I see the issue know, can you check if it is fixed?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

jd1900, 2021-03-18:
> I see the issue know, can you check if it is fixed?
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

I'm using https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/ which doesn't have that commit (https://github.com/Koenkk/zigbee2mqtt/commit/1acccc002a623f5251d7f51cb2828fb3fe7d26d3).

If I find enough time today I'll try to add that commit manually and try.

Koenkk, 2021-03-18:
@jd1900 you should switch to the edge version of the addon.

jd1900, 2021-03-20:
I mean I don't have this repo:
https://github.com/Koenkk/zigbee2mqtt/

But this:
https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/ which doesn't have that commit (1acccc0).

So that edge version should be different, right?

Koenkk, 2021-03-20:
That repo uses zigbee2mqtt, the edge version of that addon is equal to the zigbee2mqtt dev branch.

jd1900, 2021-03-20:
It works without any problem in the edge repo!
Thanks!

(I close the issue, reopen if needed)

closed 2021-03-20 by jd1900