Koenkk/zigbee2mqtt #6725: AduroSmart 81809/81813 always report Failed to setup reporting during startup

groenmarsmannetje, 2021-03-15:
## What happened
When I start zigbee2mqtt the log shows this error for all 3 AduroSmart 81809/81813 bulbs (actual brand is Trust).

`Zigbee2MQTT:error 2021-03-15 16:33:20: Failed to setup reporting for '0x00158d00011d7de1' - Error: ConfigureReporting 0x00158d00011d7de1/2 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 7745 - 2 - 8 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)`

## What did you expect to happen
No error during startup

## How to reproduce it (minimal and precise)
n/a

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: zzh
Adapter firmware version: 20201026

Koenkk, 2021-03-16:
The report feature is considered deprecated: https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.17.1

closed 2021-03-16 by Koenkk

groenmarsmannetje, 2021-03-17:
Okay I have removed the deprecated report configuration. But now I have had multiple devices that become unavailable. Actually I do not understand with what is meant with “rebind your bound devices” in the migration instruction. Is a power cycle sufficient or should I remove and pair each device again? It is not clear. Also manually setting up reporting per device requires some kind of documentation because I have no clue how that works.

Koenkk, 2021-03-17:
@groenmarsmannetje what is the reason you set up the reporting?