Koenkk/zigbee2mqtt #6728: Openwrt Support

ANARHIST1984, 2021-03-15:
I try to install Zigbee2Mqtt to Openwrt router.
openwrt_19_07_7_ramips_mt7621_zbtlink_zbt_we3526
I manage to compile from here https://github.com/nxhack/openwrt-node-packages/tree/dev_19.07
node_v12.20.0, node-npm_v12.20.0, node-zigbee2mqtt_1.17.1.
And just in case node-bcrypt_5.0.1, node-serialport_9.0.6, node-serialport-bindings_9.0.7, node-serialport-list_9.0.4.
also install driver kmod-usb-acm and i can see in system /dev/ttyACM0

when i try to start "npm start" i receive this
root@OpenWrt:/opt/zigbee2mqtt# npm start
npm info it worked if it ends with ok
npm info using npm@6.14.8
npm info using node@v14.15.1
npm info lifecycle zigbee2mqtt@1.16.2prestart: zigbee2mqtt@1.16.2
npm info lifecycle zigbee2mqtt@1.16.2start: zigbee2mqtt@1.16.2

zigbee2mqtt@1.16.2 start /opt/zigbee2mqtt
node index.js

Zigbee2MQTT:info 2021-03-14 22:43:56: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-03-14.22-43-49' filename: log.txt
Zigbee2MQTT:info 2021-03-14 22:43:57: Starting Zigbee2MQTT version 1.16.2 (commit #unknown)
Zigbee2MQTT:info 2021-03-14 22:43:57: Starting zigbee-herdsman (0.13.37)
Zigbee2MQTT:error 2021-03-14 22:43:57: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-03-14 22:43:57: Failed to start zigbee
Zigbee2MQTT:error 2021-03-14 22:43:57: Exiting...
Zigbee2MQTT:error 2021-03-14 22:43:57: Error: Error relocating /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node: _ZN2v82V88MakeWeakEPPNS_8internal6ObjectEPvPFvRKNS_16WeakCallbackInfoIvEEENS_16WeakCallbackTypeE: symbol not found
at Object.Module._extensions..node (internal/modules/cjs/loader.js:1122:18)
at Module.load (internal/modules/cjs/loader.js:928:32)
at Function.Module._load (internal/modules/cjs/loader.js:769:14)
at Module.require (internal/modules/cjs/loader.js:952:19)
at require (internal/modules/cjs/helpers.js:88:18)
at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
at Object. (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
at Module._compile (internal/modules/cjs/loader.js:1063:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
at Module.load (internal/modules/cjs/loader.js:928:32)
npm info lifecycle zigbee2mqtt@1.16.2~start: Failed to exec start script
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.16.2 start: node index.js
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.16.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm timing npm Completed in 20808ms

npm ERR! A complete log of this run can be found in:
npm ERR! /root/.npm/_logs/2021-03-14T22_43_58_259Z-debug.log

Stik CC2531
root@OpenWrt:/opt/zigbee2mqtt# ls -la /dev | grep ttyACM0
crw-rw----    1 root     dialout   166,   0 Mar 15 17:16 ttyACM0

configuration.yaml
Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

what am I missing or doing wrong?

sjorge, 2021-03-15:
Looks like serialport's binary part is not compiled corrected:
```
Zigbee2MQTT:error 2021-03-14 22:43:57: Error: Error relocating /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node: _ZN2v82V88MakeWeakEPPNS_8internal6ObjectEPvPFvRKNS_16WeakCallbackInfoIvEEENS_16WeakCallbackTypeE: symbol not found
```

Indiates it can't find a symbol it needs.

ANARHIST1984, 2021-03-15:
> Looks like serialport's binary part is not compiled corrected:
>
> ```
> Zigbee2MQTT:error 2021-03-14 22:43:57: Error: Error relocating [...]

I compile this in ubuntu 20. Also, use package compile another people. nothing 
But, this works in xiaomi router 3G. Same CPU but memory only 256.

ANARHIST1984, 2021-03-15:
> Looks like serialport's binary part is not compiled corrected:
>
> ```
> Zigbee2MQTT:error 2021-03-14 22:43:57: Error: Error relocating [...]

npm ci --production

../src/serialport_unix.cpp:176:13: note: 'snprintf' output 20 or more bytes (assuming 1043) into a destination of size 1024
     snprintf(data->errorString, sizeof(data->errorString), "Error %s Cannot open %s", strerror(errno), data->path);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../src/serialport_unix.cpp: In function 'void EIO_Open(uv_work_t*)':
../src/serialport_unix.cpp:79:6: warning: '%s' directive output may be truncated writing up to 1023 bytes into a region of size 1003 [-Wformat-truncation=]
 void EIO_Open(uv_work_t* req) {
      ^~~~~~~~
../src/serialport_unix.cpp:86:13: note: 'snprintf' output 22 or more bytes (assuming 1045) into a destination of size 1024
     snprintf(data->errorString, sizeof(data->errorString), "Error: %s, cannot open %s", strerror(errno), data->path);
     ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
/usr/bin/ld: cannot find -lpthread
collect2: error: ld returned 1 exit status
make: *** [bindings.target.mk:134: Release/obj.target/bindings.node] Error 1
make: Leaving directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
gyp ERR! System Linux 4.14.221
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v14.15.1
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm info lifecycle @serialport/bindings@9.0.4~install: Failed to exec install script
npm info teardown Done in 0.002s
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.0.4 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the @serialport/bindings@9.0.4 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm timing npm Completed in 519818ms

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-15T21_50_14_832Z-debug.log
root@OpenWrt:/opt/zigbee2mqtt# opkg install make
Package make (4.2.1-4) installed in root is up to date.
root@OpenWrt:/opt/zigbee2mqtt# nano /root/.npm/_logs/2021-03-15T21_50_14_832Z-de
bug.log

nurikk, 2021-03-18:
`/usr/bin/ld: cannot find -lpthread
collect2: error: ld returned 1 exit status`

Ant then the very first link in [google](https://forum.openwrt.org/t/usr-bin-ld-cannot-find-lpthread/18404)

closed 2021-04-25 by github-actions[bot]