Koenkk/zigbee2mqtt #6729: Stable OTA check error

Erelen-Laiquendi, 2021-03-16:
## What happened
Can't check for update of any device in my network.

IKEA E1524/E1810 (1 device)
IKEA E1525/E1745 (2 devices)
IKEA E1603/E1702 (1 device)
IKEA LED1623G12 (6 devices)
Tuya TS0601_thermostat (3 devices)

For any device I receive "Failed to check if update available for '<...>' (Device didn't respond to OTA request)" afret 30 seconds from check start.

All recomendations from FAQ (turn on/off for lights, remove and retrn battery for other) didn't help. Nothing else like z2m restart, stick re-insert, turning light on/off during check - too.

Everything, except OTA, else works ok (network - ~60 devices).

```
debug 2021-03-16 04:00:10: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Balcony Light","transaction":"lqqwb-1"}'
info  2021-03-16 04:00:10: Checking if update available for 'Balcony Light'
info  2021-03-16 04:00:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Balcony Light'","meta":{"device":"Balcony Light","status":"checking_if_available"},"type":"ota_update"}'
info  2021-03-16 04:00:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Balcony Light' (Device didn't respond to OTA request)","meta":{"device":"Balcony Light","status":"check_failed"},"type":"ota_update"}'
info  2021-03-16 04:00:40: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Balcony Light"},"error":"Failed to check if update available for 'Balcony Light' (Device didn't respond to OTA request)","status":"error","transaction":"lqqwb-1"}'
error 2021-03-16 04:00:40: Failed to check if update available for 'Balcony Light' (Device didn't respond to OTA request)
```

## Debug info
Zigbee2MQTT version: 1.18.1 (addon 1.18.1-2)
Adapter hardware: slaesh's CC2652RB
Adapter firmware version: 20210120

Twit123, 2021-03-20:
Perhaps add "ikea_ota_use_test_url: true" to your config (just guessing).

On my IKEA buttons (battery powered), click the update button in the webui and before I click the "are you shore" dialog, I start clicking the button and keep clicking (every 3-5 seconds) until it starts updating. Once the update starts (2 hours) then it seems fine.

Simon

davidstosik, 2021-04-25:
I've had the same problem, even with `ikea_ota_use_test_url: true`.

Erelen-Laiquendi, 2021-04-27:
`ikea_ota_use_test_url: true` - doesn't work.
And I can't check OTA for "Tuya TS0601_thermostat" (not IKEA device) too.

closed 2021-05-05 by github-actions[bot]

KitHubek, 2022-04-08:
hmm. the same problem here with a lot of ikea bulbs and switches

ChrisLane, 2023-03-17:
I have the same issue I think on my ikea bulb that I bought new recently:

Zigbee Model - TRADFRI bulb E27 CWS 806lm
Model - [LED1924G9](https://www.zigbee2mqtt.io/devices/LED1924G9.html#ikea-led1924g9)
Firmware build date - 2019062886LEL08N
Firmware version - 1.0.021

I get the following logs when attempting an OTA update:
```
info 2023-03-17 09:34:33Checking if update available for 'Floor Lamp'
info 2023-03-17 09:34:33MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Floor Lamp'","meta":{"device":"Floor Lamp","status":"checking_if_available"},"type":"ota_update"}'
info 2023-03-17 09:34:59MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Floor Lamp' (Device didn't respond to OTA request)","meta":{"device":"Floor Lamp","status":"check_failed"},"type":"ota_update"}'
info 2023-03-17 09:34:59MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Floor Lamp"},"error":"Failed to check if update available for 'Floor Lamp' (Device didn't respond to OTA request)","status":"error","transaction":"b2oye-1"}'
error 2023-03-17 09:34:59Failed to check if update available for 'Floor Lamp' (Device didn't respond to OTA request)
```
This happens with the ikea test url too.

I have had success updating my lamp bulbs without needing to do anything special.
Lamp bulbs:
Zigbee Model - TRADFRIbulbE14WScandleopal470lm
Model - [LED1949C5](https://www.zigbee2mqtt.io/devices/LED1949C5.html#ikea-led1949c5)
Updated to firmware date - 20211201
Updated to firmware version - 1.1.003

beshrkayali, 2024-03-03:
> Perhaps add "ikea_ota_use_test_url: true" to your config (just guessing).
>
> On my IKEA buttons (battery powered), click the update button in the webui and before I click the "are you shore" dialog, I [...]

This worked for me. In zigbee2mqtt logs, I had `Update of 'IkeaButton' failed (OTA: Device didn't respond to OTA request)` every time I tried to update. Mashing the buttons every few seconds right after you click "Install" in HomeAssistant worked.

Zolric, 2025-01-16:
> > On my IKEA buttons (battery powered), click the update button in the webui and before I click the "are you sure" dialog, I start clicking the button and keep clicking (every 3-5 seconds) until it starts updating. Once the update starts (2 hours) then it seems fine.
> > Simon
> 
> This worked for me. In zigbee2mqtt logs, I had `Update of 'IkeaButton' failed (OTA: Device didn't respond to OTA request)` every time I tried to update. Mashing the buttons every few seconds right after you click "Install" in HomeAssistant worked.

same here - I did like a binding to start the update.

defanator, 2025-03-07:
> Mashing the buttons every few seconds right after you click "Install" in HomeAssistant worked.

Just been able to initiate OTA update for IKEA's Rodret E2201 by simply pushing buttons a few times after initiating update.

Before it always ended up with:
```
[2025-03-07 14:56:15] info:     z2m: Updating 'controls/rodret1' to latest firmware
[2025-03-07 14:56:23] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "62446".
[2025-03-07 14:56:25] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "62446".
[2025-03-07 14:56:26] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "62446".
[2025-03-07 14:56:28] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "62446".
[2025-03-07 14:57:20] info:     z2m:mqtt: MQTT publish: topic 'z2m/bridge/response/device/ota_update/update', payload '{"data":{},"error":"Update of 'controls/rodret1' failed (Device didn't respond to OTA request)","status":"error","transaction":"iwd98-8"}'
[2025-03-07 14:57:20] error:    z2m: Update of 'controls/rodret1' failed (Device didn't respond to OTA request)
```

Still running, but at least it's something. Thanks for mentioning!