Koenkk/zigbee2mqtt #6732: Inappropriate ioctl for device setting custom baud rate of 115200

denvro, 2021-03-16:
## My Zigbee2MQTT stopped working
I tried to update to the latest version. Running on an Debian 10 LXC container.

## What did you expect to happen
- Update went without any problems
- My USB port is visible and i have write access
- Would expect that I could start Zigbee2MQTT without any problems

## Debug info
Platform: Debian 10 LXC container running in Proxmox
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2652

## Logging
``` 
Mar 16 11:01:33 zigbee2mqtt-lxc systemd[1]: Started zigbee2mqtt.
Mar 16 11:01:33 zigbee2mqtt-lxc npm[558]: > zigbee2mqtt@1.18.1 start /opt/zigbee2mqtt
Mar 16 11:01:33 zigbee2mqtt-lxc npm[558]: > node index.js
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: Zigbee2MQTT:info  2021-03-16 11:01:34: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-03-16.11-01-33' filename: log.txt
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: Zigbee2MQTT:info  2021-03-16 11:01:34: Starting Zigbee2MQTT version 1.18.1 (commit #a43d1f3)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: Zigbee2MQTT:info  2021-03-16 11:01:34: Starting zigbee-herdsman (0.13.71)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: Zigbee2MQTT:error 2021-03-16 11:01:34: Error while starting zigbee-herdsman
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: Zigbee2MQTT:error 2021-03-16 11:01:34: Failed to start zigbee
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: Zigbee2MQTT:error 2021-03-16 11:01:34: Exiting...
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: Zigbee2MQTT:error 2021-03-16 11:01:34: Error: Error while opening serialport 'Error: Error: Inappropriate ioctl for device setting custom baud rate of 115200'
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at Generator.next (<anonymous>)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at new Promise (<anonymous>)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]:     at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR! code ELIFECYCLE
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR! errno 1
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR! Exit status 1
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR!
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR! Failed at the zigbee2mqtt@1.18.1 start script.
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR! A complete log of this run can be found in:
Mar 16 11:01:34 zigbee2mqtt-lxc npm[558]: npm ERR!     /root/.npm/_logs/2021-03-16T11_01_34_620Z-debug.log
Mar 16 11:01:34 zigbee2mqtt-lxc systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Mar 16 11:01:34 zigbee2mqtt-lxc systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 16 11:01:34 zigbee2mqtt-lxc systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Mar 16 11:01:34 zigbee2mqtt-lxc systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 4.
Mar 16 11:01:34 zigbee2mqtt-lxc systemd[1]: Stopped zigbee2mqtt.
```

closed 2021-04-23 by github-actions[bot]

AuHau, 2021-06-06:
@denvro did you solve your problem? I have run into the same issue 😳

foskamon, 2021-07-04:
same problem here

xiki808, 2021-08-23:
Same problem here, Running on debian 11 LXC