Koenkk/zigbee2mqtt #6733: ORVIBO - smart sticker switch

risozhor, 2021-03-16:
## What happened
When I'm adding an [ORVIBO smart sticker switch](https://www.zigbee2mqtt.io/devices/CR11S8UZ.html)  it gets added as unsupported device even though it is supported.

[image]

## What did you expect to happen
Can help to create external device converter?
## How to reproduce it (minimal and precise)
Push "permit join all" in Zigbee2MQTT and hold buttons (7 and 8) on the smart sticker for reset it and  join.
## Debug info
Zigbee2MQTT version: 1.18.1-2
Adapter hardware: CC2538
Adapter firmware version: 2020 October
[database.db.zip](https://github.com/Koenkk/zigbee2mqtt/files/6148569/database.db.zip)
Part of my log ...

<pre><code>
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,3,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,11,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,7,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,15,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,3,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,11,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,7,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,15,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,3,0,2],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
Debug Received Zigbee message from '0x00124b000cc8dbf4', type 'raw', cluster '23', data '{"data":[25,0,8,3,0,3],"type":"Buffer"}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model '51725b7bcba945c8a595b325127461e9' and manufacturer name '欧瑞博'
</code></pre>

Koenkk, 2021-03-16:
Looks that you have a slightly different model then the supported one. Try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/1a3b473fac24215965db9d85f4c911b2d9a68da2/devices.js#L14048 to `zigbeeModel: ['3c4e4fc81ed442efaf69353effcdfc5f', '51725b7bcba945c8a595b325127461e9'],`.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to edit `devices.js`.

risozhor, 2021-03-16:
> Looks that you have a slightly different model then the supported one. 

Thanks! Its work for me. Can you add this into next commit?

closed 2021-03-16 by risozhor

Koenkk, 2021-03-17:
Added! 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

SAOPP, 2022-09-01:
@Koenkk btw I got two devices, and they all is not fully interviewed also... So it's not only one-device issue.

Koenkk, 2022-09-05:
@SAOPP can you provide the herdsman debug log of that?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Keeze, 2022-12-15:
So I tried to pair the Orvibo sticker with herdman logs enabled:

```
Zigbee2MQTT:info  2022-12-15 23:45:22: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"zzw9n-29"}'
2022-12-15T22:45:24.979Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 162
2022-12-15T22:45:24.981Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-12-15T22:45:24.981Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-12-15T22:45:28.701Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-12-15T22:45:28.701Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-12-15T22:45:28.704Z zigbee-herdsman:deconz:driver query aps data indication
2022-12-15T22:45:28.708Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 163
2022-12-15T22:45:28.710Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 163 srcAddr: 0xc9ec destAddr: 0x0 profile id: 0x0 cluster id: 0x13 lqi: 191
2022-12-15T22:45:28.710Z zigbee-herdsman:deconz:frameParser response payload: 0,236,201,224,17,220,19,0,75,18,0,128
2022-12-15T22:45:28.710Z zigbee-herdsman:controller:log Device '0x00124b0013dc11e0' joined
2022-12-15T22:45:28.710Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-12-15T22:45:28.710Z zigbee-herdsman:controller:log Device '0x00124b0013dc11e0' accepted by handler
2022-12-15T22:45:28.710Z zigbee-herdsman:controller:log Delete device '0x00124b0013dc11e0' joined, undeleting
Zigbee2MQTT:info  2022-12-15 23:45:28: Device '0x00124b0013dc11e0' joined
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0013dc11e0","ieee_address":"0x00124b0013dc11e0"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0013dc11e0"},"type":"device_connected"}'
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'homeassistant/sensor/0x00124b0013dc11e0/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 action","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'homeassistant/sensor/0x00124b0013dc11e0/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'homeassistant/sensor/0x00124b0013dc11e0/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 last seen","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'zigbee2mqtt/0x00124b0013dc11e0', payload '{"action":null,"last_seen":1671144328827,"linkquality":199}'
2022-12-15T22:45:28.829Z zigbee-herdsman:controller:log Interview '0x00124b0013dc11e0' start
Zigbee2MQTT:info  2022-12-15 23:45:28: Starting interview of '0x00124b0013dc11e0'
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0013dc11e0","ieee_address":"0x00124b0013dc11e0","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-12-15 23:45:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0013dc11e0"},"type":"pairing"}'
2022-12-15T22:45:28.936Z zigbee-herdsman:controller:device:log Interview - start device '0x00124b0013dc11e0'
2022-12-15T22:45:28.936Z zigbee-herdsman:controller:device:log Interview - skip node descriptor request for '0x00124b0013dc11e0', already got it
2022-12-15T22:45:28.936Z zigbee-herdsman:controller:device:log Interview - Detected potential TuYa end device, reading modelID and manufacturerName...
[... cut ...] urerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1285,"dataType":33,"attrData":227}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":50600,"endpoint":1,"linkquality":111,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2022-12-15T22:45:29.699Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

Koenkk, 2022-12-17:
@Keeze  interesting

- Does this also happen when you force remove the device (through z2m frontend), restart z2m and pair the device?
- can you provide the data/database.db entry of this device after trying to pair it?

Keeze, 2022-12-17:
This also happens after I force remove and restart Z2M.

Here is the database entry

`{"id":3,"type":"EndDevice","ieeeAddr":"0x00124b0013dc11e0","nwkAddr":51692,"manufId":0,"manufName":"欧瑞博","modelId":"3c4e4fc81ed442efaf69353effcdfc5f","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"3c4e4fc81ed442efaf69353effcdfc5f","manufacturerName":"欧瑞博"}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1671271089039,"defaultSendRequestWhen":"immediate"}
`

Koenkk, 2022-12-17:
How are you running z2m? (docker/via ha addon?)

Keeze, 2022-12-17:
situation:

situation:
1) I started with Home assistant in docker and Z2M in docker with a CC2531 stick and the device worked
2) due to performance problems I migrated to the ConBee2 stick and everything worked except this device ( I already made an issue some time ago)
- At the moment I changed servers and run home assistant os in a proxmox VM with Z2M as an add-on. and same as situation 2, everything works execpt the Orvibo smart sticker.

Koenkk, 2022-12-18:
I need some more logging to debug this. For this follow https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#testing-changes-locally , instead of `vi app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js` do `vi app/node_modules/zigbee-herdsman/dist/controller/model/device.js` and around line 500 change this:

```js
        debug.log(`Interview - got active endpoints for device '${this.ieeeAddr}'`);
        console.log('ACTIVE EP', activeEndpoints.endpoints);
        console.log('_endpoints', this._endpoints);
        for (const endpointID of activeEndpoints.endpoints) {
            const endpoint = this.getEndpoint(endpointID);
            console.log('EP ', endpointID, endpoint);
            const simpleDescriptor = await entity_1.default.adapter.simpleDescriptor(this.networkAddress, endpoint.ID);
            endpoint.profileID = simpleDescriptor.profileID;
```

Restart z2m as documented in the next step, pair again and provide me the STDOUT logging. (`docker logs CONTAINER_ID`)

SAOPP, 2022-12-19:
@Keeze for me this devices worked with 2531 too, I want to pair it again too 2531 also, and one again to make sure this works on it. But with ZigStar it's didn't works.

Keeze, 2022-12-19:
@Koenkk I finally gained access to through SSH. I am not shure what to edit in the device.js file.....

SAOPP, 2022-12-19:
Tested it with the latest release and cc2531, the same behavior, device is not paired. So it's not stick issue, as I has think before, it's general bug I think.

Koenkk, 2022-12-19:
@Keeze I updated https://github.com/Koenkk/zigbee2mqtt/issues/6733#issuecomment-1356785460 with the lines to update (it is around line number 500)

Keeze, 2022-12-19:
[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/10262591/logs.txt)

Looks like around line 596...

Keeze, 2022-12-19:
@Koenkk  Just one other thing. In the "old" forum there was a thread  with the right sequence for resetting this device. Do you still have acces to this old forum (or do you know the right reset-sequence (rule thing out)..

Koenkk, 2022-12-20:
The logging I added is not logged to the log files, only to stdout (use `docker logs -f CONTAINER_ID`)

Keeze, 2022-12-20:
I did use docker logs and redirected the output to a file..

Koenkk, 2022-12-20:
Are you sure the changes are still in `app/node_modules/zigbee-herdsman/dist/controller/model/device.js`? note that you have to restart via `docker restart CONTAINER_ID`, not via HA itself (then changes will be gone)

Keeze, 2022-12-20:
restarted Z2M in HA..

New logs: 
[logs2.txt](https://github.com/Koenkk/zigbee2mqtt/files/10272951/logs2.txt)

Koenkk, 2022-12-20:
💯  , pushed a fix

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Keeze, 2022-12-20:
I run Z2M as a add-on in home assistant, how can I go from the stable to the dev version (and back)?

SAOPP, 2022-12-21:
@Keeze look at 3th point ova here: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#installation

Keeze, 2022-12-21:
I am aware of the Edge version, but my concerns are current configuration, do I have to remove the current install (stable) and revert when this fix is in the stable version. (edge can't be upgraded automaticly).

Keeze, 2023-01-01:
@Koenkk I just noticed a new version of Z2M in the HA add-ons and installed it. I did not see the Orvibo under fixes? The device seems to be recognized now, but the device does not stop flashing the red led after interviewing and there are no actions registerd in Z2M and HA.

Koenkk, 2023-01-02:
@Keeze I only fixed the interview issues, I was not aware there are more. Can you provide the debug log when triggering actions on the device?

Keeze, 2023-01-02:
I again rejoined the device and this is the log-output. Initialy the sticker keeps flashing. Pairing is just pressing the bottom right buttons at the same time right?

Here is the debug output

```
Info 2023-01-02 09:57:07Device '0x00124b0013dc11e0' joined
Info 2023-01-02 09:57:07MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0013dc11e0","ieee_address":"0x00124b0013dc11e0"},"type":"device_joined"}'
Info 2023-01-02 09:57:07MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0013dc11e0"},"type":"device_connected"}'
Info 2023-01-02 09:57:07MQTT publish: topic 'homeassistant/sensor/0x00124b0013dc11e0/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 action","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info 2023-01-02 09:57:07MQTT publish: topic 'homeassistant/sensor/0x00124b0013dc11e0/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2023-01-02 09:57:07MQTT publish: topic 'homeassistant/sensor/0x00124b0013dc11e0/last_seen/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 last seen","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Info 2023-01-02 09:57:07MQTT publish: topic 'zigbee2mqtt/0x00124b0013dc11e0', payload '{"action":null,"last_seen":1672649828747,"linkquality":191}'
Info 2023-01-02 09:57:07Starting interview of '0x00124b0013dc11e0'
Info 2023-01-02 09:57:07MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0013dc11e0","ieee_address":"0x00124b0013dc11e0","status":"started"},"type":"device_interview"}'
Info 2023-01-02 09:57:07MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0013dc11e0"},"type":"pairing"}'
Debug 2023-01-02 09:57:08Received MQTT message on 'homeassistant/sensor/0x00124b0013dc11e0/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 action","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Debug 2023-01-02 09:57:08Received MQTT message on 'homeassistant/sensor/0x00124b0013dc11e0/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug 2023-01-02 09:57:08Received MQTT message on 'homeassistant/sensor/0x00124b0013dc11e0/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0013dc11e0"],"manufacturer":"ORVIBO","model":"Smart sticker switch (CR11S8UZ)","name":"0x00124b0013dc11e0"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","json_attributes_topic":"zigbee2mqtt/0x00124b0013dc11e0","name":"0x00124b0013dc11e0 last seen","state_topic":"zigbee2mqtt/0x00124b0013dc11e0","unique_id":"0x00124b0013dc11e0_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Debug 2023-01-02 09:57:08Received Zigbee message from '0x00124b0013dc11e0', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"欧瑞博","modelId":"3c4e4fc81ed442efaf69353effcdfc5f"}' from endpoint 1 with groupID null
[... cut ...] value_min":0}],"model":"CR11S8UZ","options":[],"supports_ota":false,"vendor":"ORVIBO"},"friendly_name":"0x00124b0013dc11e0","ieee_address":"0x00124b0013dc11e0","status":"successful","supported":true},"type":"device_interview"}'
Info 2023-01-02 09:57:10MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart sticker switch","friendly_name":"0x00124b0013dc11e0","model":"CR11S8UZ","supported":true,"vendor":"ORVIBO"},"type":"pairing"}'
```

Koenkk, 2023-01-02:
@Keeze I don't know the details of this device (I don't have it)

@risozhor does this device still work in z2m 1.29.0?

SAOPP, 2023-01-02:
@Koenkk works for me homie at 1.29.0

[image]

Keeze, 2023-01-02:
I get the same, but does it also register keypresses?

SAOPP, 2023-01-02:
@Keeze yep, all single clicks, hold and release action is detected, try to replace batterie.

SAOPP, 2023-01-02:
https://user-images.githubusercontent.com/5647009/210259620-e1b5d649-c3aa-4480-b020-4d3ba27949c2.mp4

Keeze, 2023-01-06:
Situation: I seems to pair ok. But no button presses are seen. Also the last seen does not update... but, when I remove the device, without forcing, as soon as I press a button, it leaves the network... Weird behavoir... @Koenkk It is of no use now for me, can I send it to you, just for fun.... (fits a bubble envolope and you are even in the same province)

SAOPP, 2023-01-07:
@Koenkk here is on:

```
Info 2023-01-07 21:07:15MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_3_hold","last_seen":"2023-01-07T21:07:15+02:00","linkquality":178}'
Info 2023-01-07 21:07:15MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:15+02:00","linkquality":178}'
Info 2023-01-07 21:07:15MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_3_hold'
Debug 2023-01-07 21:07:17Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,7,0,3],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:17MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_3_release","last_seen":"2023-01-07T21:07:17+02:00","linkquality":174}'
Info 2023-01-07 21:07:17MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:17+02:00","linkquality":174}'
Info 2023-01-07 21:07:17MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_3_release'
Debug 2023-01-07 21:07:19Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,15,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:19MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_4_click","last_seen":"2023-01-07T21:07:19+02:00","linkquality":178}'
Info 2023-01-07 21:07:19MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:19+02:00","linkquality":178}'
Info 2023-01-07 21:07:19MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_4_click'
Debug 2023-01-07 21:07:20Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,11,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:20MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_2_click","last_seen":"2023-01-07T21:07:20+02:00","linkquality":178}'
Info 2023-01-07 21:07:20MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:20+02:00","linkquality":178}'
Info 2023-01-07 21:07:20MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_2_click'
Debug 2023-01-07 21:07:21Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,15,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:21MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_4_click","last_seen":"2023-01-07T21:07:21+02:00","linkquality":174}'
Info 2023-01-07 21:07:21MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:21+02:00","linkquality":174}'
Info 2023-01-07 21:07:21MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_4_click'
Debug 2023-01-07 21:07:21Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,15,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:21MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_4_click","last_seen":"2023-01-07T21:07:21+02:00","linkquality":171}'
Info 2023-01-07 21:07:21MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:21+02:00","linkquality":171}'
Info 2023-01-07 21:07:21MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_4_click'
Debug 2023-01-07 21:07:22Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,11,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:22MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_2_click","last_seen":"2023-01-07T21:07:22+02:00","linkquality":171}'
Info 2023-01-07 21:07:22MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:22+02:00","linkquality":171}'
Info 2023-01-07 21:07:22MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_2_click'
Debug 2023-01-07 21:07:23Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,7,0,2],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:23MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_3_hold","last_seen":"2023-01-07T21:07:23+02:00","linkquality":171}'
Info 2023-01-07 21:07:23MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:23+02:00","linkquality":171}'
Info 2023-01-07 21:07:23MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_3_hold'
Debug 2023-01-07 21:07:24Received Zigbee message from '0x00124b000cc9ae47', type 'raw', cluster '23', data '{"data":[25,0,8,7,0,3],"type":"Buffer"}' from endpoint 1 with groupID 0
Info 2023-01-07 21:07:24MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"button_3_release","last_seen":"2023-01-07T21:07:24+02:00","linkquality":174}'
Info 2023-01-07 21:07:24MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47', payload '{"action":"","last_seen":"2023-01-07T21:07:24+02:00","linkquality":174}'
Info 2023-01-07 21:07:24MQTT publish: topic 'z2m-zigstar-poe/0x00124b000cc9ae47/action', payload 'button_3_release'
```

Koenkk, 2023-01-08:
@Keeze can you try to bind cluster 23 to the coordinator by sending to `zigbee2mqtt/bridge/request/device/bind` `{"from": "0x00124b0013dc11e0", "to": "coordinator", "clusters": [23]}`. Right before sending this make sure to wakeup the device by pressing a button on it.

Keeze, 2023-01-08:
When I do that, I immediately get
[image]

Koenkk, 2023-01-09:
Could you check if the issue is fixed with the following external converter: 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['3c4e4fc81ed442efaf69353effcdfc5f', '51725b7bcba945c8a595b325127461e9'],
    model: 'CR11S8UZ',
    vendor: 'ORVIBO',
    description: 'Smart sticker switch',
    fromZigbee: [fz.orvibo_raw_1],
    exposes: [e.action(['button_1_click', 'button_1_hold', 'button_1_release', 'button_2_click', 'button_2_hold', 'button_2_release',
        'button_3_click', 'button_3_hold', 'button_3_release', 'button_4_click', 'button_4_hold', 'button_4_release'])],
    toZigbee: [],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await reporting.bind(endpoint1, coordinatorEndpoint, [23]);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- z2m frontend -> device -> reconfigure device with yellow button (make sure it succeeds, you should get a green popup)

Keeze, 2023-01-09:
In first instance it seems to reconfigure, but no luck. I already have an external_converter configured. Do I have to add it?

I tried it with the other external converter removed from configuration.yaml, and tried to remove it from Z2M and rejoin (after restart of Z2M). The device seems to join, but never stops flashing the red light (telling it is functional I presume).

Thanks for all your effort!

Koenkk, 2023-01-09:
Maybe the issue is caused by the Conbee coordinator, can you try updating you conbee firmware to the latest version?

Keeze, 2023-01-09:
Hmm, how can I best update the firmware of my conbee stick, I run HAos in a Proxmox VM....

SAOPP, 2023-01-09:
Last chance, still try on another coordinator, if the problem remains, then this is a device, if not, then it's in the conbee.