Koenkk/zigbee2mqtt #6734: mocked Home Assistant messages cause unwanted recursion/flood

GitOnHub, 2021-03-16:
TL;DR: homeassistant.js:onPublishEntityState infinitely recurses when cache_state is configured as false. This can be circumvented by emiting only a single message _after_ the `for`-loop.

## What happened
Upon updating from 1.16.2 the log started flooding with the following infinitely repeated message pattern:
```
debug [...] 16:28:24: Mocking 'update_available' value for Home Assistant
info [...] 16:28:24: MQTT publish: topic 'zigbee/mysensor', payload '{"linkquality":60,"update_available":false}'
debug [...] 16:28:24: Mocking 'update' value for Home Assistant
info  [...] 16:28:24: MQTT publish: topic 'zigbee/mysensor', payload '{"linkquality":60,"update":{"state":"idle"}}'

```
(date removed for clarity)

Such a flood would be directly following a normal message from that device (a button press, or a periodic checkin from a router)

The spurious MQTT messages quickly choke up Home Assistant on the same host, and cause the whole host to become sluggish.

The mentioned devices are mostly but not exclusively the Tradfri product range, and it looks like many (or all) have OTA support.

## What did you expect to happen
I expected just the single operational message and no spurious messages at all.

## How to reproduce it (minimal and precise)
configuration.yaml:
```
homeassistant: true
cache_state: false
```
**I know, yes, cache_state should be `true` for HA** because it expects all values to be there in every single message. But HA is not my only concern: **I consume the messages with other software as well**, and for that purpose I really need to receive ONLY the updated value instead of all values repeated. Additionally I've noticed that for me, HA runs just fine with `cache_state: false`.

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC26X2R1
Adapter firmware version: CC2652R_coordinator_20210120.hex
Device types: seems to affect any device that supports OTA updates

##Code Analysis
Inspecting the code, the issue seems to be in [homeassistant.js](https://github.com/Koenkk/zigbee2mqtt/blob/a43d1f3917fee5516429bf1bf80419f1d2e0e7f8/lib/extension/homeassistant.js#L506), where each property to be mocked for HA is emitted separately. When configured with `cache_state: true` this would probably result in a single message or at most a double message, since the skipping conditions (.hasOwnProperty) would be satisfied. With `cache_state: false` however, the `update` and `update_available` properties are emitted separately, each triggering another invocation of the `onPublishEntityState` handler that will then emit a spurious message with the converse mocked property (`update` -> `update_available` -> `update`>etc.). This will recurse indefinitely until a crash or until stopped.

## Solution
A solution (/workaround) could be to collect the mocked properties during the `for` loop and emit them immediately afterwards instead of emitting them as we go, still inside the loop. I am not intimately familiar with the code, but this change seems to me to have little impact on the rest of the code.

GitOnHub, 2021-03-16:
After thinking and tinkering some more, I seem to have been able to work around this issue by adding `legacy_api: false` to prevent triggering the involved code path.

Koenkk, 2021-03-16:
I didn't investigate this, but if `legacy_api: false` fixes it I don't think this is worth investigating further (since the legacy api will be removed after some time).

closed 2021-04-17 by Koenkk