Koenkk/zigbee2mqtt #6736: Failed to connect to the adapter

jzeeman01, 2021-03-16:
## What happened
During the Z2M start I get the following error:

Zigbee2MQTT:error 2021-03-16 18:06:58: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE

I have run both the the Zigbee2MQTT addon for Loxberry and running Z2M its own docker container
I have run for versions 1.16.2, 1.17.1 and 1.18.1 on both a Raspbeerypi 3 & 4 also trying both adapters:
CC26X2R1 with ( CC2652R_coordinator_20210120.zip ) and 
CC2531 with ( CC2531_SOURCE_ROUTING_20201128.zip ) 

My std setup working for the last 6 months was the CC26X2R1 on a pi4 - but for the last couple of days its been connecting and working intermittently and since yesterday just stopped - for me there was no direct software update of z2M that seems to be  causing the issue for some

Gibbz, 2021-03-18:
I have this issue commonly, I find pulling out the usb device and pluging in solves the issue for me... Which is a real pain.

jzeeman01, 2021-03-18:
Hey Gibbz
I've tried that - plugging in each different port - rebooting with it in , rebooting with it out :-(
No luck

mamrai1, 2021-03-18:
If you are running z2m in docker remove the --privileged=true

jzeeman01, 2021-03-18:
Thx Mamrai1 - I will try - although I was running with and without Docker...
I have actually gone to a previous version of the PI software I'm using as a base... its picked up the CC2531 but still has an error on the CC26X2R1... I've actually ordered a zzh! although its the same chip - hoping !!! getting a bit desperate now :-(

Ulli2k, 2021-03-19:
I do have the same issue.
After ~24h z2m just says adapter disconnected. Just restarting my z2m does not solve the issue. I must replugin the adapter and it works again.
Maybe an issue of current firmeware and 1.18.1?

Koenkk, 2021-03-19:
@Ulli2k what adapter?

Ulli2k, 2021-03-19:
Firmware zStack3x0  20210218
CC2652P
z2m 1.18.1

Koenkk, 2021-03-19:
Can you provide the herdsman debug log of this? From starting zigbee2mqtt till crash.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

henkiejan1, 2021-03-21:
Hmm, i has a same sort of problem. It was after i installed the latest NPM what suggested after i run ./update.sh:

`
   ╭───────────────────────────────────────────────────────────────╮
   │                                                               │
   │      New major version of npm available! 6.14.11 → 7.6.3      │
   │   Changelog: https://github.com/npm/cli/releases/tag/v7.6.3   │
   │               Run npm install -g npm to update!               │
   │                                                               │
   ╰───────────────────────────────────────────────────────────────╯

`

But if i do the whole system hangs and has a lot of cpu usage. After removing whole nodejs with: `sudo apt remove nodejs` and then manually removed `sudo rm -r /usr/lib/node_modules` and then reinstall nodejs with `sudo apt install nodejs` everythings works like a charm with the old NPM.

Maybe the same problem?

krema, 2021-03-25:
I have the same problem. The coordinator has been broken since the upgrade to 1.18.1. I could reproduce this with another stick as well. Now I have to flash the coordinators again. I will stick to version 1.17.1 for now.

jamess5, 2021-03-26:
Same issue here, sticking with 1.17.1 - can't upgrade

Ulli2k, 2021-03-28:
Its getting worse. Z2m can not be started up again

Zigbee2MQTT:error 2021-03-28 08:14:11: Failed to call 'Bridge' 'onMQTTConnected' (Error: SRSP - ZDO - extNwkInfo after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
Zigbee2MQTT:error 2021-03-28 08:14:17: Publish 'set' 'state' to '0x00158d00042a01db' failed: 'Error: Command 0x00158d00042a01db/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

I did not get the herdsman debug stdout. I run it on docker compose.

mamrai1, 2021-03-28:
Can you please check if you have an ip under bind host address in settings > frontend? I had 0.0.0.0 and sometimes z2m failed to start

Ulli2k, 2021-03-28:
After several reboots and unplugs of the zigbee stick I have it back up.
The bind address was configured as 0.0.0.0 ...I changed it now....lets see if it helps

closed 2021-05-06 by github-actions[bot]