Koenkk/zigbee2mqtt #6737: Device FB56+ZSW1HKJ2.7 not report change state when use it manually

roumano, 2021-03-16:
## What happened
On this hardware ( i got 30 other kind of hardware without any issue )
- Zigbee model : FB56+ZSW1HKJ2.7
- Description : Smart light switch - 2 gang
- Support status : Supported
- IEEE address : 0x00124b001995750c
- Network address : 0x66B8
- Firmware build date : 20190601 
- Vendor : Nue / 3A
- Model : HGZB-042
- Interview completed : Yes

* It's correctly sync and usable from zigbee2mqtt (and from HA) 
* but it's not get update state when use it manually , i don't have any log when do it.
* if i click on the refresh button ( in exposes of this equipment in zigbee2mqtt interface), it's get the new and correct state and get this kind of log
```
mars 16 20:27:40 rasp npm[5919]: Zigbee2MQTT:info  2021-03-16 20:27:40: MQTT publish: topic 'zigbee2mqtt/Cuisine', payload '{"linkquality":31,"state_bottom":"ON","state_top":"OFF"}'
mars 16 20:27:42 rasp npm[5919]: Zigbee2MQTT:info  2021-03-16 20:27:42: MQTT publish: topic 'zigbee2mqtt/Cuisine', payload '{"linkquality":31,"state_bottom":"OFF","state_top":"OFF"}'

```

## What did you expect to happen
Refresh the state when use it manually (via hand behind the switch)

## How to reproduce it (minimal and precise)
move behind the switch
press of the two button (or both), the state is never update in the zigbee2mqqt.

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: Conbee II
Adapter firmware version: deCONZ_ConBeeII_0x26680700.bin.GCF  , deCONZ firmware version 26580700

sjorge, 2021-03-16:
Does the frontend show reporting configure for the device?

roumano, 2021-03-16:
Not sure to understand, you mean this ? : 
[image]

sjorge, 2021-03-16:
Yes! Looks like there is no reporting setup. So if the device changes state without going through z2m (or ha which goes through z2m) it won’t report its state.

@Koenkk i guess we might need to setup the genOnOff reporting for this device with the new reporting system... since the device can be physically changed?

sjorge, 2021-03-17:
So I double checked, https://github.com/Koenkk/zigbee-herdsman-converters/blob/58d0756414530bbcca38644bded7a2e74183b46c/devices.js#L6958 the configuration should set up the reporting.

Can you try reconfiguring the device? 2 entries should show up on that tab if all goes well.

One for endpoint 16 and one for endpoint 16, both for the OnOff cluster.

juslex, 2021-03-17:
Look this. The same issue: HGZB-42 and conbee II
https://github.com/Koenkk/zigbee2mqtt/issues/4461#issuecomment-778833458

Koenkk, 2021-03-17:
Duplicate of #4461 indeed (technical details: https://github.com/Koenkk/zigbee2mqtt/issues/4461#issuecomment-742710987). TLDR; has to be fixed in Conbee firmware.

closed 2021-03-17 by Koenkk