Koenkk/zigbee2mqtt #6738: Adeo lexman e14 bulb

Gwenlad, 2021-03-16:
Hello,
I tried to add new device, but after hours I let it down, it's to complicated for me. 
If it's possible to add this on a next update, it would be great :)

   {
        zigbeeModel: ['LXEK-3'],
        model: '9CZA-P470T-A1A',
        vendor: 'ADEO',
        description: 'ENKI LEXMAN E14 LED RGBW',
        extend: preset.light_onoff_brightness_colortemp_color(),
    },

thanks a lot !

sjorge, 2021-03-17:
Can post the databases.db entry so we can get the colorTempPhysicalMax and colorTempPhysicalMin values?

Gwenlad, 2021-03-17:
I don't know how to do that sorry :/
I just know the bulb is 2700 to 6500kelvin and 470lumen maximum.

I finally succeeded to have the devices.js in my docker, but the code doesn't work.

sjorge, 2021-03-17:
Ah you need to get it working first else the data wont be populated but once it is you can do something like

```
[root@amethyst /opt/zigbee2mqtt]# grep 0x00158d000361b295 data/database.db
{"id":50,"type":"Router","ieeeAddr":"0x00158d000361b295","nwkAddr":117,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RB 285 C","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RB 285 C","manufacturerName":"innr","powerSource":1,"zclVersion":2,"appVersion":16,"stackVersion":21,"hwVersion":1,"dateCode":"20180813-1","swBuildId":"2.0"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"lightingColorCtrl":{"attributes":{"colorTemperature":555,"currentX":45914,"currentY":19615,"currentHue":254,"currentSaturation":254,"colorMode":1,"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":555,"options":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":7,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":1,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":3,"minRepIntval":3,"maxRepIntval":300,"repChange":1},{"cluster":768,"attrId":4,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{"scenes":{"42_10105":{"state":{"state":"ON","color":{"x":0.7006,"y":0.2993}}}}}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":21,"hwVersion":1,"dateCode":"20180813-1","swBuildId":"2.0","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1615990936407}
```

Or if you have jq you can filter the data you want

```
[root@amethyst /opt/zigbee2mqtt]# grep 0x00158d000361b295 data/database.db | jq '[.endpoints."1".clusters|.genBasic.attributes.modelId,.lightingColorCtrl.attributes.colorTempPhysicalMin,.lightingColorCtrl.attributes.colorTempPhysicalMax]'
[
  "RB 285 C",
  153,
  555
]
```

Where 153 is the min and 555 is the max, this example is for a "RB 285 C", it differes per manufacture/bulb.

Gwenlad, 2021-03-17:
I found something like that ? 

`{"id":15,"type":"Router","ieeeAddr":"0x5c0272fffe2dff87","nwkAddr":20807,"manufId":4727,"manufName":"Adeo","powerSource":"Mains (single phase)","modelId":"LXEK-3","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64642],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LXEK-3","manufacturerName":"Adeo","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200526","swBuildId":"2.4.0"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200526","swBuildId":"2.4.0","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1615984528062}`

sjorge, 2021-03-17:
Yes, well once you get the devices.js change to work. The fields won't show up in the entry until the light is configured.

Gwenlad, 2021-03-17:
hmmm I'm not sure to understand :/ (my english is not very good)

Gwenlad, 2021-04-04:
I added external converters in my HA on docker. now I can modify the devices.js in /opt/zigbee-herdsman-converters 
I put the code of my first message, but it's still not working. I don't know what to do. (I restart Zigbee2mqtt and reboot my HA)

closed 2021-05-12 by github-actions[bot]