Koenkk/zigbee2mqtt #6740: Conbee II

hamdanfadi, 2021-03-17:
Hi,
I have conbee II. It is seems it is not working well. I have seen that it showing experimental status! What is that mean? Do I have to flash it? If yes, how I can go back to the original Firmware? 
Cheers,

m0wlheld, 2021-03-17:
I have a Raspbee II which is basically the same but for GPIO. Works well. 

What host do you use and how is the Conbee II connected to it?

hamdanfadi, 2021-03-17:
Thanks,
Mu host is Intel NUC. Has Home Assistant (Intel NUC edition)

Cheers,

Hamdan, Fadi
24B Carrick Glen Avenue, Flat Bush, Auckland 2019
New Zealand
***@***.******@***.***>
M+64 21 066 0088

m0wlheld, 2021-03-17:
Ok. And what errors do you face? Communication with the stick and/or network issues? Does the NUC have USB3 and - if so - do you use an extension cable, like it is recommended?

hamdanfadi, 2021-03-17:
I have the following issues:

  1.  502 Bad Gateway - I have resolved it by adding "adapter: deconz" in configuration, but when I restart my HA it returns back!!!! No solution yet.
  2.  I have Smartthing sensors all are discovered, but the buttons are not showing Power/Temperature status.

Do I have to upgrade its firmware? If yes which version and how?

Cheers,

Hamdan, Fadi
24B Carrick Glen Avenue, Flat Bush, Auckland 2019
New Zealand
***@***.******@***.***>
M+64 21 066 0088

hamdanfadi, 2021-03-17:
Just missed something to add, Yes there is a usb 3.0 and I'm using usb hub 3.0. What is the recommended?

Hamdan, Fadi
24B Carrick Glen Avenue, Flat Bush, Auckland 2019
New Zealand
***@***.******@***.***>
M +64 21 066 0088

m0wlheld, 2021-03-17:
Okay, just to clarify:
* you are using Home Assistant OS on an Intel NUC PC (the all-in-one version of Home Assistant)?
* you are using the Zigbee2MQTT add-on of Home Assistant or a dedicated Zigbee2MQTT installation?
* you have plugged in the Conbee II directly into the NUC which has USB3 ports in  use.

In any case, you should / must a USB extention cable to move the Conbee II away from the USB3 device (see https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html)

You should file an issue at [Home Assistant's Zigbee2MQTT add-on repository](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt) as the Conbee II is known to work with "plain" Zigbee2MQTT (although it is still marked "experimental")

You should also learn how Home Assistant is designed, that is uses ephemeral Linux container technologie and how it handles configuration.

Many issues here, but none directly related to this project IMHO.

hamdanfadi, 2021-03-17:
Here are my answers next to your questions.

  *   you are using Home Assistant OS on an Intel NUC PC (the all-in-one version of Home Assistant)? HA OS NUC Version.
  *   you are using the Zigbee2MQTT add-on of Home Assistant or a dedicated Zigbee2MQTT installation? Zigbee2MQTT addon from GitHub Repository
  *   you have plugged in the Conbee II directly into the NUC which has USB3 ports in use. USB HUB 3.0 plugged in USB 3.0 Port.
  *

hamdanfadi, 2021-03-18:
Hi there,
May I know what I should do to make it working?

Cheers,

m0wlheld, 2021-03-18:
> May I know what I should do to make it working?

Since communication with the stick works, but communication with devices is unreliable, I suggest reading https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html 

Start with an USB extension cable, than try to change the Zigbee Network Channel (or 2.4GHz WiFi channel) to reduce interferences. Warning: Chaning Zigbee channel requires re-pairing of devices.

closed 2021-04-25 by github-actions[bot]