Koenkk/zigbee2mqtt #6741: You suggested CC2652RB by slae.sh, I bought, but non answer on any channel, no good way

arripeo, 2021-03-17:
As I need a new powerful adapter I bought as suggested Cc2652RB by else.sh. I understood that is not Amazon, but I think that some must answer if I want to know when I can receive the order. No answer on any channel (telegram or mail). Anyone could think that this is a fraud...... be careful. More over it's impossible to write on the GitHub associated... any other contacts? Think about that you suggest it!

henkiejan1, 2021-03-17:
I don't have ever buyed something there, i don't nothing about that seller. But i take a look at there site and i read a lot of problems there. E-mail problems, payment problems and problems with the stick self...

https://slae.sh/projects/cc2652/#get-one

sjorge, 2021-03-17:
There is a discussion on people not getting updates and stuff here: https://github.com/Koenkk/zigbee2mqtt/discussions/6702

m0wlheld, 2021-03-17:
Why do you open an issue(!) for this?

Koenkk, 2021-03-17:
Let's keep this in #6702

closed 2021-03-17 by Koenkk