Koenkk/zigbee2mqtt #6746: Xiaomi Aqara WXKG12LM hold/release actions not detected by Home Assistant

derekoharrow, 2021-03-18:
I have several of [these Xiaomi WKG12LM buttons](https://www.zigbee2mqtt.io/devices/WXKG12LM.html). They are setup and detected fine in Z2M, and passed through to Home Assistant just fine.

Home Assistant detects and processes "clicks" just fine, but doesn't seem to detect "hold", "release" or "shake" actions. I think this may be because of the way Z2M is processing these events.

In Z2M, it defines the Home Assistant sensor as:

```
MQTT TOPIC: homeassistant/sensor/0x00158d000429e66a/action/config

{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state"
    },
    {
      "topic": "zigbee2mqtt/Cabin Button/availability"
    }
  ],
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x00158d000429e66a"
    ],
    "manufacturer": "Xiaomi",
    "model": "Aqara wireless switch (with gyroscope) (WXKG12LM)",
    "name": "Cabin Button",
    "sw_version": "Zigbee2MQTT 1.18.1"
  },
  "icon": "mdi:gesture-double-tap",
  "json_attributes_topic": "zigbee2mqtt/Cabin Button",
  "name": "Cabin Button action",
  "state_topic": "zigbee2mqtt/Cabin Button",
  "unique_id": "0x00158d000429e66a_action_zigbee2mqtt",
  "value_template": "{{ value_json.action }}"
}
```

This seems to define that the action should be defined by the JSON value called Action within the zigbee2mqtt/Cabin Button topic.

```
zigbee2mqtt/Cabin Button

{
  "action": "",
  "battery": 86,
  "last_seen": "2021-03-18T10:23:11+00:00",
  "linkquality": 57,
  "voltage": 2975
}
```

However, that value seems to be empty (or change very quickly to be empty). The real value for hold/release/shake seems to be sent in the MQTT topic zigbee2mqtt/Cabin Button/action:

```
zigbee2mqtt/Cabin Button/action
	hold
```

I wonder if the home assistant setup is wrong? Should the state topic instead be set to "zigbee2mqtt/Cabin Button/action" which does seem to hod the correct vlaue?

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: ZZH
Adapter firmware version: 20210120

Koenkk, 2021-03-18:
Make sure to check https://www.zigbee2mqtt.io/integration/home_assistant.html#responding-to-button-clicks . The clearing of the `action` is to trigger a state change for the sensor (so expected behaviour). 

If you see that the values are published to MQTT but you can't respond to it with HA I suggest to open an issue @ Home Assistant.

derekoharrow, 2021-03-22:
Thanks. I'll use MQTT directly.

closed 2021-03-22 by derekoharrow