Koenkk/zigbee2mqtt #6755: Unable to pair devices outside of coordinator range to any router

TTDARK, 2021-03-18:
## What happened
I just received Slaesh's CC2652RB stick today and trying to re-pair all the devices again migrating from CC2531 source routing firmware. 
I noticed that any device outside of the new coordinator's range won't pair at all (no message in logs), even they are in range of a router (in my case a OSRAM light bulb). I know they are in range because they worked fine before, using CC2531 with source routing firmware. 
Now I'm confused if a device has to be in range of the coordinator to be able to pair for the first time or is something in my setup defect.
It would be annoying to move them downstairs for pairing since a lot of them are glued in place and they just paired fine when I was using souce routing.

## What did you expect to happen
Pairing through a router should be possible.

## How to reproduce it (minimal and precise)
Start a pairing process with the device outside of the coordinator's range but in range of a connected router.

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: Slaesh's CC2652RB
Adapter firmware version: unknown

juslex, 2021-03-19:
what type of router are you using?  if it is nue / zemismart there are problems. Check: 

https://github.com/Koenkk/zigbee2mqtt/issues/5386#issue-774009388

TTDARK, 2021-03-19:
I'm using 4 Osram E27 light bulbs directly connected to the Slaesh stick (coordinator) and a CC2531 with antenna flashed as router upstairs. They used to communicate well with each other. But with the Slaesh stick, once the CC2531 is no longe in range of the coordinator (but still in range of one of the bulbs), it doesn't get recognized.
Looks like a problem with the bulb, which doesn't get the permit join command? I will let the CC2531 stick sit for a little while to see if it picks up the network by itself.
I saw another issue yesterday where koenkk said that the 2652 firmwares are all using source routing, so I'm confused now why they don't just do what they always did, since I was already using source routing with the old CC2531 stick.

closed 2021-04-26 by github-actions[bot]