Koenkk/zigbee2mqtt #6760: Last seen status in the device overview and last updated in dashboard

brjhaverkamp, 2021-03-19:
## Is your feature request related to a problem? Please describe
Indirectly. I have a problem with my temperature sensors not updating correctly in Domoticz, but reporting correctly in Z2M

## Describe the solution you'd like
I would like to have and "last seen" column with the date of the last message received in the device overview. So next to LQI and battery status.
Secondly I would like to have an update timestamp in the dashboard for the values shown. A small "date hour:minute" tag under   A bit like domoticz shows this. the tag doesn't have to be big, but is usefull for verifying how old the reading is, when in doubt.

## Describe alternatives you've considered
Currently I am reading the log to see if and when updates came into z2m:-)
[image]

## Additional context
I added a small screenshot of how the update time tag is currently shown on my dashboard in domoticz/dashticz.  I'm looking for something similar.

Thanks for this great piece of software and all the time you put in!

Koenkk, 2021-03-19:
> but reporting correctly in Z2M

If you see the data is published correctly to MQTT, Zigbee2MQTT isn't the issue. I suggest to create an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin

closed 2021-03-19 by Koenkk

brjhaverkamp, 2021-03-19:
Ai,Sorry Koenkk I shouldn't have tried to be so complete and try to fill in a "problem'.:-) It was just the background to my request.
The feature request is only loosely related. It would just be very handy information to have, in case you lose a sensor on your network and don't receive updates. I can't see this anywhere in the gui, except for diggin in the logs for one specific sensor.

The  last seen time would allow a user to check for sensors that have died or something and haven't reported in for a while.
The update time on a sensor reading would just be very handy!

Kind regards,
Bert

Koenkk, 2021-03-19:
For this you can use the `last_seen` option (see https://www.zigbee2mqtt.io/information/configuration.html#configuration)

brjhaverkamp, 2021-03-20:
Hi Koen,

Thanks for the pointer. But this adds the 'last_seen' value to an MQTT message. I'm not sure how this would help. I guess it would help in Node-red to create some mechanism to find lost devices. But my request is to have the information in the z2m GUI. 
Do I miss/overlook something?

  
Last_seen: 'disable'   # Optional: Add an elapsed attribute to MQTT messages, contains milliseconds since the previous msg (default: false)

brjhaverkamp, 2021-03-20:
Appologies, I just learned from a friend that this option also adds the column in the devices list in the gui. Problem solved