Koenkk/zigbee2mqtt #6761: ADDING BAC-002ALZB THERMOSTAT (FAN COIL UNIT AIR CONDITIONING)

juan11perez, 2021-03-19:
## Information about the device + link
[BAC-002ALZB](https://www.aliexpress.com/item/1005001441524917.html?spm=a2g0s.9042311.0.0.34534c4d0ZdyPu)

## data/database.db entry of the device
`{"id":9,"type":"Router","ieeeAddr":"0x5c0272fffedd2138","nwkAddr":108,"manufId":4098,"manufName":"_TZE200_dzuqwsyg","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_dzuqwsyg","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"reporting":1},"lastSeen":1616151697791}`

My first attempt at adding a device, so way over my head. Followed the instructions and I got it working, but I think my product is a bit of a hack.

So the device is very similar to [BHT-002-GCLZB](https://www.zigbee2mqtt.io/devices/BHT-002-GCLZB.html)

So i copied the corresponding devices.js entry with following minor mods:

`    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_dzuqwsyg'}],
        model: 'BAC-002-ALZB',
        vendor: 'HKGK',
        description: 'BAC series Thermostat',
        fromZigbee: [fz.hgkg_thermostat],
        toZigbee: [tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.hgkg_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
        exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 30, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['off', 'cool'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program']).withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
        onEvent: tuya.onEventSetLocalTime,
    },`

Main difference is that this device controls cooling as opposed to heating.
So for fromZigbee  I copied [fz.moes_thermostat] labelled it  [fz.hgkg_thermostat] and made only this change: 
return {system_mode: value ? 'cool' : 'off'}. see below complete entry:

`    hgkg_thermostat: {
        cluster: 'manuSpecificTuya',
        type: ['commandGetData', 'commandSetDataResponse'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp;
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
            let temperature;
            /* See tuyaThermostat above for message structure comment */
            switch (dp) {
            case tuya.dataPoints.moesSchedule:
                return {
                    program: [
                        {weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
                        {weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
                        {weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
                        {weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
                        {saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
                        {saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
                        {saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
                        {saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
                        {sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
                        {sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
                        {sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
                        {sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
                    ],
                };
            case tuya.dataPoints.state: // Thermostat on standby = OFF, running = ON
                return {system_mode: value ? 'cool' : 'off'};
            case tuya.dataPoints.moesChildLock:
                return {child_lock: value ? 'LOCK' : 'UNLOCK'};
            case tuya.dataPoints.moesHeatingSetpoint:
                return {current_heating_setpoint: value};
            case tuya.dataPoints.moesMaxTempLimit:
                return {max_temperature_limit: value};
            case tuya.dataPoints.moesMaxTemp:
                return {max_temperature: value};
            case tuya.dataPoints.moesDeadZoneTemp:
                return {deadzone_temperature: value};
            case tuya.dataPoints.moesLocalTemp:
                temperature = value & 1<<15 ? value - (1<<16) + 1 : value;
                return {local_temperature: parseFloat((temperature / 10).toFixed(1))};
            case tuya.dataPoints.moesTempCalibration:
                temperature = value;
                // for negative values produce complimentary hex (equivalent to negative values)
                if (temperature > 4000) temperature = temperature - 4096;
                return {local_temperature_calibration: temperature};
            case tuya.dataPoints.moesHold: // state is inverted, preset_mode is deprecated
                return {preset_mode: value ? 'program' : 'hold', preset: value ? 'program' : 'hold'};
            case tuya.dataPoints.moesScheduleEnable: // state is inverted, preset_mode is deprecated
                return {preset_mode: value ? 'hold' : 'program', preset: value ? 'hold' : 'program'};
            case tuya.dataPoints.moesValve:
                return {heat: value ? 'OFF' : 'ON'};
            case tuya.dataPoints.moesSensor:
                switch (value) {
                case 0:
                    return {sensor: 'IN'};
                case 1:
                    return {sensor: 'AL'};
                case 2:
                    return {sensor: 'OU'};
                default:
                    return {sensor: 'not_supported'};
                }
            default: // DataPoint 17 is unknown
                meta.logger.warn(`zigbee-herdsman-converters:Moes BHT-002: NOT RECOGNIZED DP #${
                    dp} with data ${JSON.stringify(msg.data)}`);
            }
        },
    }, `

For toZigbee i copied  tz.moes_thermostat_standby labelled it  tz.hgkg_thermostat_standby, as follows:

`    hgkg_thermostat_standby: {
        key: ['system_mode'],
        convertSet: async (entity, key, value, meta) => {
            await tuya.sendDataPointBool(entity, tuya.dataPoints.state, value === 'cool');
        },
    }, `

francisgambin, 2021-03-21:
Thanks Juan,

I am impatient to see the official support of this device, I have 5 in home

juan11perez, 2021-03-22:
@francisgambin 
Set up as per my instructions above and they will work.

francisgambin, 2021-03-22:
Doesn't work for me, I'm doing something wrong

Please help me

Pd. Mejor en Español Gracias!

> zigbee2mqtt@1.18.1 start /app
> node index.js
(node:470) UnhandledPromiseRejectionWarning: Error: Cannot find module 'https-proxy-agent'
Require stack:
- /share/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js
- /share/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/ledvance.js
- /share/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/index.js
- /share/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices.js
- /share/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js
- /app/lib/util/utils.js
- /app/lib/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/share/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:3:25)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Module.require (internal/modules/cjs/loader.js:952:19)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:470) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:470) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[21:41:14] INFO: Handing over control to Zigbee2mqtt Core ...

juan11perez, 2021-03-23:
Ok Francis, como es tu instalacion de z2m? Docker, directo?
Yo uso docker así que primero hay que instalar el módulo zigbee-herdsman-converters con estas [instrucciones](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)  final de la página. Luego montas el nuevo volumen en tu docker.

Después es sólo copiar las configuraciones que pegue arriba en los respectivos archivos.

Primero en débiles.js. Busca el device moes que ya existe y pega mi configuración debajo de ese.

Luego busca converters/fromZigbee y pega lo que puse para eso, luego converters/toZigbee y haces lo mismo.

francisgambin, 2021-03-23:
Pues no me funciona, mi instalación es en directo desde hassio con una Pi4 y el addon Zigbee2mqtt.

No se como hacer funcionar el converter, pero si tengo claro y como modificar los archivos *.js que propones

Gracias!

juan11perez, 2021-03-24:
@francisgambin 
No uso hassio asi que no puedo ayudar ahi. Las instrucciones para montar el modulo estan el la misma pagina que mencione anteriormente.
Tienes que montar el modulo para poder modificar esos archivos.

francisgambin, 2021-03-24:
Gracias @juan11perez 
Pues se trandrá que esperar al soporte oficial, espero que llegue algún día (igual podrías agregarlo tu mismo)

**Please official support!!**

francisgambin, 2021-03-24:
An advance, I am testing from windows and works!!

{"id":2,"type":"Router","ieeeAddr":"0x60a423fffeb280cf","nwkAddr":44904,"manufId":4098,"manufName":"_TZE200_dzuqwsyg","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_dzuqwsyg","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1616620295995}

juan11perez, 2021-03-27:
@francisgambin Added this pull requests:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/2420
https://github.com/Koenkk/zigbee2mqtt.io/pull/681

closed 2021-03-27 by juan11perez

francisgambin, 2021-03-28:
Muchas gracias @juan11perez 
Pues uso CC2531 con Z-Stack-firmware
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

juan11perez, 2021-03-30:
@francisgambin 
el termostato ya esta en el dev branch. si usas homeassintant addon, creo que puedes seleccionar latest-dev

Prim0z, 2021-07-08:
BAC-002-ALZB thermostat supports heating, cooling and ventilation. With your submit you can only control cooling? Is your use case only for cooling? I also use heating and ventilation.

At current stage we are missing mode setpoint (heating/cooling/ventilation) and fan speed setpoint (1/2/3/auto).

Is anyone willing to implement this futures? I don't have deep enough technical knowledge about zigbee2mqtt to do it myself, but I can provide data logs and testing resources. I can also send you a BAC-002-ALZB thermostat for testing and development. No need to return it, you can keep it.

Levia1han, 2021-12-05:
How did you solve the problem of a large number of packets on the zigbee network? The device is sending more than 5 message per second.

Prim0z, 2021-12-05:
You can't. 

> How did you solve the problem of a large number of packets on the zigbee network? The device is sending more than 5 message per second.

You can't. That is up to device firmware. You can contact the manufacturer or add debounce to zigbee2mqtt to filter out the duplicates.

a2nt, 2022-07-12:
Were you able to control Fun speed?
[image]