Koenkk/zigbee2mqtt #6772: 'No network route' (205))' after changing port from usb 3.0 to usb 2.0 (cc2531 in a raspberry pi4)

kitopopo, 2021-03-20:
Dear friend @Koenkk ,

When all change my stick from usb 3.0 to usb 2.0 some zigbee devices no works. i can show the following log messange in a device:

```
mar 20 22:47:23 raspberrypi npm[27485]: Zigbee2MQTT:error 2021-03-20 22:47:23: Publish 'set' 'state' to '0x00158d00030a6c7e' failed: 'Error: Command 0x00158d00030a6c7e/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'
mar 20 22:47:23 raspberrypi npm[27485]: Zigbee2MQTT:info  2021-03-20 22:47:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x00158d00030a6c7e' failed: 'Error: Command 0x00158d00030a6c7e/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x00158d00030a6c7e"},"type":"zigbee_publish_error"}'
```
If i make a network map i can see also 2 zigbee repeaters (xiaomi plugs) with LQI error.

mar 20 23:00:41 raspberrypi npm[29885]: Zigbee2MQTT:info  2021-03-20 23:00:41: Starting network scan (includeRoutes 'true')
mar 20 23:00:54 raspberrypi npm[29885]: Zigbee2MQTT:error 2021-03-20 23:00:54: Failed to execute LQI for '0x00158d0002675428'
mar 20 23:01:04 raspberrypi npm[29885]: Zigbee2MQTT:error 2021-03-20 23:01:04: Failed to execute routing table for '0x00158d0002675428'
mar 20 23:01:15 raspberrypi npm[29885]: Zigbee2MQTT:error 2021-03-20 23:01:15: Failed to execute LQI for '0x00158d00039bba90'
mar 20 23:01:25 raspberrypi npm[29885]: Zigbee2MQTT:error 2021-03-20 23:01:25: Failed to execute routing table for '0x00158d00039bba90'
mar 20 23:01:25 raspberrypi npm[29885]: Zigbee2MQTT:info  2021-03-20 23:01:25: Network scan finished
Especially xiaomi plugs fail me. and xiaomi temperature sensors. the aperture sensors and xiaomi swtichs seem to work fine.

I have updated my cc2531 stick firmware from initial version to 20201128 but same problem
With the stick inserted in the usb 3.0 i test: ls -l /dev/serial/by-id and appear the cc2531 conected to ttyACM0 , same in usb 2.0 and same in the configuration.yaml file.- All zigbee2mqtt los seems stopping and having very little activity. It seems there are no interviews.

Changin the stick again to usb 3.0 port all works correctly again. 

Any idea of whats happens and why i can't use a usb 2.0 port in my raspberry pi 4?

Thanks in advanced, best regards

## What did you expect to happen

Thath the cc2531 should be works correctly in all usb ports of raspberry pi 4.

## How to reproduce it (minimal and precise)

Changing cc2531 stick from usb 3.0 to usb 2.0 in a Raspberry pi 4

## Debug info
Zigbee2MQTT version: 1.18.1-dev
Adapter hardware: CC2531
Adapter firmware version: revision":20201128

Koenkk, 2021-03-21:
Is the CC2531 positioned +- 20cm from the pi using a usb extension cable?

kitopopo, 2021-03-21:
ohh, wonderful, you are a teacher . This really works, what strange things. I am waiting the shipment for the Zzh stick to, I hope everything works as well as with the cc2531. Thanks. Best regards

closed 2021-03-21 by kitopopo

kitopopo, 2021-04-06:
Dear @Koenkk ,

why can't the stick be close to the raspberry pi4? What does the stick interfere with? the new zzh stick cannot be in any usb port. neither 2.0 nor 3.0. It has to be quite far from the raspberry, further than 20cm. to what is due? thanks in advance

Koenkk, 2021-04-06:
Because the Pi 4 generates a lot of interference especially the USB 3.0 ports.

kitopopo, 2021-04-09:
> Because the Pi 4 generates a lot of interference especially the USB 3.0 ports.

the new stick ssh CC2652R no works connected to any usb 2.0 port and also no works connected to any. 3.0. port. It only works with an extension of more than 0.20 cm