Koenkk/zigbee2mqtt #6775: Attribute 'last_seen' is not correctly saved into 'state.json'

snoopyCode, 2021-03-21:
## What happened

The current state of a device is periodically seen saved into 'state.json' every (currently) 5 min. The 'last_seen' attribute in this file does not reflect the real last seen timestamp of the device (relative to the save interval for sure).

## What did you expect to happen

In 'state.json' there should the 'last_seen' value of the device which is the current at the point in time where 'state.json' is saved.

## How to reproduce it (minimal and precise)

Compare the 'last_seen' attribute of a device in 'state.json' compared to the 'last_seen' value in the message which is put to mqtt. This deviates quite a bit in some cases.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: 20190223

Koenkk, 2021-03-21:
`last_seen` is kept in the underlying zigbee-herdsman, there is no intention to keep it in the `state.json`. What is your use case for this?

snoopyCode, 2021-03-24:
I have seen that `last_seen` is stored in 'state.json' already but the dates are not the same as been sent over mqtt. In my opinion confusing. Would be better to remove it completely or store the last right value.

With storing in the underlying layer you probably mean in 'database.db'. That's true, however I recognised that it is also outdated there because the file is not saved often. It seems to me only if e.g. new devices are paired but I haven't checked that.

But well, `state.json` and `database.db` are internal data structures, probably not intended to be used directly for use cases.

I am currently working on a controller which checks the health of the instance and fails over to another server in case it detects issues with the instance, zigbee device or simply if the server failed. Unfortunately zigbee does not have a real failover concept in case the controller dies. For that use cases I used the most direct way to see if messages are received on the zigbee device. Using mqtt adds another component which can fail. Open for better ideas :)

Koenkk, 2021-03-25:
I've checked again, the `last_seen` you have in your `state.json` are probably outdated, I expect that after removing those manually they will not be added back.

> But well, state.json and database.db are internal data structures, probably not intended to be used directly for use cases.

True, they are indeed not meant for this.

> For that use cases I used the most direct way to see if messages are received on the zigbee device. Using mqtt adds another component which can fail. Open for better ideas :)

You can use https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgerequesthealth_check and https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgestate

closed 2021-05-03 by github-actions[bot]