Koenkk/zigbee2mqtt #6778: Startup Error on latest update "AREQ - ZDO - stateChangeInd after 60000ms"

MrLeonida, 2021-03-21:
Today I have updated my Home Assistant Core and OS on my RB Pi4 to the latest version.
I have also updated my Mosquito and Zigbee2MQTT addon to the latest version.
Now, starting Zigbee2MQTT service I receive this error.

Zigbee2MQTT:debug 2021-03-21 17:17:45: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2021-03-21 17:17:45: Logging to console and directory: '/share/zigbee2mqtt/log/2021-03-21.17-17-44' filename: log.txt
Zigbee2MQTT:debug 2021-03-21 17:17:45: Removing old log directory '/share/zigbee2mqtt/log/2021-03-21.17-06-19'
Zigbee2MQTT:info  2021-03-21 17:17:45: Starting Zigbee2MQTT version 1.18.1 (commit #1.18.1)
Zigbee2MQTT:info  2021-03-21 17:17:45: Starting zigbee-herdsman (0.13.71)
Zigbee2MQTT:debug 2021-03-21 17:17:45: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_3A_10-if00-port0"}}'
Zigbee2MQTT:error 2021-03-21 17:18:53: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-03-21 17:18:53: Failed to start zigbee
Zigbee2MQTT:error 2021-03-21 17:18:53: Exiting...
Zigbee2MQTT:error 2021-03-21 17:18:53: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-03-21T16_18_53_720Z-debug.log

Adapter hardware: Slaesh's CC2652RB stick

I've read other similar posts before, but I don't really know what to do.

Can someone help me?

Koenkk, 2021-03-21:
After reinstalling, did you recover the `coordinator_backup.json` and `configuration.yaml`? Otherwise you have to change your pan id and repair your devices: 3rd last point of https://www.zigbee2mqtt.io/information/supported_adapters.html#notes

MrLeonida, 2021-03-22:
Thank you @Koenkk.

I finally solved it with a new flash of the adapter, following the guides proposed on the manufacturer's website. It was not necessary to re-pair all devices. I believe that sometimes updating MQTT can corrupt the adapter fw.

closed 2021-03-22 by MrLeonida