Koenkk/zigbee2mqtt #6779: Hive SLR2 not pairing correctly

peterguy04, 2021-03-21:
Hi i am trying to get z2m to work for a few things but having bother with 1 device.

Hive SLR2 is failing to pair properly, i see the support is there but the log is complaining that its not supported. i have tried multiple times to pair it with same result 
log shows this 
```Zigbee2MQTT:info  2021-03-21 16:24:37: SLR Heating (0x001e5e0902******): Not supported (Router)```
and occasionally pops up this 
```Recieved message from unsupported device with Zigbee model 'undefined' and manufacturer name 'computime'```

in the database.db it shows `"manufId":4153,"manufName":"Computime"` but no model ID in here , model identifier is displayed as SLR2 in deconz so not sure if this has anything to do with it

## Debug info
Zigbee2MQTT version: 1.18.1 (:latest docker tag today)
Adapter hardware: Conbee II
Adapter firmware version: 26680700

Koenkk, 2021-03-22:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails

peterguy04, 2021-03-22:
@Koenkk thanks, i have read that and enabled herdsman debug, interview did complete and the only thing it really showed wrong was ```supported: false```, i put the coordinator on a extension, re-paired about 5 times but still same issue on those specific devices.

going on i decided to have a look in the databse.db file, all devices in there had a `modelId` except the 2 i mentioned above ... i decided to try manually insert ```"modelId":"SLR2"``` for the hive reciever into the line for that device  and the same for the light with its own model . now they both seem to work ok from the testing i did quickly... no idea why i had this issue to begin with but thought it might be worth mentioning how i fixed it. i guess it just wasnt reading the modelid .

closed 2021-04-30 by github-actions[bot]