Koenkk/zigbee2mqtt #6780: Group turns other groups lights on

DenisLug, 2021-03-21:
## What happened
I have created two groups of lights. Group 1 with 10 and group 2 with 3 Philips Hue lights. I can turn on and off all 10 lights of group 1 independently of group 2. When I turn group 2 on or off then also the lights of group 1 are triggered.

## What did you expect to happen
Lights of group 2 should turn on/off independently to group 1.

## How to reproduce it (minimal and precise)
I added the lights in the configuration.yaml and when it did not worked I also tried it by using the MQTT messages.

This is my database.db after adding the light with MQTT:
```
{"id":33,"type":"Group","groupID":1,"members":[{"deviceIeeeAddr":"0x00178801084e87b1","endpointID":11},{"deviceIeeeAddr":"0x00178801083f63e7","endpointID":11},{"deviceIeeeAddr":"0x00178801083f6ad1","endpointID":11},{"deviceIeeeAddr":"0x0017880108461b3e","endpointID":11},{"deviceIeeeAddr":"0x0017880108461b59","endpointID":11},{"deviceIeeeAddr":"0x00178801084e9514","endpointID":11},{"deviceIeeeAddr":"0x001788010846212d","endpointID":11},{"deviceIeeeAddr":"0x00178801084e87d4","endpointID":11},{"deviceIeeeAddr":"0x0017880108462bd8","endpointID":11}],"meta":{}}
{"id":34,"type":"Group","groupID":2,"members":[{"deviceIeeeAddr":"0x00178801084905f9","endpointID":11},{"deviceIeeeAddr":"0x0017880108461b4d","endpointID":11},{"deviceIeeeAddr":"0x00178801084ede43","endpointID":11}],"meta":{}}
```

This is the resulting configuration.yaml entry for groups:
```
groups:
  '1':
    friendly_name: '{"friendly_name": Sofa, "id":1}'
    devices:
      - 0x00178801084e87b1/11
      - 0x00178801083f63e7/11
      - 0x00178801083f6ad1/11
      - 0x0017880108461b3e/11
      - 0x0017880108461b59/11
      - 0x00178801084e9514/11
      - 0x001788010846212d/11
      - 0x00178801084e87d4/11
      - 0x0017880108462bd8/11
  '2':
    friendly_name: '{"friendly_name": Esstisch, "id":2}'
    devices:
      - 0x00178801084905f9/11
      - 0x0017880108461b4d/11
      - 0x00178801084ede43/11
```

By sending "ON" or "OFF" to the topic "zigbee2mqtt/2/set" the lights of both groups turn on/off.

## Debug info
Zigbee2MQTT version: 1.18.1 (commit #a43d1f3, Docker on Synology)
Adapter hardware: Conbee II
Adapter firmware version: 0x26580700

attkri, 2021-03-22:
**Note**

Just noticed that there is a configuration of group with switches and lamps there HA is updated. The other 10 groups are not. Unfortunately, the settings regarding bind, reporting and settings are 100% the same for both groups.

DenisLug, 2021-03-22:
The problem occurs in zigbee2mqtt not in HA. I don't know what you mean with "the other 10 groups".

attkri, 2021-03-22:
I have 11 zigbee2mqtt groups with lamps and one physical switch bound per group. And as I said only 1 group behaves as expected. When I press this switch a) the lamps in the zigbee2mqtt group are turned off and b) in HA the lamp symbols go off too. In the other groups the lamps in HA stay on although they actually went out.

DenisLug, 2021-04-02:
I could reproduce the problem with other lights. It has something to do with the group id. When I swapped the IDs between the two groups it worked as expected. Maybe a developer could take a closer look at this.

attkri, 2021-04-05:
Note: As part of a channel change, I have to pair all devices again. However, during this action, the grouping of the lamps was not retransmitted as expected. Have also changed in the groups.yaml (in which all is still as it should be) from IEEE-Adr to Friendly_Name, restarted, without success.

I will now try to assign the lamps to their groups via MQTT commands because it is very tedious via the interface.

attkri, 2021-05-06:
No change even after updating to core-2021.5.0. Problem persists:

1. **fan.set_speed = auto** is not set:

According to documentation https://www.home-assistant.io/integrations/xiaomi_miio/#service-fanset_speed the values are not accepted but only the following: low, medium and high. **auto** is missing and the given values do not work at all on device zhimi.humidifier.ca4.

2. **water_level** is also still missing.

attkri, 2021-05-06:
Problem still exists. The switch of a switch that does not switch light by automation but to the switching of a Zigbee group lights group controls the light but in HA the state change is not displayed at the lights.

closed 2021-06-14 by github-actions[bot]