Koenkk/zigbee2mqtt #6784: Xiaomi QBKG11LM reporting actions when none happening

derekoharrow, 2021-03-22:
I'm using a [Xiaomi QBKG11LM](https://www.zigbee2mqtt.io/devices/QBKG11LM.html) which I have set to "decoupled" mode as per the wiki.

The relay is working fine and I'm monitoring the following topic to look for actions:

`zigbee2mqtt/Cabin Master Switch/action`

For some reason, the action value seems to be being set to single every so often, even though the button isn't being pressed on the switch.

Here's a debug log file output (during which time the switch wasn't pressed):

```
Zigbee2MQTT:debug 2021-03-22 09:50:26: Received Zigbee message from 'Cabin Master Switch', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":1.0800000429153442}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2021-03-22 09:50:26: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"consumption":92.7,"energy":92.7,"last_seen":"2021-03-22T09:50:26+00:00","linkquality":63,"operation_mode":"decoupled","power":1.08,"state":"ON","temperature":21}'
Zigbee2MQTT:debug 2021-03-22 09:53:02: Received Zigbee message from 'Cabin Master Switch', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-03-22 09:53:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"single","click":"single","consumption":92.7,"energy":92.7,"last_seen":"2021-03-22T09:53:02+00:00","linkquality":72,"operation_mode":"decoupled","power":1.08,"state":"ON","temperature":21}'
Zigbee2MQTT:info  2021-03-22 09:53:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"","consumption":92.7,"energy":92.7,"last_seen":"2021-03-22T09:53:02+00:00","linkquality":72,"operation_mode":"decoupled","power":1.08,"state":"ON","temperature":21}'
Zigbee2MQTT:info  2021-03-22 09:53:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"click":"","consumption":92.7,"energy":92.7,"last_seen":"2021-03-22T09:53:02+00:00","linkquality":72,"operation_mode":"decoupled","power":1.08,"state":"ON","temperature":21}'
Zigbee2MQTT:info  2021-03-22 09:53:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch/action', payload 'single'
Zigbee2MQTT:info  2021-03-22 09:53:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch/click', payload 'single'
```

To me it doesn't look like z2m is receiving anything from the switch that indicates the action should be activated. It seems to be doing this every few minutes, making the switch effectively useless.

Any ideas?

Thanks

Zigbee2MQTT version: 1.18.1
Adapter hardware: ZZH - CC2652R
Adapter firmware version: 20210120

derekoharrow, 2021-03-22:
Just noticed that this is happening exactly every 5 minutes, which I presume is some kind of regular status reporting (perhaps Temperature & power). But the switch isn't pressed at any point:

```
Zigbee2MQTT:info  2021-03-22 10:03:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"single","click":"single","consumption":92.7,"energy":92.7,"last_seen":"2021-03-22T10:03:02+00:00","linkquality":39,"operation_mode":"decoupled","power":9.08,"state":"ON","temperature":22}'
Zigbee2MQTT:info  2021-03-22 10:08:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"single","click":"single","consumption":92.71,"energy":92.71,"last_seen":"2021-03-22T10:08:02+00:00","linkquality":75,"operation_mode":"decoupled","power":1.09,"state":"ON","temperature":22}'
Zigbee2MQTT:info  2021-03-22 10:13:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"single","click":"single","consumption":92.71,"energy":92.71,"last_seen":"2021-03-22T10:13:02+00:00","linkquality":72,"operation_mode":"decoupled","power":8.9,"state":"ON","temperature":22}'
Zigbee2MQTT:info  2021-03-22 10:18:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"single","click":"single","consumption":92.71,"energy":92.71,"last_seen":"2021-03-22T10:18:02+00:00","linkquality":72,"operation_mode":"decoupled","power":1.14,"state":"ON","temperature":22}'
Zigbee2MQTT:info  2021-03-22 10:23:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"single","click":"single","consumption":92.71,"energy":92.71,"last_seen":"2021-03-22T10:23:02+00:00","linkquality":69,"operation_mode":"decoupled","power":8.9,"state":"ON","temperature":22}'
Zigbee2MQTT:info  2021-03-22 10:28:02: MQTT publish: topic 'zigbee2mqtt/Cabin Master Switch', payload '{"action":"single","click":"single","consumption":92.71,"energy":92.71,"last_seen":"2021-03-22T10:28:02+00:00","linkquality":63,"operation_mode":"decoupled","power":1.08,"state":"ON","temperature":21}'
```

Koenkk, 2021-03-22:
Can you try setting `legacy: false` for this device in the `configuration.yaml`?

e.g.

```yaml
devices:
  '0x12345678':
    friendly_name: my_QBKG11LM
    legacy: false  # <- add this
```

ademobile, 2021-03-22:
I also have one doing exactly the same, however mine is not running in legacy mode or decoupled

derekoharrow, 2021-03-22:
> Can you try setting `legacy: false` for this device in the `configuration.yaml`?
> 
> e.g.
> 
> ```yaml
> devices:
>   '0x12345678':
>     friendly_name: my_QBKG11LM
>     legacy: false  # <- add this
> ```

Done, and so far it looks good. Only been running like that for about 30 mins, but no updates to the action every 5 mins now.

Thanks

closed 2021-04-30 by github-actions[bot]

knoopx, 2022-01-14:
I'm having also this problem and I think it's happening since an OTA upgrade a few months ago. I have two devices, one is working properly but the other is randomly turning on and off. If I shut down the pi, it automatically turns on every 4:30min aprox. Legacy mode is disabled yet still happening. I removed the device and repaired thousand times.