Koenkk/zigbee2mqtt #6785: Heiman water Sensor not recognized

kirelam, 2021-03-22:
I tried following: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

the following I did not get working: 
Now add the volumes to the Docker container by adding the following to your docker run command.:

-v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters

The bought device is the following: 
https://nl.aliexpress.com/item/4000028480488.html?spm=a2g0s.9042311.0.0.27424c4dYllaqE

## Information about the device + link
Friendly name
0x842e14fffeed6807
Device type
EndDevice
Zigbee model
WaterSensor-EF-3.0
Support status
Unsupported

IEEE address
0x842e14fffeed6807
Network address
0xFA00
Firmware build date
2019.5.23
Power source
Battery 
Interview completed
Yes

Nonetheless the device is in the supported list. The device is not recognized.

## data/database.db entry of the device
..

Koenkk, 2021-03-22:
Your version is slightly different and was not supported yet. Added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-03-22 by Koenkk

kirelam, 2021-03-22:
Thank you sooooooo much <3