Koenkk/zigbee2mqtt #6786: Xiaomi QBKG25LM SOFTWARE_FAILURE

balonchiks, 2021-03-22:
## What happened

The device I am referring to is Aqara D1 3 Gang switch ([QBKG25LM](https://www.zigbee2mqtt.io/devices/QBKG25LM.html))

Upon z2m start, i see the following lines in the log:

```
info  2021-03-20 20:10:46: Configuring 'Aqara Wall Switch Office 1'
error 2021-03-20 20:10:48: Failed to configure 'Aqara Wall Switch Office 1', attempt 1 (Error: ConfigureReporting 0x04cf8cdf3c7f3df4/3 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'SOFTWARE_FAILURE')
    at Endpoint.checkStatus (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23)
    at Endpoint.<anonymous> (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:419:26)
    at Generator.next (<anonymous>)
    at fulfilled (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
    at <anonymous>)
```

There are 3 attempts of this configuration, and all 3 fail with the same error. The device itself is somewhat working (skipping 2 out of 3 (or so) button press), yet it is barely usable due to skipping button presses.

## What did you expect to happen
Just paired the switch with the z2m and start using it.

## How to reproduce it (minimal and precise)
Pair device to z2m, restart z2m and observe logs. the issue happens right after restart, 3 times with about 2 minutes interval in between the errors.

## Debug info
Zigbee2MQTT version: 1.18.1 (same thing happens on 1.17.1 and 1.18.0 and latest dev, haven't tried earlier versions)
Adapter hardware: CC2538 (modkam.ru stick v3)
Adapter firmware version: 20200211

fennec622, 2021-03-28:
Same problem

balonchiks, 2021-04-04:
apart from this, what i've also noticed is that I cannot control the relay state from homeassistant (via mqtt). It always timeouts:

` MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Aqara Wall Switch Corridor 1' failed: 'Error: Command 0x04cf8cdf3c7f3df4/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'Timeout' (9999))'","meta":{"friendly_name":"Aqara Wall Switch Corridor 1"},"type":"zigbee_publish_error"}'`

it does not work for all 3 buttons (left, center and right)

balonchiks, 2021-04-08:
so it turns out that all of this only happens when you turn some of the buttons to `decoupled` mode, and it does not happen always. 

so after 3 attempts (_remove from z2m -> start pairing -> set 1 or more buttons to decouple mode -> check if turning on relay are working from homeassistant/mqtt/z2m dashboard_) - mine got working fine. maybe it's the timing that matters cause first 2 attempts that i made were made like 15 minutes after pairing the switch into the network, and 3rd attempt was done instantly after pairing.

If you do not use `decoupled` mode - there seems to be no issues with the switch.

closed 2021-04-08 by balonchiks

ahmadarifabdulkadir, 2021-05-07:
Hi,

i am having the same problem but i did not use decouple mode. 

Before the existance of this error:
- i was able to use this perfectly

After the existance of this error:
- the error only comes up when zigbee2mqtt restarts.
- if i publish mqtt message to the device , time out.
- if i press physical button after the error, the log shows the mqtt messages. only then it starts working again.

My current workaround, everytime zigbee2mqtt restarts, i re-pair the device to the coordinator.

Additional question, 
1) is this device capable of doing OTA update? in the zigbee2mqtt page, it says that this device is capable of doing OTA update. 
2) Can this device act as router?